arendst/Tasmota #9915: Tasmota 9.1.0.2 Shelly Dimmer 2

oziee, 2020-11-19:
### PROBLEM DESCRIPTION
flashed latest Tasmota 9.1.0.2 to Shelly Dimmer 2.. lights just are ON.. 
no toggle for light on or off.. there is no dim either

My shelly is connected to a momentary switch.. when this is pressed the log changes to 
```
13:58:14 MQT: stat/tasmota_95A634/RESULT = {"POWER":"ON"}
13:58:14 MQT: stat/tasmota_95A634/POWER = ON
13:58:18 MQT: stat/tasmota_95A634/RESULT = {"POWER":"OFF"}
13:58:18 MQT: stat/tasmota_95A634/POWER = OFF
```

however the light never turns on or off.. not even if I hold the button in

Configuration output here:
```
{"NAME":"Lounge Dimmer 2","GPIO":[0,3200,0,3232,5504,5536,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
```

log output 
```
00:00:00 CFG: Loaded from flash at FB, Count 9
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.1.0.2(tasmota)-2_7_4_7
00:00:00 WIF: Connecting to AP1 OziesNoT Channel 6 BSSId 86:83:C2:91:4C:56 in mode 11N as tasmota_95A634-1588...
00:00:01 WIF: Connected
00:00:02 HTP: Web server active on tasmota_95A634-1588 with IP address 10.0.4.67
13:56:04 MQT: Attempting connection...
13:56:04 MQT: Connected
13:56:04 MQT: tele/tasmota_95A634/LWT = Online (retained)
13:56:04 MQT: cmnd/tasmota_95A634/POWER = 
13:56:04 MQT: tele/tasmota_95A634/INFO1 = {"Module":"Lounge Dimmer 2","Version":"9.1.0.2(tasmota)","FallbackTopic":"cmnd/DVES_95A634_fb/","GroupTopic":"cmnd/tasmotas/"}
13:56:04 MQT: tele/tasmota_95A634/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_95A634-1588","IPAddress":"10.0.4.67"}
13:56:04 MQT: tele/tasmota_95A634/INFO3 = {"RestartReason":"Software/System restart"}
13:56:05 MQT: stat/tasmota_95A634/STATUS10 = {"StatusSNS":{"Time":"2020-11-19T13:56:05","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.0},"TempUnit":"C"}}
13:56:05 MQT: stat/tasmota_95A634/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_95A634","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:56:05 MQT: stat/tasmota_95A634/STATUS11 = {"StatusSTS":{"Time":"2020-11-19T13:56:05","Uptime":"0T00:00:06","UptimeSec":6,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"OziesNoT","BSSId":"86:83:C2:91:4C:56","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:56:07 QPC: Reset
13:56:08 MQT: tele/tasmota_95A634/STATE = {"Time":"2020-11-19T13:56:08","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"OziesNoT","BSSId":"86:83:C2:91:4C:56","Channel":6,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:56:08 MQT: tele/tasmota_95A634/SENSOR = {"Time":"2020-11-19T13:56:08","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.0},"TempUnit":"C"}
```

### EXPECTED BEHAVIOUR
ability to turn off or on or dim.. its just on 

to be able to turn it off

closed 2020-11-19 by close-issue-app[bot]

arendst, 2020-11-19:
And again we have a user using the wrong template. See #9900 and

```
 * {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5568,5600,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
 * {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
```

oziee, 2020-11-19:
> And again we have a user using the wrong template. See #9900 and
> 
> ```
>  * {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5568,5600,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
>  * {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
> ```

https://templates.blakadder.com/shelly_Dimmer_2.html

used the Template from the docs

arendst, 2020-11-19:
OK. That's wrong. I will update it. Thx.

oziee, 2020-11-19:
awesome work:)

do you know the setting to change so that the switch can be a momentary ??

arendst, 2020-11-19:
Play with different ``SwitchMode`` options.