arendst/Tasmota #9916: Adding RC522 RFID reader to supported drivers - first working attempt attached

pookycade, 2020-11-19:
**Have you looked for this feature in other issues and in the docs?**  Yes

**Is your feature request related to a problem? Please describe.**  No
_A clear and concise description of what the problem is._

**Describe the solution you'd like** 
_A clear and concise description of what you want to happen._

RC522 readers are far more prevalent than PN532 readers simply due to cost.  There is not current support for them in Tasmota

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

I have made an RC522 work at its most rudimentary level by hacking into the PN532 ino. It works, but its obviously not clean and I have to compile my own BIN for this project. Perhaps someone with more time/skills can take this further or pretty it up so it is more acceptable to the main code base

This can be renamed .ino to use and just plug it right in and you have to also add the MFRC522.h library to the build.

Its only function right now is that when you scan an RFID card it will report out in the console the number of this card in the sensor data. It is formatted along the lines of tele/CardSwipe5/SENSOR = {"Time":"2020-11-19T07:20:53","PN532":{"UID":AFF2BA}}  It works for my purposes but is rudimentary and an admitted complete hack.  Someone else can do better
[xsns_40_pn532.txt](https://github.com/arendst/Tasmota/files/5567626/xsns_40_pn532.txt)

**(Please, remember to close the issue when the problem has been addressed)**

pookycade, 2020-11-20:
One additional bit of info.  I am trying to get an entity recognized in HA for this hacked sensor routine. I try to follow the expected syntax that is documented for the PN532 sensor as shown below from the Console

MQT: tele/Cardswipe6/SENSOR = {"Time":"2020-11-20T08:38:50","PN532":{"UID":"77B0A65F", "DATA":"ILOVETASMOTA"}}

The New Tasmota(Beta) 9.10 Integration picks up my V53l0x, my PCF8574 output, but will not see this RFID no matter how I seem to format the MQT messages.  

Now I realize that my expectation that any of this should work is ZERO.  But if its something simple, I'd love to get it working. 
 Being clueless about this, can anyone perhaps provide insight into what is needed for an auto integration via Tasmota inside HA ? Why does it correctly set up the V53l0x sensor which is integrated in much the same manner

MQT: tele/Cardswipe6/SENSOR = {"Time":"2020-11-20T08:44:50","VL53L0X":{"Distance":9999}}

That above works perfectly.  

I even tried paring it down to this

MQT: tele/Cardswipe6/SENSOR = {"Time":"2020-11-20T08:38:50","PN532":{"UID":"77B0A65F"}}

No go.

ascillato2, 2020-12-06:
Closing this request as it has been added by Theo.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-06 by ascillato2

rapi3, 2021-02-21:
Hi
At this moment the reader work but antena range it is low.
Do you think Is it possible to add SetOptions to set antenna gain value RFCfgReg 48.
thank you