arendst/Tasmota #9923: zbBridge Aqara Button WXKG12LM - Shake not reportet

toto79, 2020-11-20:
### PROBLEM DESCRIPTION
in tasmota zbBridge Firmware action shake (Gyro function) for WXKG12LM is not reportet. Other actions are 1,2,3 instead of click, hold, ... like for WXKG11LM -> solution for me, see at bottom of this post

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ ] Tasmota binary firmware version number used: 9.1.0.2 | 20201120 00:17
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ota update

- [x] Provide the output of this command: `Status 0`:
```
06:34:29 CMD: status 0
06:34:29 MQT: stat/zbBridge/STATUS = {"Status":{"Module":75,"DeviceName":"ZigbeeBridge","FriendlyName":["Tasmota"],"Topic":"zbBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:34:29 MQT: stat/zbBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:22:53","StartupUTC":"2020-11-20T05:11:36","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-10-30T18:05:26","SaveCount":127,"SaveAddress":"FA000"}}
06:34:29 MQT: stat/zbBridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(zbbridge)","BuildDateTime":"2020-11-20T05:08:53","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"419/699"}}
06:34:29 MQT: stat/zbBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Streaming2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006002","00000000"]}}
06:34:29 MQT: stat/zbBridge/STATUS4 = {"StatusMEM":{"ProgramSize":576,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
06:34:29 MQT: stat/zbBridge/STATUS5 = {"StatusNET":{"Hostname":"SonZbBridge1","IPAddress":"192.168.4.233","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"A4:CF:12:DC:0B:55","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:34:29 MQT: stat/zbBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.4.3","MqttPort":1883,"MqttClientMask":"zbBridge","MqttClient":"zbBridge","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:34:29 MQT: stat/zbBridge/STATUS7 = {"StatusTIM":{"UTC":"2020-11-20T05:34:29","Local":"2020-11-20T06:34:29","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"08:07","Sunset":"17:03"}}
06:34:29 MQT: stat/zbBridge/STATUS10 = {"StatusSNS":{"Time":"2020-11-20T06:34:29"}}
06:34:29 MQT: stat/zbBridge/STATUS11 = {"StatusSTS":{"Time":"2020-11-20T06:34:29","Uptime":"0T00:22:53","UptimeSec":1373,"Vcc":3.475,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Streaming2G","BSSId":"FE:EC:DA:FA:1D:CA","Channel":12,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
```
06:13:51 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00158D00054D81A9","ShortAddr":"0x3A21","ParentNetwork":"0x0000","Status":1,"Decision":0}}
06:13:52 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00158D00054D81A9","ShortAddr":"0x3A21","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
06:13:52 MQT: tele/zbBridge/SENSOR = {"ZbReceived":{"0x3A21":{"Device":"0x3A21","ModelId":"lumi.sensor_switch.aq3","AppVersion":5,"Endpoint":1,"LinkQuality":110}}}
06:13:53 MQT: tele/zbBridge/SENSOR = {"ZbReceived":{"0x3A21":{"Device":"0x3A21","BatteryVoltage":3.04,"BatteryPercentage":100,"Xiaomi_98":1,"Xiaomi_99":0,"Endpoint":1,"LinkQuality":107}}}
06:13:53 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
06:13:54 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 270 bytes)
06:13:56 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
06:13:58 MQT: tele/zbBridge/SENSOR = {"ZbReceived":{"0x3A21":{"Device":"0x3A21","Manufacturer":"LUMI","ModelId":"lumi.sensor_swit","Endpoint":1,"LinkQuality":68}}}
06:13:58 MQT: tele/zbBridge/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x5F01","DeviceVersion":1,"InClusters":["0x0000","0x0012","0x0006","0x0001"],"OutClusters":["0x0000"]}}
06:14:00 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 268 bytes)
06:14:00 ZIG: auto-bind `ZbBind {"Device":"0x3A21","Endpoint":1,"Cluster":"0x0001"}`
06:14:02 ZIG: auto-bind `ZbBind {"Device":"0x3A21","Endpoint":1,"Cluster":"0x0006"}`
06:14:04 ZIG: auto-bind `ZbSend {"Device":"0x3A21","Config":{"BatteryVoltage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":0.1},"BatteryPercentage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":1}}}`
06:14:06 MQT: tele/zbBridge/RESULT = {"ZbBind":{"Device":"0x3A21","Status":0,"StatusMessage":"SUCCESS"}}
06:14:06 ZIG: auto-bind `ZbSend {"Device":"0x3A21","Config":{"Power":{"MinInterval":1,"MaxInterval":3600}}}``

```

### TO REPRODUCE
Flash a Sonoff bridge and bind a Aqara WXKG12LM

### EXPECTED BEHAVIOUR
WXKG12LM should be detected as it is.

### ADDITIONAL CONTEXT

**Solution:** 
Aqara Button is detected as tele/zbBridge/SENSOR = {"ZbReceived":{"0x3A21":{"Device":"0x3A21","Manufacturer":"LUMI","ModelId":"lumi.sensor_**swit**","Endpoint":1,"LinkQuality":68}}}
but code only refers to "ModelId":"lumi.sensor_**switch**"
(-> see the missing "ch" in "switch")

change/add in line 1707 in xdrv_23_zigbee_5_converters.ino
`} else if (modelId.startsWith(F("lumi.remote")) || modelId.startsWith(F("lumi.sensor_switch")) || modelId.startsWith(F("lumi.sensor_swit"))) {   // only for Aqara buttons WXKG11LM & WXKG12LM`

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-20:
Thanks. I will check this tomorrow.

toto79, 2020-11-21:
Thanks @s-hadinger ... i assume PR will be merged soon,  so i can close this issue.

PR #9938 (2020-11-21): Zigbee fix lumi.switch #9923
## Description:

**Related issue (if applicable):** fixes #9923

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The [... cut]
closed 2020-11-21 by toto79