arendst/Tasmota #9926: BLE scan universal (ESP32)

RaspiGame, 2020-11-20:
**Have you looked for this feature in other issues and in the docs?**  
Yes, in "ESP32 using build-in BLE" part
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
It's a new feature about pairing BLE as a server (e.g.of this switch https://a.aliexpress.com/_BS9eKN). But made it more universal and flexible due there are multiple BLE devices
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd like to have a part on webside (similar to mqtt, wifi... Sides) where to be able to put the ESP in scanning mode BLE. And then, at receive firts data, be able to split it in differents variable. All of it by UI.

And for a bit future, made the same but as client, to pair the ESP to other device (e.g. esp as switch client connected to other esp server)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
A simple alternative could be made this by commands. I saw you implemented these features for specifics device's (as I can understand in "ESP32 using build-in BLE" part). 
P.S. I don't know how much difficult is what I proposed. Thanks for your great job ;)
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-11-22:
Adding the switch seems possible, once somebody publishes the data it advertises.

Building a complex UI on an ESP32 is not entirely impossible, but IMHO not such a good idea. We already have some web-UI in some places of Tasmota and debugging/maintaining it is „pure fun“ 😁 .
Tasmota is MQTT-centric and the (limited) developing firepower will be concentrated on this target in the foreseeable future.

If I would want to have such a feature, I would definitely pass this job to another device in the network like a raspberry pi and do the complex things i.e. in Node-Red or whatever you like.

closed 2020-11-22 by ascillato2

RaspiGame, 2020-11-23:
I'm pending to receive the switch mentioned, once received, I'll publish it here.

My idea was made home lightning system entirely domotic, but for the light (high priority), have two ways to toggle it (BLE switch and wifi app). With switch BLE I pretend to have a direct communication device to device (bypassing router or zigbee port) and also have wifi support communication. For this option, I'll make my own hw with esp32.

Other idea, could be easy to implement the mobilephone Bluetooth signal level as a sensor?

Staars, 2020-11-23:
I have the slight feeling, that you try to overengineer the things.

I can only speak for myself, but in my setup with relatively many dependent service and devices, I never had a problem to access devices in my home network and I have a Homebridge setup.

But that does not mean, that it is not enjoying, to tinker around 👍 .

RaspiGame, 2020-11-23:
I'm guilty, sir! My brain sometimes run around and around... I work in quality dept., sorry. Haha

I thought make that to have a system more strong and secure. I'll keep working in it. 💪

RaspiGame, 2021-01-04:
Hello, I finally received the device!

I don't know how BLE works exactly, but I tried. Tell me if the following debug lines are suficient for implementation. Thank you again ;)

```
> nRF Connect, 2021-01-04
> YLKG (28:D1:27:20:52:54)
> I	23:25:35.297	[Server] Server started
> V	23:25:35.308	Unknown Service (00000100-0065-6c62-2e74-6f696d2e696d)
> V	23:25:35.680	Connecting to 28:D1:27:20:52:54...
> D	23:25:35.680	gatt = device.connectGatt(autoConnect = true, TRANSPORT_LE, preferred PHY = LE 1M)
> D	23:25:36.008	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
> D	23:25:36.047	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
> I	23:25:36.047	[Server] Device with address 28:D1:27:20:52:54 connected
> D	23:25:36.047	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
> I	23:25:36.047	Connected to 28:D1:27:20:52:54
> V	23:25:36.074	Discovering services...
> D	23:25:36.074	gatt.discoverServices()
> I	23:25:36.585	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 900ms)
> I	23:25:36.945	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
> D	23:25:37.276	[Callback] Services discovered with status: 0
> I	23:25:37.277	Services discovered
> V	23:25:37.339	Generic Attribute (0x1801)
> - Service Changed [I] (0x2A05)
>    Client Characteristic Configuration (0x2902)
> Generic Access (0x1800)
> - Device Name [R] (0x2A00)
> - Appearance [R] (0x2A01)
> - Peripheral Preferred Connection Parameters [R] (0x2A04)
> - Central Address Resolution [R] (0x2AA6)
> Battery Service (0x180F)
> - Battery Level [N R] (0x2A19)
>    Client Characteristic Configuration (0x2902)
> Device Information (0x180A)
> - Manufacturer Name String [R] (0x2A29)
> - Model Number String [R] (0x2A24)
> - Hardware Revision String [R] (0x2A27)
> - Firmware Revision String [R] (0x2A26)
> Unknown Service (0000fe95-0000-1000-8000-00805f9b34fb)
> - Unknown Characteristic [R] (00000004-0000-1000-8000-00805f9b34fb)
> - Unknown Characteristic [N WNR] (00000010-0000-1000-8000-00805f9b34fb)
>    Client Characteristic Configuration (0x2902)
> - Unknown Characteristic [N WNR] (00000019-0000-1000-8000-00805f9b34fb)
>    Client Characteristic Configuration (0x2902)
> - Unknown Characteristic [N W] (00000017-0000-1000-8000-00805f9b34fb)
>    Client Characteristic Configuration (0x2902)
> - Unknown Characteristic [N WNR] (00000018-0000-1000-8000-00805f9b34fb)
>    Client Characteristic Configuration (0x2902)
> Unknown Service (00000100-0065-6c62-2e74-6f696d2e696d)
> - Unknown Characteristic [WNR] (00000101-0065-6c62-2e74-6f696d2e696d)
> - Unknown Characteristic [N] (00000102-0065-6c62-2e74-6f696d2e696d)
>    Client Characteristic Configuration (0x2902)
> D	23:25:37.339	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.342	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.346	gatt.setCharacteristicNotification(00000010-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.350	gatt.setCharacteristicNotification(00000019-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.355	gatt.setCharacteristicNotification(00000017-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.358	gatt.setCharacteristicNotification(00000018-0000-1000-8000-00805f9b34fb, true)
> D	23:25:37.360	gatt.setCharacteristicNotification(00000102-0065-6c62-2e74-6f696d2e696d, true)
> I	23:25:37.363	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 900ms)
> V	23:25:45.052	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:25:45.052	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
> I	23:25:45.102	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
> A	23:25:45.102	"Notifications and indications disabled" received
> V	23:25:46.365	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:25:46.366	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
> I	23:25:46.421	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
> A	23:25:46.421	"Notifications and indications disabled" received
> V	23:25:46.600	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:25:46.600	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
[... 156 lines cut ...]
> A	23:26:45.733	"Notifications enabled" sent
> V	23:26:45.742	Notifications enabled for 00000017-0000-1000-8000-00805f9b34fb
> V	23:26:46.939	Enabling notifications for 00000018-0000-1000-8000-00805f9b34fb
> D	23:26:46.939	gatt.setCharacteristicNotification(00000018-0000-1000-8000-00805f9b34fb, true)
> D	23:26:46.943	gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0100)
> I	23:26:46.991	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
> A	23:26:46.991	"Notifications enabled" sent
> V	23:26:47.000	Notifications enabled for 00000018-0000-1000-8000-00805f9b34fb
> V	23:26:48.653	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:26:48.653	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
> I	23:26:48.735	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
> A	23:26:48.735	"Notifications enabled" received
> V	23:26:48.954	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:26:48.954	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
[... cut ...]
> D	23:26:49.438	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
> I	23:26:49.484	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
> A	23:26:49.484	"Notifications enabled" received
> V	23:26:49.904	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
> D	23:26:49.904	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
> I	23:26:49.965	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
> A	23:26:49.965	"Notifications enabled" received
```

RaspiGame, 2021-01-05:
I found this Home Assistant integration that already use this device (YLAI003). I hope it'll be useful :)

https://github.com/AlexxIT/XiaomiGateway3/blob/master/custom_components/xiaomi_gateway3/core/bluetooth.py

RaspiGame, 2021-02-23:
The guys of the custom component "Passive BLE Monitor integration" have implemented this device. 

Here is the pull request info and comments: https://github.com/custom-components/ble_monitor/issues/268

And here is the implementarion (for the custom component), i hope it could be a good help: 
https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/__init__.py
https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/sensor.py
https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/const.py

RaspiGame, 2021-06-15:
Hello, Tasmota improved a lot since I open that!! Ouyea!! 

Using [new BLE ESP32 commads](https://tasmota.github.io/docs/Commands/#ble-esp32) I've managed to do the connection and decryption of the BLE switch! Also I've maneged to sent it by mqtt. I just have tree questions:
[image]

> This is the debug response once ble switch is pressed:
>```
>20:10:51.216 M32: Decryption Key found
>20:10:51.223 M32: Err:1, Decrypted : 10 03 00 00 00
>...
>20:10:51.264 MQT: tele/tasmota_ble/S1 = {"Time":"2021-06-15T....
>```
> The important thing here is the **Decrypted** part, in this case: 10 03 00 00 **00**. And the values representing the pressed options are representated in the last bit as 0 = one press, 1 = double press and 2 = long press. That is what I've observed from Tasmota console (in Web Log Level: 3).
> - The thing is that Tasmota isn't using that bit to indicate the button state, because in the Tasmota Home Page and mqtt message show: 0 / 0 / 6 respectively. May it's configurable  by some command, but until now I not find the way to do it. Some suggestion?

```
single press
00:00:11.834 M32: Search key for MAC: 2**********4
00:00:11.839 M32: Decryption Key found
00:00:11.843 M32: Err:1, Decrypted : 10 03 00 00 00
00:00:11.844 M32 2**********4: Payload decrypted
00:00:11.846 M32: Frame 146, last 146
00:00:11.866 MQT: tele/tasmota_ble/S1 = {"Time":"2021-06-15T00:00:11","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPoint":null,"Btn":0,"Battery":84,"RSSI":-74}
00:00:11.875 M32: MI32: triggered 1 {"Time":"2021-06-15T00:00:11","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPo...

Double press
23:53:34.710 M32: Search key for MAC: 2**********4
23:53:34.711 M32: Decryption Key found
23:53:34.713 M32: Err:1, Decrypted : 10 03 00 00 01
23:53:34.714 M32 2**********4: Payload decrypted
23:53:34.715 M32: Frame 121, last 120
23:53:34.730 MQT: tele/tasmota_ble/S1 = {"Time":"2021-06-14T23:53:34","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPoint":null,"Btn":0,"Battery":84,"RSSI":-60}
23:53:34.737 M32: MI32: triggered 1 {"Time":"2021-06-14T23:53:34","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPo...

Long press
00:03:15.856 M32: Search key for MAC: 2**********4
00:03:15.858 M32: Decryption Key found
00:03:15.860 M32: Err:1, Decrypted : 10 03 00 00 02
00:03:15.862 M32 2**********4: Payload decrypted
00:03:15.864 M32: Frame 148, last 148
00:03:15.876 MQT: tele/tasmota_ble/S1 = {"Time":"2021-06-15T00:03:15","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPoint":null,"Btn":6,"Battery":84,"RSSI":-53}
00:03:15.885 M32: MI32: triggered 1 {"Time":"2021-06-15T00:03:15","alias":"S1","mac":"2**********4","Temperature":null,"Humidity":null,"DewPo...
```

>- Other  question, how can I use the button's value in Rule? Something like this `Rule1 ON S1#state=3 do Power1 2`?

>- And the last thing, but not that less important, seams that when you make a reset (by button in home page) all BLE configuration is deteled