arendst/Tasmota #9927: i2cscan and display doesn't work

panciunio, 2020-11-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
i2cscan doesn't work with precompiled tasmota-display.bin image. displayxxx commands doesn't work as well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____
  - [ x] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____ PlatformiO
- [ x] Flashing tools used: _____ esptool/PlatformiO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Generic","GPIO":[1,1,1,1,640,608,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:10:16 CMD: status 0
00:10:16 RSL: stat/tasmota_1B02CA/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1B02CA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:18","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"1970-01-01T00:00:00","SaveCount":31,"SaveAddress":"FB000"}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS2 = {"StatusFWR":{"Version":"9.1.0(display)","BuildDateTime":"2020-11-07T12:00:03","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Kameleon",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":28,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAAA59E","04049FA1","000000C3","00000000","00004080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS5 = {"StatusNET":{"Hostname":"DISPLAY3","IPAddress":"10.100.100.185","Gateway":"10.100.100.1","Subnetmask":"255.255.255.0","DNSServer":"85.11.68.99","Mac":"5C:CF:7F:1B:02:CA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1B02CA","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:10:16","Local":"1970-01-01T00:10:16","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"19:10","Sunset":"04:38"}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:10:16"}}
00:10:16 RSL: stat/tasmota_1B02CA/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:10:16","Uptime":"0T00:10:18","UptimeSec":618,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Kameleon","BSSId":"18:E8:29:B1:00:a2","Channel":11,"RSSI":22,"Signal":-89,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 Project tasmota Tasmota Version 9.1.0(display)-2_7_4_5
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Kameleon Channel 11 BSSId":"18:E8:29:B1:00:a2 in mode 11N as DISPLAY3...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:04 HTP: Web server active on DISPLAY3 with IP address 10.100.100.185
00:00:05 RSL: tele/tasmota_1B02CA/INFO1 = {"Module":"Generic","Version":"9.1.0(display)","FallbackTopic":"cmnd/DVES_1B02CA_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: tele/tasmota_1B02CA/INFO2 = {"WebServerMode":"Admin","Hostname":"DISPLAY3","IPAddress":"10.100.100.185"}
00:00:05 RSL: tele/tasmota_1B02CA/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 16
00:00:08 CFG: Saved to flash at F6, Count 36, Bytes 4096
00:00:09 RSL: tele/tasmota_1B02CA/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Kameleon2","BSSId":"18:E8:29:B1:00:a2","Channel":11,"RSSI":26,"Signal":-87,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:23 WIF: Checking connection...

00:04:06 CMD: i2cscan
00:04:06 SRC: WebConsole from 10.100.100.182
00:04:06 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
00:04:06 RSL: stat/tasmota_1B02CA/RESULT = {"Command":"Error"}

00:04:32 CMD: display
00:04:32 SRC: WebConsole from 10.100.100.182
00:04:32 CMD: Group 0, Index 1, Command "DISPLAY", Data ""
00:04:32 RSL: stat/tasmota_1B02CA/RESULT = {"Command":"Unknown"}

00:04:51 CMD: displayaddress
00:04:51 SRC: WebConsole from 10.100.100.182
00:04:51 CMD: Group 0, Index 1, Command "DISPLAYADDRESS", Data ""
00:04:51 RSL: stat/tasmota_1B02CA/RESULT = {"Command":"Unknown"}

00:05:09 CMD: i2cdriver
00:05:09 SRC: WebConsole from 10.100.100.182
00:05:09 CMD: Group 0, Index 1, Command "I2CDRIVER", Data ""
00:05:09 RSL: stat/tasmota_1B02CA/RESULT = {"I2CDriver":"3,4,5,6,47"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Jus upload last firmware...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

i2cscan, display should return proper value

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I checked 3 other ESP based devices with the same method. Tried to flash by using precompiled tasmota-display.bin and own compiled with additional options in  user_config_override.h:

#define USE_I2C
#define USE_DISPLAY
#define USE_DISPLAY_LCD
#define USE_DISPLAY_SSD1306
#define SHOW_SPLASH

Morover I tried use other displays with the same SSD1306 driver and still no proper output. Additional I checked drivers with my Cubieboard development envt. and displays working fine (base adress: 0x3c).

What I'm doing wrong? 

Regards,
pancio.net 

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-11-20:
You failed to provide all the output of `Backlog Template; Module; GPIO 255`
Especially the last, as your symptom would match your template not being activated.
FTR, I've got a SSD1306 running on Tasmota 9.1.0, no problems.

panciunio, 2020-11-20:
Thanks for your comment. I learnt more and added required details.

sfromis, 2020-11-20:
Thanks. That confirmed my expressed suspicion of that happened. You need to activate the template, otherwise no I2C bus is configured.

panciunio, 2020-11-20:
You have totally right.  I activated the template and now the I2C display was discovered... and now the display working fine :-)
I didn't realize that I'm not activated template earlier (I'm pretty sure that I did it), neverless the most important is that now it's working :-)
Thanks a lot @sfromis  for your effort and good advice!
 

```
00:00:00 CFG: Loaded from flash at F8, Count 42
00:00:00 QPC: Count 1
00:00:00 I2C: SSD1306 found at 0x3c
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 9.1.0(display)-2_7_4_5
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Kameleon2_MiWiFi2 Channel 11 BSSId 18:E8:29:A1:00:92 in mode 11N as DISPLAY3...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:04 HTP: Web server active on DISPLAY3 with IP address 10.100.100.185
00:00:05 RSL: tele/tasmota_1B02CA/INFO1 = {"Module":"Generic","Version":"9.1.0(display)","FallbackTopic":"cmnd/DVES_1B02CA_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: tele/tasmota_1B02CA/INFO2 = {"WebServerMode":"Admin","Hostname":"DISPLAY3","IPAddress":"10.100.100.185"}
00:00:05 RSL: tele/tasmota_1B02CA/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 RSL: stat/tasmota_1B02CA/RESULT = {"POWER":"OFF"}
00:00:05 RSL: stat/tasmota_1B02CA/POWER = OFF
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 21
00:00:08 CFG: Saved to flash at F7, Count 43, Bytes 4096
00:00:09 RSL: tele/tasmota_1B02CA/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Kameleon2","BSSId":"18:E8:29:f1:00:d2","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:23 WIF: Checking connection...
00:00:37 CMD: i2cscan
00:00:37 SRC: WebConsole from 10.100.100.182
00:00:37 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
00:00:37 RSL: stat/tasmota_1B02CA/RESULT = {"I2CScan":"Device(s) found at 0x3c"}
00:00:43 WIF: Checking connection...
00:00:44 CMD: display
00:00:44 SRC: WebConsole from 10.100.100.182
00:00:44 CMD: Group 0, Index 1, Command "DISPLAY", Data ""
00:00:44 RSL: stat/tasmota_1B02CA/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":1,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
00:01:03 WIF: Checking connection...
```

effelle, 2020-11-20:
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-20 by effelle

riker65, 2020-12-23:
Hi, how do you wire the display?

panciunio, 2020-12-23:
Hi,
You may find the wiring on the screenshot above.
For Wemos D1 MINI PRO:
I2C_SDA - GPIO4 - D2
I2C_SCL - GPIO5 - D1

Regards,
panciunio