arendst/Tasmota #9929: lights scheme 4 / random: device freezes after some time

maichai, 2020-11-20:
### PROBLEM DESCRIPTION

Using HASS Tasmota (beta) integration for lights to use the "random" color effect the lights work for quite a while and show the effect and are responsive. After quite a while they become unresponsive and need to be power cycled to "come back" as "available". That did not seem to happen with the previous release (I think 8.5.1).

This has been tested with 7 lights on 9.1.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [no discord account] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Moko GU10
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

On HASS UI set lights to use effect "random". Wait until the colors don't change any more.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Colors should change indefinitely, lights should remain available and responsive.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

HASS Core 0.118.0

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-11-21:
I'm not sure it is an issue with HaTasmota, since it just set the effect and ask Tasmota to reproduce it. 
Anyway, without a complete template is pretty impossible to know what is going on.

maichai, 2020-11-21:
Sorry that was as much as I could do. But since it is the Tasmota devices that become unresponsive it seems to me pretty clear that there is at least a problem with Tasmota, as this should not happen, even if HASS has a problem.

ascillato, 2020-11-21:
> ...But since it is the Tasmota devices that become unresponsive it seems to me pretty clear that there is at least a problem with Tasmota....

Please, give us more information to track this problem. Please, complete the template presented when you have opened this issue. As it is, I could not reproduce this, so more information is needed to troubleshoot it.

maichai, 2020-11-22:
There is not anything more to say to reproduce. It is really that simple as I wrote it.
The additional information I cannot produce. Sorry.

effelle, 2020-11-22:
> This has been tested with 7 lights on 9.1.0

If all the lights are from the same brand/model could be something related to the hardware type. Can you confirm if they are the same? 
If they are not based on TuyaMCU please share at least the result of `Backlog Template; Module; GPIO 255` to help us reduce the possible causes. 
As a side note, I left a WS1228b ring running on scheme 4 for more than 12 hours and I can't reproduce the issue.
The only think that comes in mind is check the Home Assistant logs at debug level to see if there is something about it, but I doubt yopu will find something related. 

Obs.: Even if the Tasmota official integration is made (also) by me is not tracked on this GitHub. Please open an issue [here](https://github.com/emontnemery/hatasmota).

ascillato, 2020-11-22:
> There is not anything more to say to reproduce. It is really that simple as I wrote it.
The additional information I cannot produce. Sorry.

I'm sorry, but the information requested is needed to troubleshoot your issue. If you don't want to provide it, we can't help you. The device freeze could be another problem not related to the HA integration. If you provide the status 0, we can see the restart reason, may be it is just hunging because of bot enough energy, but without info I'm just guessing. 

Please, provide the info requested in order to have something to work on. Thanks.

ascillato2, 2020-11-22:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-22 by ascillato2

emontnemery, 2020-11-27:
@maichai When you see this problem, is the light completely dead including not possible to access Tasmota web console?