arendst/Tasmota #9939: USE_PING not mentioned at `BUILDS.md` and not available at least at tasmota.bin

juliogonzalez, 2020-11-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Status of `USE_PING` is not documented at https://github.com/arendst/Tasmota/blob/development/BUILDS.md but it seems it's not available at least for `tasmota.bin` (maybe same case for other builds).

I think it should be documented, or at least the `BUILDS.md` should mention that if a feature is not mentioned at all, it is just not available for any of the builds.

Right now the doc mentions that `ping` *(requires #define USE_PING)* at https://tasmota.github.io/docs/Commands/ but having only that is (IMHO) not enough information to assume that `USE_PING` was not used for the pre-compiled binaries.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S2X
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: `tasmotizer.py`
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:20:40 CMD: Backlog Template; Module; GPIO 255
19:20:40 RSL: stat/tasmota_3E082C/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:20:40 RSL: stat/tasmota_3E082C/RESULT = {"Module":{"8":"Sonoff S2X"}}
19:20:40 RSL: stat/tasmota_3E082C/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
19:22:14 CMD: Backlog Rule1; Rule2; Rule3
19:22:14 RSL: stat/tasmota_3E082C/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":304,"Free":207,"Rules":"ON system#boot do Increase 3 ENDON   ON system#boot do IpPing 3 ENDON   ON Increase#State>1439 DO Increase 1439 ENDON    ON Time#Minute|%Increase% DO Ping4 %IpPing% ENDON   ON Ping#%IpPing%#Success==0 DO backlog Mult1 3; Power1 0; Delay 10; Power1 1 ENDON   ON Ping#%IpPing%#Success>0 DO Increase 3 ENDON"}
19:22:14 RSL: stat/tasmota_3E082C/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:22:14 RSL: stat/tasmota_3E082C/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
19:21:35 CMD: Status 0
19:21:35 RSL: stat/tasmota_3E082C/STATUS = {"Status":{"Module":8,"DeviceName":"Tasmota-Wifi-Madrid","FriendlyName":["Tasmota Wifi Madrid"],"Topic":"tasmota_3E082C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:11:29","StartupUTC":"2020-11-21T18:10:06","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-11-21T17:56:32","SaveCount":26,"SaveAddress":"FA000"}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"428/699"}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Republica","Tasmota"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_3E082C-2092","IPAddress":"192.168.3.94","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.7","Mac":"84:F3:EB:3E:08:2C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3E082C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS7 = {"StatusTIM":{"UTC":"2020-11-21T18:21:35","Local":"2020-11-21T19:21:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:09","Sunset":"17:02"}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS10 = {"StatusSNS":{"Time":"2020-11-21T19:21:35"}}
19:21:35 RSL: stat/tasmota_3E082C/STATUS11 = {"StatusSTS":{"Time":"2020-11-21T19:21:35","Uptime":"0T00:11:29","UptimeSec":689,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Republica","BSSId":"8C:E1:17:E9:09:BC","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
19:26:05 CMD: ping4 192.168.3.10
19:26:05 SRC: WebConsole from 192.168.3.107
19:26:05 CMD: Group 0, Index 4, Command "PING", Data "192.168.3.10"
19:26:05 RSL: stat/tasmota_3E082C/RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Install `tasmota.bin` on any supported device and try to run the `ping` command.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The status of `USE_PING` on the builds should be documented

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

I think they are not needed in this case.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

This was already reported at https://github.com/arendst/Tasmota/issues/9078, but since the user did not fill the template, the bot closed the issue.

If you tell me how you want to handle this, I think I can provide a PR :-)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-11-22:
> The status of USE_PING on the builds should be documented

Agree.

Please, help us and provide a PR to add that information. Thanks.

closed 2020-11-22 by ascillato2

juliogonzalez, 2020-11-22:
Will do but... why was this issue closed if I still need to prepare the PR?

According to https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md `Issues for unmerged PRs will be closed. If there is an issue with a PR, the explanation should be added to the PR itself.`

PR #9947 (2020-11-22): Document state of USE_PING at BUILDS.md
## Description:

As far I can see it's disabled for all builds, as according to `tasmota/my_user_config.h` it is `+2k code`.

**Related issue (if applicable):** fixes #9939 and fixes #9078
## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] [... cut]