arendst/Tasmota #9941: Tasmota Zigbee Sensor use State instead of Power

tyjtyj, 2020-11-22:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Not Really. But to maintain consistency. 
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Use state for sensor instead of Power
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
N/A
**Additional context**  
_Add any other context or screenshots about the feature request here._
Consider using state for zigbee sensor. Example below. 

Tasmota with Power monitoring report Power as in Watt while Tasmota Zigbee 
**Power Monitoring Plug** Power as in Watt
11:26:31 MQT: tele/bsd1/SENSOR = {"Time":"2020-11-22T11:26:31","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":125.335,"Yesterday":0.070,"Today":0.025,**"Power":14**,"ApparentPower":7,"ReactivePower":7,"Factor":0.32,"Voltage":235,"Current":0.031}}
**Zigbee Switch with power monitoring** Power as Switch State 
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1",**"Power":1**,"Endpoint":1,"LinkQuality":31}}
**Zigbee Sensor (Contact and Motion)** Power as in State
{"door_grill":{"Device":"0x1112","Name":"door_grill",**"Power":1**,"Endpoint":1,"LinkQuality":86}}

Suggestion
Use state as for Switch/Sensor state. This may cause HUGE breaking change
**Zigbee Switch with power monitoring** Power as Switch State 
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1",**"State":1**,"Endpoint":1,"LinkQuality":31}}
**Zigbee Sensor (Contact and Motion)** Power as in State
{"door_grill":{"Device":"0x1112","Name":"door_grill",**"State":1**,"Endpoint":1,"LinkQuality":86}}

In addition to power monitoring on zigbee plug. Also a bug
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1",**"AqaraRotate":0.42**,"Endpoint":2,"LinkQuality":18}}
AqaraRotate is actually Power in Watt
Suggestion:
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1","Power":0.42,"Endpoint":2,"LinkQuality":18}}
Auto reporting power
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1","Xiaomi_64":1,**"Xiaomi_98"**:0.93,"Xiaomi_95":31.03,"Xiaomi_9A":16,"Endpoint":1,"LinkQuality":45}}
Suggestion
{"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1","Xiaomi_64":1,**"Power"**:0.93,"EnergyTotal":31.03,"Xiaomi_9A":16,"Endpoint":1,"LinkQuality":45}}

Note:
Xiaomi_64: Power State
Xiaomi_98: Power Watt
Xiaomi_95: Energy Total
Xiaomi_9A: ??????????

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-22:
This is very unfortunate that the same `Power` label was used both for Relays and plug power. I generally try to stick to Zigbee names but I did some exceptions for the commonly used Tasmota attributes.

The ZCL equivalent to `Power` is OnOff, but I preferred to reuse Tasmota's `Power` instead.

I would argue that this consufion exists within Tasmota independently from Zigbee.

@arendst Shouldn't we rename the Energy reporting instead? Like `PowerWatt` or `TruePower`?

Let me support for Xiaomi Plug.

arendst, 2020-11-22:
If we need to change energy ``Power`` it would need to become ``ActivePower``

s-hadinger, 2020-11-22:
Indeed, but just changing `D_JSON_POWERUSAGE` from `Power` to `ActivePower` would certainly break some HA integration. Maybe it would be wiser to display both `Power` and `ActivePower` for a certain period until we deprecate `Power` for ENERGY.

s-hadinger, 2020-11-22:
It should be fixed now for the lumi.plug. Can you please confirm using the latest development version?

PR #9953 (2020-11-22): Zigbee support for lumi_plug
## Description:

Better support for lumi_plug. Also renamed `CurrentSummDelivered` to `EnergyTotal` 

**Related issue (if applicable):** fixes #9941

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tas [... cut]
closed 2020-11-22 by s-hadinger