arendst/Tasmota #9942: Xiaomi Sensor LYWSD03 (MI_ESP32). Negative temperature values (less than zero) not OK.

Einstein67, 2020-11-22:
### PROBLEM DESCRIPTION
Negative temperature values are shown as high positive values.
For example: "-1.6 ° C" on the sensor results in approximately "+6550.0 ° C" in Tasmota. 
The error occurs with stock firmware and custom firmware (ATC).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32-DevKit Modul
- [ ] Tasmota binary firmware version number used: Tested with 9.1.0.1, 9.1.0.2 (and MI ESP32 v0916a/v0917)
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: portable VSC/PlatformIO (Jason)
- [ ] Flashing tools used: _____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
07:11:40 CMD: Backlog Template; Module; GPIO 255
07:11:40 MQT: stat/esp32/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
07:11:40 MQT: stat/esp32/RESULT = {"Module":{"0":"ESP32-DevKit"}}
07:11:41 MQT: stat/esp32/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"480":"Buzzer"},"GPIO3":{"0":"None"},"GPIO4":{"1376":"WS2812"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"226":"Relay3"},"GPIO14":{"227":"Relay4"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"No ...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
07:14:11 CMD: status 0
07:14:11 MQT: stat/esp32/STATUS = {"Status":{"Module":0,"DeviceName":"Mi Temps und AlZet","FriendlyName":["Res 1","Res 2","Res 3","Alarm"],"Topic":"esp32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"LedMask":"0001","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:14:11 MQT: stat/esp32/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bix","RestartReason":"Software reset CPU","Uptime":"0T00:27:56","StartupUTC":"2020-11-22T05:46:15","Sleep":10,"CfgHolder":4617,"BootCount":118,"BCResetTime":"2020-04-21T17:34:41","SaveCount":1208}}
07:14:11 MQT: stat/esp32/STATUS2 = {"StatusFWR":{"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020.11.17 06:33:39","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"423/699"}}
07:14:11 MQT: stat/esp32/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Einstein-N450",""],"TelePeriod":60,"Resolution":"458180C0","SetOption":["00108009","2805C80001000600003C5A00000000000000","000004A0","00006000","00000002"]}}
07:14:11 MQT: stat/esp32/STATUS4 = {"StatusMEM":{"ProgramSize":1221,"Free":1600,"Heap":108,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000407","0F008683","00100001","00000007","000000C0","00000001","00000000","00000000"],"Drivers":"1,2,3,4,9,10,20,21,24","Sensors":"1,2,3,62"}}
07:14:11 MQT: stat/esp32/STATUS5 = {"StatusNET":{"Hostname":"197_Mi-Temps_Alzet","IPAddress":"192.168.0.197","Gateway":"192.168.0.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.2","Mac":"24:0A:C4:61:D7:D8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:14:11 MQT: stat/esp32/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.201","MqttPort":1883,"MqttClientMask":"Esp32","MqttClient":"Esp32","MqttUser":"Sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:14:11 MQT: stat/esp32/STATUS7 = {"StatusTIM":{"UTC":"2020-11-22T06:14:11","Local":"2020-11-22T07:14:11","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:23","Sunset":"16:22"}}
07:14:11 MQT: stat/esp32/STATUS10 = {"StatusSNS":{"Time":"2020-11-22T07:14:11","ATC-7ca255":{"Temperature":6549.7,"Humidity":65,"DewPoint":3795.1,"Battery":57,"RSSI":-88},"ATC-8a6769":{"Temperature":17.1,"Humidity":70,"DewPoint":11.6,"Battery":77,"RSSI":-69},"MJ_HT_V1-3ac46e":{"Temperature":16.3,"Humidity":75,"DewPoint":11.9,"Battery":89,"RSSI":-90},"TempUnit":"C"}}
07:14:11 MQT: stat/esp32/STATUS11 = {"StatusSTS":{"Time":"2020-11-22T07:14:11","Uptime":"0T00:27:56","UptimeSec":1676,"Heap":104,"SleepMode":"Normal","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Dimmer":63,"Color":"A19400","HSBColor":"55,100,63","Channel":[63,58,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Einstein-N450","BSSId":"50:C7:BF:D3:1A:3C","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Error occurs every time a negative temperature is measured

### EXPECTED BEHAVIOUR
Display of the temperature in Tasmota as on the display of the sensor

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-11-22:
Thanks for reporting.
I misinterpreted the manual of the device, stating temperature range of 0 - 60 C. Maybe the meant the range with reliable results?

Can you compile a test version, because I have slight problems to find a place with negative values of degrees celsius?

Einstein67, 2020-11-22:
Yes, I know the specifications 0 - (+ 60 °). But the (negative) values displayed on the sensor LCD, are OK compared to another thermometer.

I can compile a test version.

Thank you for your answer

Staars, 2020-11-22:
Here you are:
https://gist.github.com/Staars/79f396478ebb05ccd9097c7a23c5745b

Einstein67, 2020-11-22:
[image]

Seems to work :)

Thanks!!

Staars, 2020-11-22:
Nice.
I will prepare a PR.

closed 2020-11-22 by Einstein67