arendst/Tasmota #9945: Sonoff Zigbee sensors - Tasmota sends OFF signal even though sensor is not off

cydia2020, 2020-11-22:
### PROBLEM DESCRIPTION
When Zigbee2tasmota is used with SONOFF Zigbee binary sensors, notably their door sensor and motion sensor, tasmota will send `"Device":"0xXXXX","Name":"entryway_motion","Occupancy":0` before the device sends its own off command, which usually looks like `{"Device":"0xXXXX","Name":"entryway_motion","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":68}`

This creates problems as the sensor in actuality is not cleared when tasmota sends the clear command itself

The `ZbOccupancy 0xXXXX,0` command only sets the timeout to immediate, where tasmota will immediately send `"Occupancy":0` after a device is triggered, it does not disable tasmota from sending `"Occupancy":0`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
- [x] Flashing tools used: FTDI

- [x] Provide the output of this command: `Status 0`:
```
12:25:46 CMD: status 0
12:25:46 MQT: zbbridge/stat/STATUS = {"Status":{"Module":75,"DeviceName":"zbbridge","FriendlyName":["zbbridge"],"Topic":"zbbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:25:46 MQT: zbbridge/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"0T18:30:49","StartupUTC":"2020-11-21T16:54:57","Sleep":0,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-11-08T14:40:39","SaveCount":72,"SaveAddress":"F6000"}}
12:25:46 MQT: zbbridge/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(zbbridge)","BuildDateTime":"2020-11-07T12:01:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"450/699"}}
12:25:46 MQT: zbbridge/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OnHub-Cheng",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006080","00000000"]}}
12:25:46 MQT: zbbridge/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
12:25:46 MQT: zbbridge/stat/STATUS5 = {"StatusNET":{"Hostname":"SONOFF_ZIGBEE_BRIDGE","IPAddress":"192.168.85.23","Gateway":"192.168.85.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.85.1","Mac":"A4:CF:12:DC:01:30","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:25:46 MQT: zbbridge/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.85.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DC0130","MqttUser":"cydia2020","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:25:46 MQT: zbbridge/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-22T11:25:46","Local":"2020-11-22T12:25:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:01"}}
12:25:46 MQT: zbbridge/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-22T12:25:46"}}
12:25:46 MQT: zbbridge/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-22T12:25:46","Uptime":"0T18:30:49","UptimeSec":66649,"Vcc":3.442,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"somewifi","BSSId":"something","Channel":6,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
12:18:07 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","0500<00":"010000000000","ZoneStatusChange":1,"Occupancy":1,"Endpoint":1,"LinkQuality":79}}}
12:18:07 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","Occupancy":0}}}
12:18:13 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":100}}}
12:18:15 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","0500<00":"010000000000","ZoneStatusChange":1,"Occupancy":1,"Endpoint":1,"LinkQuality":79}}}
12:18:15 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","Occupancy":0}}}
12:18:17 MQT: zbbridge/tele/E6C3/SENSOR = {"ZbReceived":{"0xE6C3":{"Device":"0xE6C3","Name":"bedroom_door","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":100}}}
```

### TO REPRODUCE
described above

### EXPECTED BEHAVIOUR
described above, zboccupancy 0xXXXX,0 should disable tasmota from sending `"Occupancy":0`

s-hadinger, 2020-11-22:
This was fixed in #9787 
Please use the latest development version.

cydia2020, 2020-11-22:
Cheers. I'll give it a try when the next stable version comes out, I've came up with a temporary solution in the mean time and I'll stick with that till the next release. Closing for now.

closed 2020-11-22 by cydia2020