arendst/Tasmota #9952: MQTT issue in tasmota_ir

Ulli2k, 2020-11-22:
### PROBLEM DESCRIPTION
I am using the current tasmota_ir 9.1.0 binary with an IR Blaster (Blitzwolf BW-RC1) and it basically works like a charm. I can send and receive IR commands!

The issue is the following:
Always I send an IR command by using the MQTT interface I receive at the same time the same command.
If I use the console of the tasmota web interface it just sends the command without receiving it.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
09:55:43 MQT: stat/tasmota_DF8E60/RESULT = {"IRSend":"Done"}
09:55:43 MQT: tele/tasmota_DF8E60/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x5EA1D827","DataLSB":"0x7A851BE4","Repeat":0}}

```

### TO REPRODUCE
Every time with all IRsend commands by using MQTT.

### EXPECTED BEHAVIOUR
All IRsend commands can be sent without receiving it at the same time.
No echo! The receiver turns off during sending.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-22:
There is no difference whether you use the console or Mqtt. This is strange.

sfromis, 2020-11-22:
Dong a similar IRSend via MQTT, I'm not seeing the same signal coming back immediately. Wondering if it could depend on what repeat value you specify, I used 5. Sometimes, I do get spurious IrReceived payloads (UNKNOWN protocol), but that's just the noise I'm getting frequently anyway. No visible difference doing the same via console or MQTT. My hardware is a generic Aliexpress IR blaster, which has not given me any unexpected trouble.

Ulli2k, 2020-11-28:
@sfromis could you provide your extact mqtt command as reference?

sfromis, 2020-11-28:
An example payload published to the command topic of the blaster:
`xxxxxxxx/cmnd/IRSend {"Protocol":"NEC","Bits":32,"Data":"0x00FF08F7","Repeat":5}`

ascillato2, 2020-11-29:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-29 by ascillato2

Ulli2k, 2020-12-02:
Right know I found the real issue.
Each irsend command with repeat 0 results in receiving the same command at the same time.
Each command with >0 works as expected. (No echo)

Same behavoir using console or mqtt.

rebel7580, 2020-12-02:
I can confirm tha same behaviour with a YTF IR Bridge module running 8.4.0.1

s-hadinger, 2020-12-02:
In the latest dev version, receiving is disabled while enabling. There will not be any interference anymore.

rebel7580, 2020-12-02:
Time to upgrade, I guess. :)