arendst/Tasmota #9955: KNX Setting Several Tasmotas to be controlled by one Tasmota

DelroyTaylor, 2020-11-22:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have linked two KNX devices. One is the master and one slave. When power (120VAC) is applied to the Master the Slave turns ON immediately. When the Master is switched Off (by pressing the button) the slaves turns Off immediately.
The Master functions as a sensor and is connected to the furnace Aux fan (120VAC). The slave is in a remote room to control a fan.
The problem is when the Master goes OFF by losing electricity (unplugged/no electricity/dead) the slave stays on because the Master cannot send Off state (Command) to slaves. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Slave should monitor the state of the Master to ensure the Master is ON. If Master is not ON or responding the slave should turn itself OFF.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Configurations
Master:

{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

KNX Parameters:
Data to Send to Group Addresses
Output 1 -> 3 / 1 / 1
KNX TX 1 -> 3 / 1 / 2

Rule1
Rule1 on system#boot do backlog KnxTx_Cmnd1 1 endon
[image]

Slave:
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
KNX Parameters:
Group Addresses to Receive Data from
3 / 1 / 1 -> Output 1
3 / 1 / 2 -> KNX RX 1
Rule1
Rule1 on tele-power do knxtx_val1 %value% endon

[image]

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-22:
> Slave should monitor the state of the Master to ensure the Master is ON. If Master is not ON or responding the slave should turn itself OFF.

Hi,

There is no Online/Offline on KNX protocol, sorry. So, under strict standard KNX protocol this can not be done.

Anyway, you can use rules to add this new behavior and use KNX communications to regularly check for that MAIN device to be alive. If there is no answer in some threshold time, your SECONDARY devices can turn them off by themselves. 

One solution, using rules, could be that your MAIN device sends a heartbeat message (like a watchdog timer reset) to your SECONDARY devices. This message should reset an internal `ruletimer` of your SECONDARY devices. If that message is not received, in for example 30 seconds, their relays will turn off.

* On the MAIN device:

Let's set a GroupAddress to *send* this Heartbeat like KNXTX1 3.3.3 in the KNX webmenu.

```lua
On power1#boot do ruletimer1 30 endon
On rules#timer=1 do backlog knxtx_val1 1;ruletimer1 30 endon
```

* On each of the SECONDARY devices:

Let's set a GroupAddress to *receive* this Heartbeat like KNXRX1 3.3.3 in the KNX webmenu.

```lua
On power1#boot do ruletimer1 30 endon
On rules#timer=1 do power1 0 endon
On event#knxrx_val1 do ruletimer 1 30 endon
```

Hope this helps to achieve your desired behavior. If this is useful for you, please add it as a new example in the Tasmota's docs at KNX.

closed 2020-11-22 by ascillato2

DelroyTaylor, 2020-11-22:
Thank you for your excellent advice.

DelroyTaylor, 2020-12-02:
### 7) Rules to Monitor Heartbeat of Main Tasmota controlling several Secondary Tasmota. ###
There is no Online/Offline on KNX protocol. Rules can be used to add this new behavior. KNX communications is used to regularly check for that MAIN device to be alive. If there is no answer in some threshold time, your SECONDARY devices can turn themselves off.
With this solution the MAIN device sends its relay state as a heartbeat message (like a watchdog timer reset) to your SECONDARY devices. This message should reset an internal ruletimer of your SECONDARY devices. If that message is not received, for example 30 seconds, their relays will turn off.

On the MAIN device set a GroupAddress to send this Heartbeat like KNXTX1 3.3.3 in the KNX webmenu.
 
RULES
```
Rule1 ON system#boot DO BACKLOG KnxTx_Cmnd2 1; var1 1; ruletimer1 10 ENDON
```
At system boot (120Vac power applied) the MAIN device turns on its relay. It also sends KNX commands to turn on SECONDARY devices, it stores the value 1 in var1 and starts the ruletimer1 10 seconds countdown(heartbeat).
```
Rule1 + ON power1#State DO var1 %value% ENDON
```
Whenever MAIN power state changes (by pushbutton, Web UI, timer or MQTT) the power1#State value is stored in var1 (If MAIN loses Ac power no heartbeat is sent).
```
Rule1 + ON rules#timer=1 DO BACKLOG knxtx_val1 %var1%;ruletimer1 10 ENDON
```
When the ruletimer1 countdown reaches 1 KNX transmits var1 and restart ruletimer1 at 10 seconds. The cycle repeats.
```
RULE1 1
```
OUTPUT 1 -> 4.4.4
Output1 of MAIN is sent to 4.4.4. When received by SECONDARY, the SECONDARY matches the state with MAIN.

* In the KNX Menu, set a Group Address to send data or commands by rules, as **KNX TX2**

KNXTX2 -> 4.4.4.
At boot time KnxTx_Cmnd2 1 is sent from MAIN to 4.4.4. When received by SECONDARY, the SECONDARY turns its relay on.   

MAIN

[image]

SECONDARY

[image]

On each of the SECONDARY devices set a Group Address to receive this Heartbeat like KNXRX1 3.3.3 in the KNX webmenu.

```
RULE1 ON power1#boot DO ruletimer1 20 ENDON
```
When SECONDARY boots, the relay remains off until commanded by MAIN. The SECONDARY starts the ruletimer1 countdown at 20 seconds.
```
RULE1 + ON rules#timer=1 DO BACKLOG power1 0; ruletimer1 20 ENDON
```
Whenever ruletimer1 reaches 1 second it turns the relay OFF and restarts the ruletimer1 countdown at 20 seconds.
```
RULE1 + ON event#knxrx_val1 DO BACKLOG power1 %value%; ruletimer1 20 ENDON
```
If the SECONDARY receives a heartbeat transmission from MAIN it stores the value as the state for the relay. These values have following effect:
* 0 / OFF = relay(s) OFF
* 1 / ON = turn relay(s) ON
* 2 / TOGGLE = toggle relay(s)

```
RULE1 1
```

4.4.4 -> OUTPUT 1 
Output1 of MAIN is sent to 4.4.4. When received by SECONDARY, the SECONDARY matches the state with MAIN.