arendst/Tasmota #9957: adcparam Settings - lost after reboot

FanaticJoker, 2020-11-23:
### PROBLEM DESCRIPTION
Working with: 9.1.0(sensors)
After the AdcParam Command - everthing works fine. But after a reboot, the settings get lost.
`AdcParam 6, 660, 1024, 100, 0`

I haven't found any information on "saving" the settings for a reboot

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: 9.1.0 (sensors)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:31:32 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:31:32 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"Module":{"0":"Generic"}}
08:31:33 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"GPIO0":{"224":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4704":"ADC Input"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
08:32:00 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:32:00 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:32:00 MQT: sonoff/stat/sonoff-pflanze/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

  Console output here:
08:32:24 CMD: Status 0
08:32:24 MQT: sonoff/stat/sonoff-pflanze/STATUS = {"Status":{"Module":0,"DeviceName":"Sonoff-Pflanze","FriendlyName":["Sonoff-Pflanze"],"Topic":"sonoff-pflanze","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:32:24 MQT: sonoff/stat/sonoff-pflanze/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:31","StartupUTC":"2020-11-23T07:28:53","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2020-11-17T14:05:20","SaveCount":83,"SaveAddress":"F9000"}}
08:32:24 MQT: sonoff/stat/sonoff-pflanze/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"441/699"}}
08:32:24 MQT: sonoff/stat/sonoff-pflanze/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Benetwork",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS5 = {"StatusNET":{"Hostname":"sonoff-pflanze-7699","IPAddress":"192.168.2.118","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.202","Mac":"A4:CF:12:D9:BE:13","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.201","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D9BE13","MqttUser":"iobroker","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS7 = {"StatusTIM":{"UTC":"2020-11-23T07:32:25","Local":"2020-11-23T08:32:25","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:12","Sunset":"17:00"}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS10 = {"StatusSNS":{"Time":"2020-11-23T08:32:25","ANALOG":{"Range":40}}}
08:32:25 MQT: sonoff/stat/sonoff-pflanze/STATUS11 = {"StatusSTS":{"Time":"2020-11-23T08:32:25","Uptime":"0T00:03:32","UptimeSec":212,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Benetwork","BSSId":"FC:EC:DA:8A:2A:22","Channel":11,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```

### TO REPRODUCE
_Steps to reproduce the behavior: reboot the device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen: Settings get saved after reboot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

[image]

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-11-23:
To configure the ADC parameters for sensor 6 (range), you should also configure the ADC gpio17 for range. If you do that, the values will be saved.

FanaticJoker, 2020-11-23:
> To configure the ADC parameters for sensor 6 (range), you should also configure the ADC gpio17 for range. If you do that, the values will be saved.

Thank you very much... will check that as soon as I'm at home!

FanaticJoker, 2020-11-24:
> To configure the ADC parameters for sensor 6 (range), you should also configure the ADC gpio17 for range. If you do that, the values will be saved.

thank you very much, did the trick!

closed 2020-11-24 by FanaticJoker