arendst/Tasmota #9961: Button push toggle is often not carried out with relay

pirx7, 2020-11-23:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
A button push toggle is often not carried out with corresponding relay although it is recognized

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ ESP01s
- [x] Tasmota binary firmware version number used: _____8.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ______ tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
16:30:35 CMD: Backlog Template; Module; GPIO 255
16:30:35 SRC: WebConsole from 192.168.7.21
16:30:35 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
16:30:36 SRC: Backlog
16:30:36 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
16:30:36 RSL: stat/tasmota_833883/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:30:36 SRC: Backlog
16:30:36 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
16:30:36 RSL: stat/tasmota_833883/RESULT = {"Module":{"18":"Generic"}}
16:30:36 SRC: Backlog
16:30:36 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
16:30:36 RSL: stat/tasmota_833883/RESULT = {"GPIO0":{"29":"Relay1i"},"GPIO1":{"122":"Button1i"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
16:30:41 WIF: Prüfe Verbindung...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:32:10 CMD: Status 0
16:32:10 SRC: WebConsole from 192.168.7.21
16:32:10 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
16:32:10 RSL: stat/tasmota_833883/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_833883","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:32:10 RSL: stat/tasmota_833883/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:32:12","StartupUTC":"2020-11-23T14:59:58","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2020-11-22T11:17:37","SaveCount":593,"SaveAddress":"FB000"}}
16:32:10 RSL: stat/tasmota_833883/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020.09.09 11:47:14","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
16:32:10 RSL: stat/tasmota_833883/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FGhouse2g4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:32:10 RSL: stat/tasmota_833883/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
16:32:10 RSL: stat/tasmota_833883/STATUS5 = {"StatusNET":{"Hostname":"tasmota_833883-6275","IPAddress":"192.168.7.98","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.1","Mac":"84:CC:A8:83:38:83","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:32:10 RSL: stat/tasmota_833883/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_833883","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:32:10 RSL: stat/tasmota_833883/STATUS7 = {"StatusTIM":{"UTC":"2020-11-23T15:32:10","Local":"2020-11-23T16:32:10","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:12","Sunset":"17:00"}}
16:32:10 RSL: stat/tasmota_833883/STATUS10 = {"StatusSNS":{"Time":"2020-11-23T16:32:10"}}
16:32:10 RSL: stat/tasmota_833883/STATUS11 = {"StatusSTS":{"Time":"2020-11-23T16:32:10","Uptime":"0T00:32:12","UptimeSec":1932,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FGhouse2g4","BSSId":"7C:FF:4D:D5:AE:E1","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:32:21 WIF: Prüfe Verbindung...

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Serie 1
---------------------------------------------------------------------
13:13:26 WIF: Prüfe Verbindung...
[... 194 lines cut ...]
Tried a thousand things, nothing worked.
First of all, of course, I fell into the "[Button in ON state when depressed](https://tasmota.github.io/docs/FAQ/#device-reset-to-defaults-on-its-own))" trap.
I had wrongly expected that with an inverted input "Button1i" this reset behavior would also be inverted - which would make a lot more sense for me, but is not so.

**Fall Back**
So the shutter project is finished for now and replaced by a very primitive button relay setting, see below.

But here, too, I get push-button inconsistencies:

**Current problem**
A button push toggle is often not carried out with corresponding relay although it is recognized
There are also series in which no errors occur up to 50 times and more

**Set up**

ESP01s with a dual relay module
both have their own stable power supply
PullDown Push-Button + 10 nF on GPIO1
10 k PullUp resistors on all 4 GPIOs

Tasmota version 8.5.0 (tasmota)
Build Date & Time 2020/09/09 11:47:14 AM
Core / SDK version 2_7_4_1 / 2.2.2-dev (38a443e)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-11-29:
Please, update to latest Tasmota and try again. If you need assistance, please address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-29 by ascillato2

pirx7, 2020-11-29:
I changed to ESPHome. It is working directly with the same hardware.