arendst/Tasmota #9966: 9.1 Sonoff B1 flickering

BugsonPL, 2020-11-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After upgrading to 9.1.0 bulb started to flicker. Best to notice during fading or fast changing of color/brightenss.
Looks like communication issue with MY92x1.
After downgrading to 8.5.1 everything back to normal.
Tried:
Dev release 9.1.0.2 the same problem.
With MQTT disabled same problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____Sonoff B1 v2
- [x] Tasmota binary firmware version number used: _____tasmota.bin	9.1.0	  20201107 13:58
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:31:29 MQT: Sonoff_B1/stat/RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
09:31:30 MQT: Sonoff_B1/stat/RESULT = {"Module":{"26":"Sonoff B1"}}
09:31:30 MQT: Sonoff_B1/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"143":"MY92x1 DI"},"GPIO13":{"0":"None"},"GPIO14":{"144":"MY92x1 DCKI"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
09:31:36 MQT: Sonoff_B1/tele/STATE = {"Time":"2020-11-24T09:31:36","Uptime":"0T10:35:10","UptimeSec":38110,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,0,0","White":0,"CT":500,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Removed","BSSId":"50:C7:BF:AF:B8:84","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:00:59"}}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just flash with 9.1.0 and set:
Fade 1
Speed 3
Turn on/off the bulb few times with brithness 100% color to WW .

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Remove the bug

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-11-24:
Sorry, I cannot reproduce on my Sonoff B1 v1

BugsonPL, 2020-11-24:
I can do some more test if needed.
If I move slider in WEB interface from CW<->WW suddenly it will switch to some random white state and stay there until next 
color adjustment.
Can it be V2-V1 difference?
The problem is consistent and it is only on 9.1.0.
Every time I do Reset 5

radionerd1, 2020-12-12:
I have same problem with a B1 bulb running ESPEasy. Started several weeks ago. Just one bulb of two. Polly not firmware bug.

closed 2020-12-19 by github-actions[bot]

rg2k, 2021-01-06:
I am having the same issue. I have B1 v2's - so this issue seems to be related to v2's only. Mine starts fading, then flashes bright for a moment, then continues the fade. This is for both fade in and out.