arendst/Tasmota #9973: Avoid the RfReceived to send duplicated code

byte4geek, 2020-11-25:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I have PIR sensor that send the code 4 times sequentially, the tasmota RfReceived public all of 4 code on MQTT sequentially.
Is it possible to ignore the same code (duplicated) in a range of time of the 0,5 second circa and public only the first one?

**Describe the solution you'd like**  
_no solution for this_

**Describe alternatives you've considered**  
_OpenMQTTGateway already do this_

**Additional context**  
_noting_

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-25:
Hi 

Why is this a problem for you?

Adding this feature will break backward compatibility with some users setups. For example the key buttons. If you press it once, they send just once, but if you keep it pressed it sends the same code over and over and that can be catch as a hold action. If this code duplication is added, the hold feature will be disabled or not working for some users.

Jason2866, 2020-11-25:
Agree with @ascillato  For example i use this for on / off switching and holding the on / off button for dimming up / down.
Thank you for sharing your idea. Since it breaks compability it will not be implemented.

closed 2020-11-25 by ascillato2