arendst/Tasmota #9977: MQTT Full Payload option

shanekuz, 2020-11-25:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
When the ZBGateway sends TELE MQTT messages it sends them as it receives them from the Zigbee devices with just that piece of data.  When this is picked up by OPENHAB there are many things that monitor that MQTT TOPIC using JSONPATH and as there is only one or two pieces of data OPENHAB logs are full of errors as all of the other things get NULL values passed as they were not in the JSON payload. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like an OPTION to enable full payload of data to be sent when the STAT is updated for anyitem, all of these variables are kept in Tasmota so to publish them all to that TOPIC in one JSON should not be to hard i would think.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I think it could be done with many many rules to publish to different MQTT paths but this solution seems easier.

**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-25:
I'm not in favor of implementing this because:
1/ it's not the way Zigbee works
2/ Tasmota does not memorize all data on all devices due to limited memory
3/ it would create nasty side effects 
4/ this should be handled at gateway level since gateway have much more memory and computing power. 

Anyways you can always add a rule to trigger `ZbInfo %zbdevice%` after receiving a ZbReceived event. It's maybe enough for your needs.

ascillato, 2020-11-25:
I agree with @s-hadinger. The RAM is very small. Adding this will make that new features can't be added. Sorry.

Please, check with OpenHab support because I'm aware that there is a solution for that. I don't use OpenHab to tell you exactly the solution but from an old issue I recall that there is. Thanks.

shanekuz, 2020-11-25:
Thank you for your feedback can you please give more information on ZBinfo rules?  i have many devices on my gateway hence the issue will my logs getting full of NULLs would i have to have a rule for each device etc.  im just not sure what you mean

closed 2020-11-25 by ascillato2