arendst/Tasmota #9978: Topic with wrong payload for ZigBee temperature and humidity sensors

Joseraj, 2020-11-25:
### PROBLEM DESCRIPTION
Looking values of this sensors at influx I've found strange values for temperature and humidity.
At first I thank there was some problem extracting the values from json.
Finally I've seen live the msg with the wrong data.

This is the data from the broker:
```
24/11/2020, 21:47:18	tele/Outdoor_TempHumPres/SENSOR
"{"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":7.5,"Humidity":69.4,"Endpoint":1,"LinkQuality":55}}}"
24/11/2020, 21:47:18	tele/Outdoor_TempHumPres/SENSOR
"{"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":42949572,"Pressure":1023,"Humidity":69.4,"Reachable":true,"BatteryPercentage":24,"LastSeen":0,"LastSeenEpoch":1606250837,"LinkQuality":58}}}"
24/11/2020, 21:47:18	stat/ZbBridge/RESULT
"{"ZbInfo":"Done"}"
24/11/2020, 21:47:19	tele/Outdoor_TempHumPres/SENSOR
"{"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":-100,"Humidity":654.36,"Pressure":1017,"PressureScale":-1,"PressureScaledValue":10178,"SeaPressure":1023,"Endpoint":1,"LinkQuality":55}}}"
24/11/2020, 21:47:19	tele/Outdoor_TempHumPres/SENSOR
"{"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":42949572,"Pressure":1023,"Humidity":654.36,"Reachable":true,"BatteryPercentage":24,"LastSeen":1,"LastSeenEpoch":1606250837,"LinkQuality":55}}}"
24/11/2020, 21:47:19	stat/ZbBridge/RESULT
"{"ZbInfo":"Done"}"
```
This is the output in the console:
```
21:47:28 ZIG: {"ZbEZSPReceived":"45000004010204010100010000228CBFB0F2FFFF0818030A000029ED02"}
21:47:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1026,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":58,"securityuse":0,"seqnumber":34,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x0A","payload":"000029ED02"}}
21:47:28 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0402/0000":749,"Endpoint":1,"LinkQuality":58}}
21:47:28 ZIG: {"ZbEZSPReceived":"45000004010504010100010000238CBFB0F2FFFF0818040A000021181B"}
21:47:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1029,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":58,"securityuse":0,"seqnumber":35,"fc":"0x18","manuf":"0x0000","transact":4,"cmdid":"0x0A","payload":"000021181B"}}
21:47:28 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0405/0000":6936,"Endpoint":1,"LinkQuality":58}}
21:47:28 ZIG: {"ZbEZSPReceived":"450000040103040101000100002488BEB0F2FFFF1118050A000029FA03140028FF100029C527"}
21:47:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1027,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":55,"securityuse":0,"seqnumber":36,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x0A","payload":"000029FA03140028FF100029C527"}}
21:47:28 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0403/0000":1018,"0403/0014":-1,"0403/0010":10181,"Endpoint":1,"LinkQuality":55}}
21:47:28 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":7.49,"Humidity":69.36,"Pressure":1018,"PressureScale":-1,"PressureScaledValue":10181,"SeaPressure":1024,"Endpoint":1,"LinkQuality":55}}}
21:47:28 RUL: ZBRECEIVED#? performs "zbinfo 0xF2B0"
21:47:28 SRC: Rule
21:47:28 CMD: Group 0, Index 1, Command "ZBINFO", Data "0xF2B0"
21:47:28 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":7.49,"Pressure":1024,"Humidity":69.36,"Reachable":true,"BatteryPercentage":24,"LastSeen":0,"LastSeenEpoch":1606250848,"LinkQuality":55}}}
21:47:28 MQT: stat/ZbBridge/RESULT = {"ZbInfo":"Done"}
```

In the console I see 2 msg on topic "tele/Outdoor_TempHumPres/SENSOR" with right values: "Temperature":7.49,"Humidity":69.36
The broker recevie 4 msg, first 2 with the right values, but the other 2 have wrong values: "Temperature":-100,"Humidity":654.36
The 2 msg with the wrong data are not shown in the console.

(the ZbInfo are from a rule from [https://github.com/arendst/Tasmota/issues/9788#issuecomment-726745816](url) )

This logs are from a Aqara WSDCGQ11LM, but I've similar data from Mijia WSDCGQ01LM.

It doesn't allways happen, i see wrong values from the Mijia 1 or 2 times every 24h, more on the Aquara.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff ZBBridge 
- [x] Tasmota binary firmware version number used: 
Program Version | 9.1.0.2(zbbridge)
Build Date & Time | 2020-11-22T16:58:19
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:34:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:34:00 SRC: Backlog
20:34:00 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:34:00 MQT: stat/ZbBridge/RESULT = {"NAME":"Sonoff ZbBridge","GPIO":[320,3552,0,3584,5312,0,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":75}
[... 22 lines cut ...]
20:36:32 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:36:32 MQT: stat/ZbBridge/STATUS = {"Status":{"Module":75,"DeviceName":"ZbBridge","FriendlyName":["ZbBridge"],"Topic":"ZbBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:36:32 MQT: stat/ZbBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin","RestartReason":"Software/System restart","Uptime":"0T23:28:10","StartupUTC":"2020-11-24T20:08:22","Sleep":50,"CfgHolder":4617,"BootCount":325,"BCResetTime":"2020-08-07T22:54:08","SaveCount":717,"SaveAddress":"FB000"}}
20:36:32 MQT: stat/ZbBridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(zbbridge)","BuildDateTime":"2020-11-22T16:58:19","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"403/699"}}
20:36:32 MQT: stat/ZbBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pepe",""],"TelePeriod":300,"Resolution":"958180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00008000","40006082","00000000"]}}
20:36:32 MQT: stat/ZbBridge/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":424,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
20:36:32 MQT: stat/ZbBridge/STATUS5 = {"StatusNET":{"Hostname":"ZbBridge-2578","IPAddress":"192.168.2.83","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"C8:2B:96:06:4A:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:36:32 MQT: stat/ZbBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.63","MqttPort":1883,"MqttClientMask":"ZbBridge","MqttClient":"ZbBridge","MqttUser":"openhabian","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:36:32 MQT: stat/ZbBridge/STATUS7 = {"StatusTIM":{"UTC":"2020-11-25T19:36:32","Local":"2020-11-25T20:36:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:26","Sunset":"16:33"}}
20:36:32 MQT: stat/ZbBridge/STATUS10 = {"StatusSNS":{"Time":"2020-11-25T20:36:32"}}
[... cut ...] ":"0T00:02:28"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Hard to catch as it doesnt happen everytime.
Just log the data from the sensors.

### EXPECTED BEHAVIOUR
The data received from the sensor seems right, and the first mqtt msg is sent with that data, but there shoudn't be a second mqtt because nothing new has been received.

s-hadinger, 2020-11-25:
This is weird and there seem to be 2 different bugs.
First two lines show that ZbInfo does not report the same values as ZbReceived.
The next 2 lines of logs show an anomaly in decoding Temp and Humidity...

Too bad you couldn't capture the raw logs at `21:47:18`. I will have a look tomorrow.

s-hadinger, 2020-11-26:
I reviewed more than 2 weeks of logs and I've never seen such readings with lumi.weather. I really need the low-level logs (weblog 3). Can you enable MQTT logs or syslog?

Joseraj, 2020-11-26:
I'll enable it tonight and try to force readings with the button on the sensor.

Joseraj, 2020-11-26:
Hi,
I've got some logs:
```
19:52:25 ZIG: {"ZbEZSPReceived":"45000004010204010100010000A59CC3B0F2FFFF0818050A0000297C09"}
19:52:25 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1026,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":68,"securityuse":0,"seqnumber":165,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x0A","payload":"0000297C09"}}
19:52:25 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0402/0000":2428,"Endpoint":1,"LinkQuality":68}}
19:52:25 ZIG: {"ZbEZSPReceived":"45000004010504010100010000A69CC3B0F2FFFF0818060A000021DF16"}
19:52:25 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1029,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":68,"securityuse":0,"seqnumber":166,"fc":"0x18","manuf":"0x0000","transact":6,"cmdid":"0x0A","payload":"000021DF16"}}
19:52:25 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0405/0000":5855,"Endpoint":1,"LinkQuality":68}}
19:52:25 ZIG: {"ZbEZSPReceived":"45000004010204010100010000CAA0C4B0F2FFFF0818000A000029F0D8"}
19:52:25 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1026,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":202,"fc":"0x18","manuf":"0x0000","transact":0,"cmdid":"0x0A","payload":"000029F0D8"}}
19:52:25 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0402/0000":-10000,"Endpoint":1,"LinkQuality":71}}
19:52:25 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":24.28,"Humidity":58.55,"Endpoint":1,"LinkQuality":68}}}
19:52:25 RUL: ZBRECEIVED#? performs "zbinfo 0xF2B0"
19:52:25 SRC: Rule
19:52:25 CMD: Group 0, Index 1, Command "ZBINFO", Data "0xF2B0"
19:52:25 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":42949572,"Pressure":1022,"Humidity":58.55,"Reachable":true,"BatteryPercentage":24,"LastSeen":0,"LastSeenEpoch":1606416745,"LinkQuality":71}}}
19:52:25 MQT: stat/ZbBridge/RESULT = {"ZbInfo":"Done"}
19:52:25 ZIG: {"ZbEZSPReceived":"45000004010504010100010000CBA0C4B0F2FFFF0818010A0000219CFF"}
19:52:25 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1029,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":203,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x0A","payload":"0000219CFF"}}
19:52:25 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0405/0000":65436,"Endpoint":1,"LinkQuality":71}}
19:52:25 ZIG: {"ZbEZSPReceived":"45000004010304010100010000CC98C2B0F2FFFF1118020A000029F803140028FF100029B027"}
19:52:25 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1027,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":65,"securityuse":0,"seqnumber":204,"fc":"0x18","manuf":"0x0000","transact":2,"cmdid":"0x0A","payload":"000029F803140028FF100029B027"}}
19:52:25 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0403/0000":1016,"0403/0014":-1,"0403/0010":10160,"Endpoint":1,"LinkQuality":65}}
19:52:25 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":-100,"Humidity":654.36,"Pressure":1016,"PressureScale":-1,"PressureScaledValue":10160,"SeaPressure":1022,"Endpoint":1,"LinkQuality":65}}}
19:52:25 RUL: ZBRECEIVED#? performs "zbinfo 0xF2B0"
19:52:25 SRC: Rule
19:52:25 CMD: Group 0, Index 1, Command "ZBINFO", Data "0xF2B0"
19:52:25 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":42949572,"Pressure":1022,"Humidity":654.36,"Reachable":true,"BatteryPercentage":24,"LastSeen":0,"LastSeenEpoch":1606416745,"LinkQuality":65}}}
19:52:25 MQT: stat/ZbBridge/RESULT = {"ZbInfo":"Done"}
```
and
```
19:54:18 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0405/0000":65436,"Endpoint":1,"LinkQuality":86}}
19:54:18 ZIG: {"ZbEZSPReceived":"4500000401030401010001000056B4C9B0F2FFFF1118020A000029F803140028FF100029B027"}
19:54:18 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1027,"srcaddr":"0xF2B0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":84,"securityuse":0,"seqnumber":86,"fc":"0x18","manuf":"0x0000","transact":2,"cmdid":"0x0A","payload":"000029F803140028FF100029B027"}}
19:54:18 ZIG: ZbZCLRawReceived: {"0xF2B0":{"0403/0000":1016,"0403/0014":-1,"0403/0010":10160,"Endpoint":1,"LinkQuality":84}}
19:54:18 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbReceived":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","Temperature":-100,"Humidity":654.36,"Pressure":1016,"PressureScale":-1,"PressureScaledValue":10160,"SeaPressure":1022,"Endpoint":1,"LinkQuality":84}}}
19:54:18 RUL: ZBRECEIVED#? performs "zbinfo 0xF2B0"
19:54:18 SRC: Rule
19:54:18 CMD: Group 0, Index 1, Command "ZBINFO", Data "0xF2B0"
19:54:18 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":42949572,"Pressure":1022,"Humidity":654.36,"Reachable":true,"BatteryPercentage":24,"LastSeen":0,"LastSeenEpoch":1606416858,"LinkQuality":84}}}
19:54:18 MQT: stat/ZbBridge/RESULT = {"ZbInfo":"Done"}
[... cut ...]
19:54:29 MQT: tele/Outdoor_TempHumPres/SENSOR = {"ZbInfo":{"Outdoor_TempHumPres":{"Device":"0xF2B0","Name":"Outdoor_TempHumPres","IEEEAddr":"0x00158D0002786760","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":23.67,"Pressure":1022,"Humidity":57.91,"Reachable":true,"BatteryPercentage":24,"LastSeen":1,"LastSeenEpoch":1606416868,"LinkQuality":92}}}
19:54:29 MQT: stat/ZbBridge/RESULT = {"ZbInfo":"Done"}
```

Today i not only receive temperature "-100", but also "42949572"
I think its the sensor sending wrong data, I see values -10000, and 65436 on msg ZbZCLRawReceived.
Maybe because of low battery ?   (BatteryPercentage":24)

s-hadinger, 2020-11-27:
@Joseraj Thanks, with your logs I found a bug in ZbInfo that would report wrong values for negative numbers.

For lumi.weather, unfortunately the device really sends a temperature of -100.00°C. This is unfortunate that the developers didn't use an 'invalid value' but instead send a perfectly possible temperature.

However it is unlikely that the temperature is -100°C so I'm rejecting now any temperature below -99°C.

PR #9999 (2020-11-27): Zigbee fix invalid temperature #9978
## Description:

Fixes a bug in ZbInfo when reporting negative values.
Exclude temperatures below -100°C, sometimes sent by Aqara lumi.weather.

**Related issue (if applicable):** fixes #9978 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [ [... cut]
closed 2020-11-27 by s-hadinger