arendst/Tasmota #9979: Switch not publishing to HomeAssistant

yinonco, 2020-11-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have flashed tasmota on a shelly 1 device.
I want to detach the relay from the switch and have automations based on the switch in Home Assistant.
I did `SO114 1` and I can see the status sent on MQTT:
```
21:04:07 MQT: room/stat/main_switch/RESULT = {"Switch1":{"Action":"TOGGLE"}}
21:04:13 MQT: room/stat/main_switch/RESULT = {"Switch1":{"Action":"TOGGLE"}}
```
And when I turn on option 19 for HA discovery I get:
```
21:07:22 MQT: homeassistant/device_automation/C50D2A_SW_1_TOGGLE/config = {"atype":"trigger","t":"room/stat/main_switch/SWITCH1T","pl":"{\"TRIG\":\"TOGGLE\"}","type":"button_short_press","stype":"switch_1","dev":{"ids":["C50D2A"]}} (retained)
21:07:22 MQT: homeassistant/device_automation/C50D2A_SW_1_HOLD/config = {"atype":"trigger","t":"room/stat/main_switch/SWITCH1T","pl":"{\"TRIG\":\"HOLD\"}","type":"button_double_press","stype":"switch_1","dev":{"ids":["C50D2A"]}} (retained)
21:07:22 MQT: homeassistant/switch/C50D2A_RL_1/config = {"name":"Tasmota","stat_t":"room/tele/main_switch/STATE","avty_t":"room/tele/main_switch/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"room/cmnd/main_switch/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"C50D2A_RL_1","dev":{"ids":["C50D2A"]}} (retained)
21:07:22 MQT: homeassistant/sensor/C50D2A_status/config = {"name":"Room Main Switch status","stat_t":"room/tele/main_switch/HASS_STATE","avty_t":"room/tele/main_switch/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"room/tele/main_switch/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"C50D2A_status","dev":{"ids":["C50D2A"],"name":"Room Main Switch","mdl":"Shelly 1","sw":"9.1.0(tasmota)","mf":"Tasmota"}} (retained)
```
**I Had to define `SwitchTopic` in order for it to send the first two lines (In other device where I use buttons I didn't need to do it) so I defined it as `switch` so it wont control the relay**

The problem is that now - after I turned option 19 on I don't get mqtt messages at all for the switch state change - not those I got before on `room/stat/main_switch/RESULT` and not those suppose to be sent for HA on `room/stat/main_switch/SWITCH1T`

In other device where I have buttons instead of switches it worked as I wanted.

Thanks :)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 1
- [X] Tasmota binary firmware version number used: 9.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:16:31 CMD: Backlog Template; Module; GPIO 255
21:16:31 MQT: room/stat/main_switch/RESULT = {"NAME":"Shelly 1 Custom","GPIO":[1,1,0,225,224,193,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
21:16:32 MQT: room/stat/main_switch/RESULT = {"Module":{"46":"Shelly 1"}}
21:16:32 MQT: room/stat/main_switch/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"192":"Switch_n1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

Not using rules

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:18:18 CMD: Status 0
21:18:18 MQT: room/stat/main_switch/STATUS = {"Status":{"Module":46,"DeviceName":"Room Main Switch","FriendlyName":["Tasmota"],"Topic":"main_switch","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"switch","SwitchMode":[8,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:18:18 MQT: room/stat/main_switch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:15:57","StartupUTC":"2020-11-25T20:02:21","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-11-25T18:27:37","SaveCount":105,"SaveAddress":"F4000"}}
21:18:18 MQT: room/stat/main_switch/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"436/699"}}
21:18:18 MQT: room/stat/main_switch/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cohen_HB",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00800200","00006000","00000001"]}}
21:18:18 MQT: room/stat/main_switch/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:18:18 MQT: room/stat/main_switch/STATUS5 = {"StatusNET":{"Hostname":"main_switch-3370","IPAddress":"192.168.1.58","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"8C:AA:B5:C5:0D:2A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:18:18 MQT: room/stat/main_switch/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.112","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C50D2A","MqttUser":"smarthome","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:18:18 MQT: room/stat/main_switch/STATUS7 = {"StatusTIM":{"UTC":"2020-11-25T20:18:18","Local":"2020-11-25T21:18:18","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:14","Sunset":"16:59"}}
21:18:18 MQT: room/stat/main_switch/STATUS10 = {"StatusSNS":{"Time":"2020-11-25T21:18:18","Switch1":"ON"}}
21:18:18 MQT: room/stat/main_switch/STATUS11 = {"StatusSTS":{"Time":"2020-11-25T21:18:18","Uptime":"0T00:15:57","UptimeSec":957,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Cohen_HB","BSSId":"98:1E:19:7F:3C:1E","Channel":6,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Provided above in the description

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Provided above in the description

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect it to send the status to the triggers topic it created for HomeAssistant

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-25:
Please, check in unused entities in home assistant. May be it is under another name since you have used first the Native integration `so19 0` and the the mqtt discovery `so19 1`

ascillato, 2020-11-25:
Also, please update Tasmota to latest 9.1.0.2 from the dev branch ( http://ota.tasmota.com/tasmota/tasmota.bin.gz )

ascillato, 2020-11-25:
Remember that when using `so19 0` your Tasmota device will pop up under Integrations in HomeAssistant.

yinonco, 2020-11-26:
Ive updated to 9102 and it is still the same.
I didnt try with `SO19 0` (native HA integration) yet. I actually didnt know this option, I will give it a try and update

ascillato, 2020-11-26:
Please, check in unused entities in home assistant. May be it is under another name

effelle, 2020-11-26:
MQTT discovery needs switchtopic to be set in order to have the switches discovered and the switches will respond to a specific topic.
The official integration uses SO114 and the switches will respond to the default topic. 
I strongly suggest to remove the device from MQTT discovery and use directly on the new Integration.

ascillato2, 2020-11-27:
@yinonco 

Hi, any news on this? Were you able to solve it?

ascillato2, 2020-11-29:
Please, when you have time, tell us if you were able to solve your problem. Thanks.

closed 2020-11-29 by ascillato2

yinonco, 2020-11-29:
ok so on this device I was able to get it to work with SO19 0 but, I have another device - Shelly 2.5 which I use for roller shutter and if I turn Option 19 off HA doesnt recognize it as a cover (currently Option 19 is on and HA does see it as a cover)

effelle, 2020-11-29:
Please don't mix things. This issue was about switches, not covers.
Shutters are already supported for the official integration but you need to wait Home Assistant 0.119 to have them working.

yinonco, 2020-11-29:
ok Thanks 👍