arendst/Tasmota #9981: PWM dimmer module button rules do not override native behavior as expected.

marc-gist, 2020-11-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
on PWM Module 73 tasmota v9.1.0 making a rule for the button does not override native behaviour as per docs or other modules typically do.
I.e. setting up `rule1 on button1#state=3 do endon`, then `rule1 1`
when holding the bottom button (button1) I get the expected rule output, but I also get the lights dimming change :( 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __MJ Dimmer___
- [x ] Tasmota binary firmware version number used: __v9.1.0_and 9.1.0.2(tasmota)__
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _OTA____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:33:10 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:33:10 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Module":{"73":"PWM Dimmer"}}
19:33:10 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"GPIO0":{"34":"Button3"},"GPIO1":{"33":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"323":"Led_i4"},"GPIO4":{"576":"LedLink_i"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"321":"Led_i2"},"GPIO13":{"416":"PWM1"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"96":"Button_i1"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
19:33:41 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":27,"Free":484,"Rules":"on button1#state=3 do endon"}

```
- [ ] Provide the output of this command: `Status 0`:
```
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS = {"Status":{"Module":73,"DeviceName":"MJ Dimmer 4","FriendlyName":["MJ Dimmer 4"],"Topic":"tasmota_5FF201","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:31","StartupUTC":"2020-11-26T00:30:19","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-11-13T15:32:53","SaveCount":130,"SaveAddress":"FA000"}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-11-25T17:27:47","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"398/699"}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOTWAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","1405C8000100060000005A00000000000000","00800200","00006000","00000000"]}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS4 = {"StatusMEM":{"ProgramSize":605,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS5 = {"StatusNET":{"Hostname":"tasmota_5FF201-4609","IPAddress":"","Gateway":"","Subnetmask":"255.255.255.0","DNSServer":"","Mac":"EC:FA:BC:5F:F2:01","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS6 = {"StatusMQT":{"MqttHost":"10.20.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5FF201","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS7 = {"StatusTIM":{"UTC":"2020-11-26T00:33:50","Local":"2020-11-25T19:33:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:26","Sunset":"16:46"}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS10 = {"StatusSNS":{"Time":"2020-11-25T19:33:50"}}
19:33:50 MQT: iot/tasmota/stat/tasmota_5FF201/STATUS11 = {"StatusSTS":{"Time":"2020-11-25T19:33:50","Uptime":"0T00:03:31","UptimeSec":211,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":89,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWAN","BSSId":"0E:EC:DA:A1:6F:B5","Channel":1,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
19:34:40 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"WebLog":4}
19:34:41 CFG: Saved to flash at F9, Count 131, Bytes 4096
19:34:42 WIF: Checking connection...
19:34:45 APP: Button1 multi-press 1
19:34:45 SRC: Retry
19:34:45 MQT: iot/tasmota/tele/tasmota_5FF201/STATE = {"Time":"2020-11-25T19:34:45","Uptime":"0T00:04:26","UptimeSec":266,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"POWER":"OFF","Dimmer":89,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWAN","BSSId":"0E:EC:DA:A1:6F:B5","Channel":1,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:34:45 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"POWER":"OFF"}
19:34:45 MQT: iot/tasmota/stat/tasmota_5FF201/POWER = OFF
19:34:46 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Button1":{"Action":"SINGLE"}}
19:34:46 CFG: Saved to flash at F8, Count 132, Bytes 4096
19:34:51 APP: Button1 multi-press 1
19:34:51 SRC: Retry
19:34:51 MQT: iot/tasmota/tele/tasmota_5FF201/STATE = {"Time":"2020-11-25T19:34:51","Uptime":"0T00:04:32","UptimeSec":272,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":89,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWAN","BSSId":"0E:EC:DA:A1:6F:B5","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:34:51 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"POWER":"ON"}
19:34:51 MQT: iot/tasmota/stat/tasmota_5FF201/POWER = ON
19:34:51 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Button1":{"Action":"SINGLE"}}
19:34:52 CFG: Saved to flash at F7, Count 133, Bytes 4096
19:34:54 APP: Button1 multi-press 1
19:34:55 CFG: Saved to flash at F6, Count 134, Bytes 4096
19:34:56 RUL: BUTTON1#STATE=3 performs ""
19:34:56 SRC: Rule
19:34:56 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Button1":{"Action":"HOLD"}}
19:34:56 CFG: Saved to flash at F5, Count 135, Bytes 4096
19:34:57 CFG: Saved to flash at F4, Count 136, Bytes 4096
19:34:58 CFG: Saved to flash at FB, Count 137, Bytes 4096
19:34:59 MQT: iot/tasmota/tele/tasmota_5FF201/STATE = {"Time":"2020-11-25T19:34:59","Uptime":"0T00:04:40","UptimeSec":280,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":80,"MqttCount":1,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWAN","BSSId":"0E:EC:DA:A1:6F:B5","Channel":1,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:35:00 APP: Button1 multi-press 1
19:35:01 CFG: Saved to flash at FA, Count 138, Bytes 4096
19:35:02 RUL: BUTTON1#STATE=3 performs ""
19:35:02 SRC: Rule
19:35:02 MQT: iot/tasmota/stat/tasmota_5FF201/RESULT = {"Button1":{"Action":"HOLD"}}
19:35:02 CFG: Saved to flash at F9, Count 139, Bytes 4096
19:35:02 WIF: Checking connection...
19:35:03 CFG: Saved to flash at F8, Count 140, Bytes 4096
19:35:04 CFG: Saved to flash at F7, Count 141, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
using an MJ Dimmer (or PWM Dimmer with 3 buttons), `so32 20`, set the rule to decouple button 1 hold action.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
holding down button 1 should not change the dimmer values with a rule in place to change the behaviour of button hold.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

marc-gist, 2020-11-26:
Further testing; When `so19 1` is set (which I would like to have, but removed for testing for a time).

When I do hold, the MQTT message `BUTTON1T` is not sent (which I believe is normal, the rule has overridden that behaviour); however, note that the dimmer value changes in the weblog 4 space as per when so19 is off.

pcdiem, 2020-11-26:
The PR that implemented multi-press (https://github.com/arendst/Tasmota/pull/9589) noted that "button holds can not be overridden by rules because PWM Dimmer uses a fixed .5 seconds for the hold time since SetOption32 value is typically too high."

marc-gist, 2020-11-26:
> The PR that implemented multi-press (#9589) noted that "button holds can not be overridden by rules because PWM Dimmer uses a fixed .5 seconds for the hold time since SetOption32 value is typically too high."

Is that working/integrated yet, as I flashed with the latest development at 
```

tasmota.bin.gz | http://ota.tasmota.com/tasmota/tasmota.bin.gz | 430k | tasmota.bin | http://ota.tasmota.com/tasmota/tasmota.bin | 606k | 9.1.0.2 | 20201125 19:17
-- | -- | -- | -- | -- | -- | -- | --

```
or has it not been compile in yet?

marc-gist, 2020-11-26:
I just compiled from the latest `git pull` and it still doesn't work :( 
So either "merged" doesn't actually mean its in the code base yet, i'm missing something, or its still "broken" ;)

marc-gist, 2020-11-26:
doh, I guess i'm not reading the PR correctly.  They have implemented button press overrides, but the button hold can not be overridden yet?

I just want to disable the button1 hold.  I like the button2 and 3 hold features.

marc-gist, 2020-11-26:
Perhaps a feature request if this is working as designed please!

I really don't like that a power button controls brightness levels :(  I want to be able to trigger events off the hold power button.

I've tried to comment out the sections of code that I think are responsible for this, but I'm getting strange behavour, so clearly haven't found the correct spot yet... this maybe beyond my minor coding abilities :(

pcdiem, 2020-11-26:
In xdrv_35_pwm_dimmer.ino, you would change line 771 from

          else if (button_hold_time[button_index] <= now) {

to

          else if (button_index && button_hold_time[button_index] <= now) {

marc-gist, 2020-11-26:
> In xdrv_35_pwm_dimmer.ino, you would change line 771 from
> 
> ```
>       else if (button_hold_time[button_index] <= now) {
> ```
> 
> to
> 
> ```
>       else if (button_index && button_hold_time[button_index] <= now) {
> ```

Awesome, thanks!  I used this instead because I wonder if `button_index` is always **0** for the power button? just in case its not i did:

```
          else if (button_index != power_button_index && button_hold_time[button_index] <= now) {
```

now to lookup a way to create a SO for this :)

pcdiem, 2020-11-26:
I just submitted PR https://github.com/arendst/Tasmota/pull/9986 that supports rules overriding button holds.

closed 2020-11-29 by ascillato2