arendst/Tasmota #9982: HA Tasmota issue with PZEM

ascillato, 2020-11-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When using `SO19 0`, and Tasmota starts, the HADiscovery information is sent **_before_** the Energy Driver is initializated, so the `STATUS 10` is incorrect, making HATasmota Integration the incorrect entities on HomeAssistant.

* Incorrect `Status 10` at boot time:

```lua
01:24:16 MQT: stat/tableroprincipal/STATUS10 = {"StatusSNS":{"Time":"2020-11-26T01:24:16","Switch2":"ON","ENERGY":{"TotalStartTime":"2020-11-26T00:42:57","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0,0],"ApparentPower":[0,0,0],"ReactivePower":[0,0,0],"Factor":[0.00,0.00,0.00],"Voltage":[0,0,0],"Current":[0.000,0.000,0.000]}}}
```

* Correct `Status 10` after few seconds:

```lua
02:43:01 MQT: stat/tableroprincipal/STATUS10 = {"StatusSNS":{"Time":"2020-11-26T02:43:01","Switch2":"ON","ENERGY":{"TotalStartTime":"2020-11-26T00:42:57","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":232,"Current":0.000}}}
```

See Complete Console log output below.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _NodeMCU with PZEM_
- [x] Tasmota binary firmware version number used: _9.1.0.2 (knx)_
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _Tasmotizer_
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

02:07:05 CMD: Backlog Template; Module; GPIO 255
02:07:06 MQT: stat/tableroprincipal/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
02:07:06 MQT: stat/tableroprincipal/RESULT = {"Module":{"18":"Generic"}}
02:07:06 MQT: stat/tableroprincipal/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"1504":"PZEM004 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"161":"Switch2"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

NONE
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

02:07:47 MQT: stat/tableroprincipal/STATUS = {"Status":{"Module":18,"DeviceName":"Tablero Principal","FriendlyName":["Tablero Principal"],"Topic":"tableroprincipal","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:07:47 MQT: stat/tableroprincipal/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:20","StartupUTC":"2020-11-26T01:01:27","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-11-26T00:42:57","SaveCount":38,"SaveAddress":"F6000"}}
02:07:47 MQT: stat/tableroprincipal/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(knx)","BuildDateTime":"2020-11-25T17:27:43","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"425/699"}}
02:07:47 MQT: stat/tableroprincipal/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WirelessNetA",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00028009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
02:07:47 MQT: stat/tableroprincipal/STATUS4 = {"StatusMEM":{"ProgramSize":603,"Free":400,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAE597","04268011","000000CF","010013C0","8000F181","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
02:07:47 MQT: stat/tableroprincipal/STATUS5 = {"StatusNET":{"Hostname":"tableroprincipal-2856","IPAddress":"192.168.88.44","Gateway":"192.168.88.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.88.1","Mac":"5C:CF:7F:F7:AB:28","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
02:07:47 MQT: stat/tableroprincipal/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.88.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F7AB28","MqttUser":"hassmqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
02:07:47 MQT: stat/tableroprincipal/STATUS7 = {"StatusTIM":{"UTC":"2020-11-26T01:07:47","Local":"2020-11-26T02:07:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:16","Sunset":"16:58"}}
02:07:47 MQT: stat/tableroprincipal/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
02:07:47 MQT: stat/tableroprincipal/STATUS10 = {"StatusSNS":{"Time":"2020-11-26T02:07:47","Switch2":"ON","ENERGY":{"TotalStartTime":"2020-11-26T00:42:57","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":230,"Current":0.000}}}
02:07:47 MQT: stat/tableroprincipal/STATUS11 = {"StatusSTS":{"Time":"2020-11-26T02:07:47","Uptime":"0T00:06:20","UptimeSec":380,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WirelessNetA","BSSId":"74:4D:28:6C:FD:5C","Channel":7,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```lua
  Console output here:

00:00:00 CFG: Loaded from flash at F9, Count 27
[... 64 lines cut ...]

**If, after boot time, we set `SO19 1` and then `SO19 0` (that the energy driver is initializated) the `STATUS 10` is correct and the integration in HA is correct**

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

That the HADiscovery on `SO19 0` waits for the energy driver to initialize

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

None

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Workaround by the use of rules:

```lua
rule1 on system#boot do backlog so19 1;ruletimer1 10 endon on rules#timer=1 do so19 0 endon
rule1 1
restart 1
```

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-11-26:
Thanks Adrian, I found something similar with another sensor too and I'm looking for a viable solution without adding a big delay with millis().

ascillato, 2020-11-26:
Hi,

May be just doing that status 10 don't publish data if isn't valid. I was going to test this.

I have opened this issue to discuss options and find which could be the best choice.

Staars, 2020-11-26:
I did a workaround for the BLE-driver some time ago, where I simply ignore the first autodiscovery. (i.e. from line 2008 in MI32_ESP).

IMHO the only real solution would be, if "the core" of Tasmota would know, that all drivers are initialized. But for this we would need a mechanism, in which the amount of drivers is known to the system.

Maybe it would be possible, that we make it mandatory for a driver to do a pre-init, where it is guaranteed, that the system knows every driver, that is about to init. Later every driver must report success.

I think, that this is impossible to solve from the side of the ha-driver.

effelle, 2020-11-26:
Christian the new discovery is a bit different but I understand what you mean. In fact we discussed a lot about it!  
I completely agree this will be a good solution but I need to know what @arendst think about it since is a change of the code base, and should not be considered a solution just for Home Assistant.

Staars, 2020-11-26:
Yes, indeed it would at least add some complexity again, in the best case with a very small memory overhead.
No easy design decision.

ascillato, 2020-11-26:
So, what do you think?

A solution could be that all drivers report null until they have valid data AND if a driver is not going to be consistent in its JSON until it is initialized ( like PZEM that can have "Voltage":230 or "Voltage":[0,0,0] ) ?

effelle, 2020-11-26:
Having a nullified value (something like we already have for sensor not working properly like the DHT) will be the best solution.

arendst, 2020-11-26:
The reason behind having teleperiod set to minimal 10 seconds after a restart was to allow sensors to initialize before probing them for a result.

Some sensors just take longer to initialize and/or stabilize before a useable result can/will be delivered.

I think knowing when a sensor is ready makes no real difference as teleperiod will just come along in default 300 secs to reprobe.

I understand the urge to know valid results at HA poll time and after deepsleep but that can simply not be done for all sensors.

What's the sense of returning "null" values? Does the poller need to know what sensor can report valid results someday?

With regards to PZEM the init process taking several seconds is part of tasmota's PZEM detection scheme. It just needs several seconds in order to receive responses from all connected PZEM's. The code even contains a define called ``PZEM_AC_STABILIZE`` with a value of 30 seconds. If your "null" suggestion is used it needs to cope with sensors dropping off in case of only one connected PZEM; at first tasmota's tries to reach all possible three and in that case would report "null" for all of them.

ascillato, 2020-11-26:
The `null` value in sensors is very handy to not display values in the automation system. If a value of zero is sent instead, you will see a drop in history charts.

About the HATasmota, I think that a solution could be that the topics `tasmota/discovery/5CCF7FF7AB28/config` and `tasmota/discovery/5CCF7FF7AB28/sensors` be sent after the drivers are initialized, or after like 10 seconds that Tasmota have connected to MQTT. In the logs showed before, the PZEM is answering 2 seconds AFTER the discovery topics were sent.

effelle, 2020-11-26:
I can see another issue incoming Adrian. Looking at your example the the PZEM send ALL the possible sensors at boot time and then after init just a single one is presented on STATUS10. This means I will have 2 more sets discovered that aren't needed. 
Example:
```py
"Power":[0,0,0]
```
vs
```py
"Power":0
```
Then perhaps is better for me add a flag to delay the discovery for 10 seconds just for power metering.

ascillato, 2020-11-26:
Yes, that is the main issue here. The uninitialized driver, but if changing the Init delay of HATasmota from 2 to 10, this issue is solved:

https://github.com/arendst/Tasmota/blob/25dfb8417130ec38fd7adcfc38ffa55d2fa0895b/tasmota/xdrv_12_home_assistant.ino#L1175

Proposing the PR #9985 for review of this.

effelle, 2020-11-26:
OK, but this will delay completely the discovery in all cases, for both MQTT discovery and HATasmota.

ascillato, 2020-11-26:
Yes, It makes that at Tasmota restart, you will see values coming after 10 seconds and not after 2 seconds, but they are going to be the correct and initialized values for PZEM and several other sensors.

PR #9985 (2020-11-26): Fix HA Discovery Issue with Uninitialized Sensors
## Description:

Fix HA Discovery Issue with Uninitialized Sensors

Change HA Discovery's Init delay from 2 to 10 seconds

**Related issue (if applicable):** fixes #9982

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accep [... cut]
closed 2020-11-26 by arendst