arendst/Tasmota #9984: TuyaMCU Temperature Function IDs

michanehls, 2020-11-26:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
no

**Describe the solution you'd like**  
Add fnid for temperatures

**Describe alternatives you've considered**  
no alternatives

**Additional context**  
My thermostat has the following temperature related dpids
* Target Temp: 10°C to 35°C in 1° steps
* Current Temp: in 1° steps
* Temp calibration: -9°C to +9°C in 1° steps
* Swing: 1°C to 9°C in 1° steps

Data values from TuyaMCU are in hex

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-11-26:
Temperature and set temperature are already in the work and will be shown on `/sensors` topic and on the webui. 
Temp calibration and swing will be not added, you can easily manage them with tuyasend.

michanehls, 2020-11-26:
Where does the development take place? I could flash development versions and test it.

effelle, 2020-11-26:
You can flash the devusing the bin from `http://ota.tasmota.com/tasmota/tasmota.bin.gz`. 
I should merge the new sensors on dev branch this week. I'll mention this issue when the update will be ready.

effelle, 2020-11-28:
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-28 by effelle