arendst/Tasmota #9987: Battery state for Moes HY386-ZB TRV

galsom, 2020-11-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I was able to successfully pair a Moes HY386-ZB TRV for which support was recently added 👍 

Different attributes such as e.g. the `TuyaChildLock` are reported, but the `TuyaBattery` property is not showing up.

From a search in the source code I presume it should work, but nothing seems to appear. When having a look at the websites of zigbee2mqtt.io and zigbee.blakadder.com, it is reported as working, but I can't verify that.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Moes HY386-ZB TRV
- [x] Tasmota binary firmware version number used: Tasmota 9.1.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:41:46 CMD: Backlog Template; Module; GPIO 255
17:41:46 SRC: WebConsole from 10.1.1.243
17:41:46 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
17:41:46 SRC: Backlog
17:41:46 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
17:41:46 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:41:47 SRC: Backlog
17:41:47 CMD: Group 0, Index 1, Command "MODULE", Data ""
17:41:47 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
17:41:47 SRC: Backlog
17:41:47 CMD: Group 0, Index 1, Command "GPIO", Data "255"
17:41:47 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

17:42:59 CMD: Status 0
17:42:59 SRC: WebConsole from 10.1.1.243
17:42:59 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS = {"Status":{"Module":75,"DeviceName":"Living Room Zigbee Bridge","FriendlyName":["Living Room Zigbee Bridge"],"Topic":"living-room-zigbee-bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin","RestartReason":"Hardware Watchdog","Uptime":"1T21:43:49","StartupUTC":"2020-11-24T18:59:10","Sleep":50,"CfgHolder":4617,"BootCount":70,"BCResetTime":"2020-10-15T15:53:45","SaveCount":137,"SaveAddress":"FB000"}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(zbbridge)","BuildDateTime":"2020-11-17T18:58:15","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"604/699"}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["x",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00088009","2805C8000100060000005A00000000000000","00008000","00006080","00000000"]}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS4 = {"StatusMEM":{"ProgramSize":576,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS5 = {"StatusNET":{"Hostname":"living-room-zigbee-bridge","IPAddress":"10.1.1.23","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.106","Mac":"A4:CF:12:D7:F9:6C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.101","MqttPort":1883,"MqttClientMask":"living-room-zigbee-bridge","MqttClient":"living-room-zigbee-bridge","MqttUser":"living-room-zigbee-bridge","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-26T16:42:59","Local":"2020-11-26T17:42:59","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:16","Sunset":"16:58"}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-26T17:42:59"}}
17:42:59 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-26T17:42:59","Uptime":"1T21:43:49","UptimeSec":164629,"Vcc":3.439,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"x","BSSId":"E0:63:DA:81:D2:C6","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:13"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Review the console output and look for the `TuyaBattery`message.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

From time to time that the `TuyaBattery` value is reported

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

When running the `ZBinfo` command, the following output is shown

```
17:50:29 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/tele/80F8/SENSOR = {"ZbInfo":{"0x80F8":{"Device":"0x80F8","IEEEAddr":"0x60A423FFFEAB34FF","ModelId":"TS0601","Manufacturer":"_TZE200_ckud7u2l","Endpoints":[1],"Config":[],"Reachable":true,"LastSeen":575,"LastSeenEpoch":1606408854,"LinkQuality":115}}}
17:50:29 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/RESULT = {"ZbInfo":"Done"}
```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-12-15:
Can you please provide `Weblog 3` level logs when receiving those messages?

Tuya Zigbee code has been improved lately, you should be able to read the attribute with:

`ZbSend {"Device":"0x80F8","Read":{"TuyaBattery":true}}`

kroon040, 2020-12-15:
@s-hadinger 

I also tried , build from 2 days ago

22:01:37 CMD: ZbSend {"Device":"0xCB5D","Read":{"TuyaBattery":true}}
22:01:37 SRC: WebConsole from 192.168.180.118
22:01:37 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0xCB5D","Read":{"TuyaBattery":true}}"
22:01:37 ZIG: guessing endpoint 1
22:01:37 ZIG: ZbZNPSent 2402025DCB0000000000000100000100EF57301E05001057001502
22:01:37 MQT: stat/z2t/RESULT = {"ZbSend":"Done"}
22:01:37 ZIG: {"ZbZNPReceived":"640200"}
22:01:38 ZIG: {"ZbZNPReceived":"4480000157"}
22:01:38 ZIG: {"ZbZNPReceived":"4481000000EF5DCB0101004C003B9FDF0000061857011502865DCB1D"}
22:01:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCB5D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":76,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":87,"cmdid":"0x01","payload":"150286"}}
22:01:38 ZIG: ZbZCLRawReceived: {"0xCB5D":{"Endpoint":1,"LinkQuality":76}}

s-hadinger, 2020-12-16:
Thanks, the response means that the attribute `TuyaBattery` is unknown on this device. I checked Zigbee2MQTT, and it seems that this device does not report `TuyaBattery`.

Do you have any additional information?

kroon040, 2020-12-16:
in the description of the TS0601 battery is nummeric and battery_low is as binairy available?

https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

galsom, 2020-12-16:
Thank you @s-hadinger for having a look at it. The following output is what I receive:

```
23:10:14 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x80F8","Read":{"TuyaBattery":true}}"
23:10:14 ZIG: guessing endpoint 1
23:10:14 ZIG: ZbEZSPSend 340000F880040100EF0101400100001601051016001502
23:10:14 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/stat/RESULT = {"ZbSend":"Done"}
23:10:14 ZIG: {"ZbEZSPReceived":"340000F3"}
23:10:22 ZIG: {"ZbEZSPReceived":"800042F880"}
23:10:22 MQT: buildings/casa/ground-floor/living-room/dal/zigbee-bridge/tele/RESULT = {"ZbRouteError":{"ShortAddr":"0x80F8","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
23:10:28 ZIG: {"ZbEZSPReceived":"45000000003880000040010000F4FF000000FFFF1B000000F8FF0714000F0010D0B7CDA9C0D6D0D1C4DAACA9B9C5B7D0"}
23:10:28 ZIG: Internal ZDO message 0x8038 sent from 0x0000 
23:10:38 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000F8C8CEF880FFFF0618160115028602"}
23:10:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x80F8","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":97,"securityuse":0,"seqnumber":248,"fc":"0x18","manuf":"0x0000","transact":22,"cmdid":"0x01","payload":"150286"}}
23:10:38 ZIG: ZbZCLRawReceived: {"0x80F8":{"Endpoint":1,"LinkQuality":97}}
23:10:38 ZIG: {"ZbEZSPReceived":"3F0000F880040100EF010140010000F3010000"}
```
Not sure if this is the entire output you need.

Draggon, 2020-12-21:
Hi @s-hadinger 

I'v tried to use read for known parameters like Target temp - but after send nothing is being published to mqtt:
```
12:22:14 CMD: ZbSend {"Device":"0xA819","Read":{"TuyaTempTarget":true}}
12:22:14 SRC: WebConsole from 192.168.1.100
12:22:14 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0xA819","Read":{"TuyaTempTarget":true}}"
12:22:14 ZIG: guessing endpoint 1
12:22:14 ZIG: ZbEZSPSend 34000019A8040100EF0101400100001301051013000202
12:22:14 MQT: sonoffzb/stat/RESULT = {"ZbSend":"Done"}
12:22:14 ZIG: {"ZbEZSPReceived":"34000090"}
12:22:16 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000D5B4C919A8FFFF0618130102028602"}
12:22:16 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xA819","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":84,"securityuse":0,"seqnumber":213,"fc":"0x18","manuf":"0x0000","transact":19,"cmdid":"0x01","payload":"020286"}}
12:22:16 ZIG: ZbZCLRawReceived: {"0xA819":{"Endpoint":1,"LinkQuality":84}}
12:22:16 ZIG: {"ZbEZSPReceived":"3F000019A8040100EF01014001000090010000"}
```

s-hadinger, 2020-12-23:
@Draggon This is also an error message. Do you have logs from the device reporting `TuyaTempTarget`?

Either the device does not support this attribute, or there is a bug Z2T when crafting the message.

Draggon, 2020-12-23:
It's being reported e.g when I change mode of device or when I set it:

```
11:11:15 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E68","Endpoint":1,"Write":{"TuyaTempTarget": 21.5}}"
11:11:15 ZIG: ZbEZSPSend 340000685E040100EF01014001000002010D110200000102020004000000D7
11:11:15 MQT: sonoffzb/stat/RESULT = {"ZbSend":"Done"}
11:11:15 ZIG: {"ZbEZSPReceived":"340000CA"}
11:11:16 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460C0CC010000"}
11:11:16 ZIG: {"ZbEZSPReceived":"3F0000685E040100EF010140010000CA010000"}
11:11:16 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460C0CC010000"}
11:11:16 ZIG: {"ZbEZSPReceived":"450000040100EF01010001000066C0CC685EFFFF0D096A02308702020004000000D704"}
11:11:16 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":92,"securityuse":0,"seqnumber":102,"fc":"0x09","manuf":"0x0000","transact":106,"cmdid":"0x02","payload":"308702020004000000D7"}}
11:11:16 ZIG: ZbZCLRawReceived: {"0x5E68":{"EF00/0202":215,"Endpoint":1,"LinkQuality":92}}
11:11:16 MQT: sonoffzb/tele/5E68/SENSOR = {"ZbReceived":{"0x5E68":{"Device":"0x5E68","TuyaTempTarget":21.5,"Endpoint":1,"LinkQuality":92}}} (retained)
11:11:26 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460BCCB010000"}
11:11:26 ZIG: {"ZbEZSPReceived":"450000040100EF01010001000067BCCB685EFFFF0A096B013088040400010204"}
11:11:26 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":89,"securityuse":0,"seqnumber":103,"fc":"0x09","manuf":"0x0000","transact":107,"cmdid":"0x01","payload":"30880404000102"}}
11:11:26 ZIG: ZbZCLRawReceived: {"0x5E68":{"EF00/0404":2,"Endpoint":1,"LinkQuality":89}}
```

s-hadinger, 2020-12-23:
Sorry, I realize that reading Tuya Zigbee attributes is not implemented. Only reporting and writing. Actually I'm not even sure how to read an attribute (the Tuya doc does not mention this possibility).

Before coding anything I need to confirm that it works.

@Draggon  can you please try
`ZbSend {"Device":"0x5E68","Send":"EF00!00/000102020000"}`

Draggon, 2020-12-23:
Here You are:

```
17:02:59 CMD: ZbSend {"Device":"0x5E68","Send":"EF00!00/000102020000"}
17:02:59 SRC: WebConsole from 192.168.1.100
17:02:59 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E68","Send":"EF00!00/000102020000"}"
17:02:59 ZIG: guessing endpoint 1
17:02:59 ZigbeeZCLSend device: 0x5E68, group: 0x0000, endpoint:1, cluster:0xEF00, cmd:0x00, send:"000102020000"
17:02:59 ZbSend: shortaddr 0x5E68, groupaddr 0x0000, cluster 0xEF00, endpoint 0x01, cmd 0x00, data 
17:02:59 ZIG: ZbEZSPSend 340000685E040100EF010140010000040109010400000102020000
17:02:59 MQT: sonoffzb/stat/RESULT = {"ZbSend":"Done"}
17:02:59 ZIG: {"ZbEZSPReceived":"3400000A"}
17:03:00 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460BCCB010000"}
17:03:00 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460BCCB010000"}
17:03:00 ZIG: {"ZbEZSPReceived":"450000040100EF010140010000ACC0CC685EFFFF0508040B000004"}
17:03:00 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":92,"securityuse":0,"seqnumber":172,"fc":"0x08","manuf":"0x0000","transact":4,"cmdid":"0x0B","payload":"0000"}}
17:03:00 MQT: sonoffzb/tele/RESULT = {"ZbResponse":{"Device":"0x5E68","Command":"EF00!00","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":92}}
17:03:00 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460BCCB010000"}
17:03:00 ZIG: {"ZbEZSPReceived":"3F0000685E040100EF0101400100000A010000"}
17:03:00 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000ADBCCB685EFFFF0D09220230FE020200040000009604"}
17:03:00 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":89,"securityuse":0,"seqnumber":173,"fc":"0x09","manuf":"0x0000","transact":34,"cmdid":"0x02","payload":"30FE0202000400000096"}}
17:03:00 ZIG: ZbZCLRawReceived: {"0x5E68":{"EF00/0202":150,"Endpoint":1,"LinkQuality":89}}
17:03:00 MQT: sonoffzb/tele/5E68/SENSOR = {"ZbReceived":{"0x5E68":{"Device":"0x5E68","TuyaTempTarget":15,"Endpoint":1,"LinkQuality":89}}} (retained)
```

s-hadinger, 2020-12-23:
@Draggon Was the `TuyaTempTarget` already at `15` before this command?

Now can you also try 
`ZbSend {"Device":"0x5E68","Send":"EF00!00/000115020000"}`

I hope it will trigger a `TuyaBattery` reporting.

Draggon, 2020-12-23:
As to `TuyaTempTarget  15` - good point - no it was 21,5 :)

As to above command here is result:
```
22:23:49 CMD: ZbSend {"Device":"0x5E68","Send":"EF00!00/000115020000"}
22:23:49 SRC: WebConsole from 192.168.1.100
22:23:49 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E68","Send":"EF00!00/000115020000"}"
22:23:49 ZIG: guessing endpoint 1
22:23:49 ZigbeeZCLSend device: 0x5E68, group: 0x0000, endpoint:1, cluster:0xEF00, cmd:0x00, send:"000115020000"
22:23:49 ZbSend: shortaddr 0x5E68, groupaddr 0x0000, cluster 0xEF00, endpoint 0x01, cmd 0x00, data 
22:23:49 ZIG: ZbEZSPSend 340000685E040100EF010140010000050109010500000115020000
22:23:49 MQT: sonoffzb/stat/RESULT = {"ZbSend":"Done"}
22:23:49 ZIG: {"ZbEZSPReceived":"340000A5"}
22:23:51 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460B0C8010000"}
22:23:51 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460B4C9010000"}
22:23:51 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460B8CA010000"}
22:23:51 ZIG: {"ZbEZSPReceived":"5900685EB842ABFEFF23A460B4C9010000"}
22:23:51 ZIG: {"ZbEZSPReceived":"450000040100EF010140010000B9B4C9685EFFFF0508050B000004"}
22:23:51 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":84,"securityuse":0,"seqnumber":185,"fc":"0x08","manuf":"0x0000","transact":5,"cmdid":"0x0B","payload":"0000"}}
22:23:51 MQT: sonoffzb/tele/RESULT = {"ZbResponse":{"Device":"0x5E68","Command":"EF00!00","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":84}}
22:23:51 ZIG: {"ZbEZSPReceived":"3F0000685E040100EF010140010000A5010000"}
22:23:51 ZIG: {"ZbEZSPReceived":"450000040100EF010140010000BAB8CA685EFFFF0508050B000004"}
22:23:51 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0x5E68","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":86,"securityuse":0,"seqnumber":186,"fc":"0x08","manuf":"0x0000","transact":5,"cmdid":"0x0B","payload":"0000"}}
22:23:51 MQT: sonoffzb/tele/RESULT = {"ZbResponse":{"Device":"0x5E68","Command":"EF00!00","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":86}}
```

s-hadinger, 2020-12-23:
Hmmm. No luck. It didn't work

galsom, 2021-01-18:
I did not answer the past few weeks not to pressure anyone in seeing if we can find a solution for this.

Is there anything we can do to help with e.g. testing things out?

closed 2021-01-26 by github-actions[bot]