arendst/Tasmota #9997: TuyaMCU additional fnId for Total kWh value of Energy Meter and make SetOption72 work...

krikk, 2020-11-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I have setup Tasmota on a  Hiking DDS238-2 WIFI Din Rail Energy Meter with the following Template: https://templates.blakadder.com/hiking_DDS238-2.html, but this Energy Meter also gives back the Total kWh Value on the DpID of 101. e.g:  i have a value of 30 (0x1e) on DpID101 which means 0.30 kWh Total.
Tasmota currently only supports fnId 31 to 33 for Power Monitoring (see here: https://tasmota.github.io/docs/TuyaMCU/#fnid) which are Power/Current/Voltage

**Describe the solution you'd like**  
It would be great if someone could add a fnId of e.g. 34 for Total kWh which could be mapped via TuyaMCU command, if this works it would also be great if SetOption72 1 (Set reference used for total energy) would use this Total kWh value as reference.

**Describe alternatives you've considered**  
an workaround it to read the raw reported TuyaReceived Values, but this works only with additional software like openhab or homeassistant

**Additional context**  
issue is also discussed here: https://github.com/arendst/Tasmota/issues/8237#issuecomment-734475045

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-11-27:
Not as easy as it seems.
In your case is even worse since dpid 101 means is a custom function.
To be more clear, all the dpId over the id number 100 are custom implementation and are created by the manufacturer, not by Tuya. 
The good news is Tasmota will try to calculate the total energy using the data received:

```py
23:17:01 RSL: tele/tasmota_49A3BC/SENSOR = {"Time":"2020-11-27T23:17:01","ENERGY":{"TotalStartTime":"2020-11-27T22:48:26","Total":0.006,"Yesterday":0.000,"Today":0.006,"Period":4,"Power":64,"ApparentPower":64,"ReactivePower":0,"Factor":1.00,"Voltage":13,"Current":5.000}}
```
Perhaps will be not as good as the integrated Tuya one but is surely better than nothing.
Now, about the supposed issue with voltage resolution, those are the latest data format used on a Tuya device.  
```
1 - Total forward energy | Only report | Integer | Values range:0-99999999, Pitch1, Scale2, Unit:kW·h
18 - Current | Only report | Integer | Values range: 0-80000, Pitch1, Scale0, Unit:mA 
19 - Power | Only report | Integer | Values range:0-200000, Pitch1, Scale1, Unit:W
20 - Voltage  | Only report | Integer | Values range:0-5000, Pitch1, Scale1, Unit:V
```
I will not be surprised if your device will be not compliant with those data, Tuya has changed the dpid configuration at least 2 times already for many type of devices.
Paste here a log with the relevant dpid in action, I'll have a look. (NO ETA).

krikk, 2020-11-28:
the only missing function is the total kWh, the voltage/current/power all work fine and tasmota correctly calculates the total kWh in software, so nothing needs to be changed with this values...(tested on 9.1.0.2) the total kWh value from TuyaMCU would be a nice addition :)

but i noticed another strange issue: i get a very high Power Reading Today of e.g.: 2766.857 kWh
this always happens after restarting tasmota

steps to reproduce:
Backlog EnergyReset1 0; EnergyReset1 0; EnergyReset2 0;
...everything shows 0
restart 1

After Reboot its 0, but i think after the first tuya update it is once again have 2766.857 kWh <-- where does it get this value ?
my Tasmota config: https://github.com/krikk/Hiking-DDS238-2-WIFI-Din-Rail-Energy-Meter-flashing-Tasmota/blob/main/README.md

effelle, 2020-11-28:
> where does it get this value ?

You tell me! :)
If you are sure is after an update from the MCU set the console to weblog 4 and check what `tuyareceived` got at boot time then wait for the next event in console.  Another way to see if tasmota is receiving some strange values is to enable the extra /stat topic with `tuyasend9`.

krikk, 2020-11-28:
weblog after resetting values and restart:

```console
00:00:00 CFG: Loaded from flash at FB, Count 33
00:00:00 QPC: Count 1
00:00:00 CFG: CR 412/699, Busy 0
00:00:00 SNS: Hardware Serial
00:00:00 TYA: Request MCU configuration at 9600 bps
00:00:00 TYA: Send "55aa0001000000"
00:00:00 SRC: Restart
00:00:00 TYA: Send "55aa0006000501010001010e"
00:00:00 Project tasmota Tasmota Version 9.1.0.2(tasmota)-2_7_4_7
00:00:00 TYA: Set WiFi LED 2 (-1)
00:00:00 TYA: Send "55aa000300010205"
00:00:00 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A227377396173796A6A386A343466726A6C222C2276223A22312E302E30222C226D223A307DEB","Cmnd":1,"CmndData":"7B2270223A227377396173796A6A386A343466726A6C222C2276223A22312E302E30222C226D223A307D"}}
00:00:00 TYA: MCU Product ID: {"p":"sw9asyjj8j44frjl","v":"1.0.0","m":0}
00:00:00 TYA: Send "55aa0002000001"
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 xxxxx Channel 1 BSSId xxxxxx in mode 11N as tasmota_1225D1-1489...
00:00:00 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
00:00:00 TYA: fnId=11 is set for dpId=1
00:00:00 TYA: RX Relay-1 --> MCU State: On Current State:On
00:00:00 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
00:00:00 TYA: fnId=11 is set for dpId=1
00:00:00 TYA: RX Relay-1 --> MCU State: On Current State:On
00:00:00 {"TuyaReceived":{"Data":"55AA030200020E1024","Cmnd":2,"CmndData":"0E10"}}
00:00:00 TYA: RX MCU configuration Mode=2
00:00:00 TYA: Read MCU state
00:00:00 TYA: Send "55aa0008000007"
00:00:00 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
00:00:00 TYA: fnId=11 is set for dpId=1
00:00:00 TYA: RX Relay-1 --> MCU State: On Current State:On
00:00:00 {"TuyaReceived":{"Data":"55AA03070008090200040000000020","Cmnd":7,"CmndData":"0902000400000000","DpType2Id9":0,"9":{"DpId":9,"DpIdType":2,"DpIdData":"00000000"}}}
00:00:00 TYA: fnId=0 is set for dpId=9
00:00:00 TYA: RX value 0 from dpId 9 
00:00:00 {"TuyaReceived":{"Data":"55AA0307000865020004000000209C","Cmnd":7,"CmndData":"6502000400000020","DpType2Id101":32,"101":{"DpId":101,"DpIdType":2,"DpIdData":"00000020"}}}
00:00:00 TYA: fnId=0 is set for dpId=101
00:00:00 TYA: RX value 32 from dpId 101 
00:00:00 {"TuyaReceived":{"Data":"55AA0307000812020004000000264F","Cmnd":7,"CmndData":"1202000400000026","DpType2Id18":38,"18":{"DpId":18,"DpIdType":2,"DpIdData":"00000026"}}}
00:00:00 TYA: fnId=32 is set for dpId=18
00:00:00 TYA: RX value 38 from dpId 18 
00:00:00 TYA: Rx ID=18 Current=38
00:00:01 {"TuyaReceived":{"Data":"55AA0307000813020004000000406A","Cmnd":7,"CmndData":"1302000400000040","DpType2Id19":64,"19":{"DpId":19,"DpIdType":2,"DpIdData":"00000040"}}}
00:00:01 TYA: fnId=31 is set for dpId=19
00:00:01 TYA: RX value 64 from dpId 19 
00:00:01 TYA: Rx ID=19 Active_Power=64
00:00:01 {"TuyaReceived":{"Data":"55AA03070008140200040000092559","Cmnd":7,"CmndData":"1402000400000925","DpType2Id20":2341,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00000925"}}}
00:00:01 TYA: fnId=33 is set for dpId=20
00:00:01 TYA: RX value 2341 from dpId 20 
00:00:01 TYA: Rx ID=20 Voltage=2341
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on tasmota_1225D1-1489 with IP address 192.168.11.80
00:00:01 HTP: Console
13:09:59 RTC: UTC 2020-11-28T12:09:59, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
13:09:59 TYA: Set WiFi LED 3 (0)
13:09:59 TYA: Send "55aa000300010306"
13:09:59 RSL: tele/tasmota_1225D1/INFO1 = {"Module":"Hiking DDS238-2 WIFI","Version":"9.1.0.2(tasmota)","FallbackTopic":"cmnd/DVES_1225D1_fb/","GroupTopic":"cmnd/tasmotas/"}
13:09:59 RSL: tele/tasmota_1225D1/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_1225D1-1489","IPAddress":"192.168.11.80"}
13:09:59 RSL: tele/tasmota_1225D1/INFO3 = {"RestartReason":"Software/System restart"}
13:09:59 RSL: stat/tasmota_1225D1/RESULT = {"POWER":"ON"}
13:09:59 RSL: stat/tasmota_1225D1/POWER = ON
13:09:59 RUL: SYSTEM#BOOT performs "RuleTimer1 5"
13:09:59 SRC: Rule
13:09:59 CMD: Group 0, Index 1, Command "RULETIMER", Data "5"
13:09:59 RSL: stat/tasmota_1225D1/RESULT = {"T1":5,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
13:10:03 QPC: Reset
13:10:04 HTP: Main Menu
13:10:04 RSL: tele/tasmota_1225D1/STATE = {"Time":"2020-11-28T13:10:04","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxx","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
[... 23 lines cut ...]
13:10:05 TYA: RX value 0 from dpId 9 
13:10:05 {"TuyaReceived":{"Data":"55AA0307000865020004000000209C","Cmnd":7,"CmndData":"6502000400000020","DpType2Id101":32,"101":{"DpId":101,"DpIdType":2,"DpIdData":"00000020"}}}
13:10:05 TYA: fnId=0 is set for dpId=101
13:10:05 TYA: RX value 32 from dpId 101 
13:10:05 {"TuyaReceived":{"Data":"55AA03070008120200040000002750","Cmnd":7,"CmndData":"1202000400000027","DpType2Id18":39,"18":{"DpId":18,"DpIdType":2,"DpIdData":"00000027"}}}
13:10:05 TYA: fnId=32 is set for dpId=18
13:10:05 TYA: RX value 39 from dpId 18 
13:10:05 TYA: Rx ID=18 Current=39
13:10:05 SRC: Backlog
13:10:05 CMD: Group 0, Index 1, Command "RULETIMER", Data "5"
[... cut ...]
13:10:05 {"TuyaReceived":{"Data":"55AA0307000814020004000009265A","Cmnd":7,"CmndData":"1402000400000926","DpType2Id20":2342,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00000926"}}}
13:10:05 TYA: fnId=33 is set for dpId=20
13:10:05 TYA: RX value 2342 from dpId 20 
13:10:05 TYA: Rx ID=20 Voltage=2342
13:10:06 CFG: Saved to flash at FA, Count 34, Bytes 4096
13:10:06 TYA: Send "55aa00000000ff"
13:10:06 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
13:10:06 TYA: Heartbeat
13:10:07 HTP: Main Menu
```

arendst, 2020-11-29:
@effelle pls see my fix for #10005 

Duplicate issues. Great!

closed 2020-11-29 by effelle

inorhan, 2020-12-28:
> Now, about the supposed issue with voltage resolution, those are the latest data format used on a Tuya device.

Hi all, thanks for your work!
I have several DDS238-2 wifi. There are at least 2 different versions by the external appearance and behaviour. 
The "old" version with tasmota 9.2, shows thousands of volts (normally around 3100vac, in reality should be 229vac).

22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA0301002A7B2270223A227377396173796A6A386A343466726A6C222C2276223A22312E302E30222C226D223A307DEB","Cmnd":1,"CmndData":"7B2270223A227377396173796A6A386A343466726A6C222C2276223A22312E302E30222C226D223A307D"}}
22:48:40 TYA: MCU Product ID: {"p":"sw9asyjj8j44frjl","v":"1.0.0","m":0}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA030200020E1024","Cmnd":2,"CmndData":"0E10"}}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA03070008090200040000000020","Cmnd":7,"CmndData":"0902000400000000","DpType2Id9":0,"9":{"DpId":9,"DpIdType":2,"DpIdData":"00000000"}}}
22:48:40 MQT: stat/induccionhorno/RESULT = {"T1":10,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA030700086502000400001249D7","Cmnd":7,"CmndData":"6502000400001249","DpType2Id101":4681,"101":{"DpId":101,"DpIdType":2,"DpIdData":"00001249"}}}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA03070008120200040000000029","Cmnd":7,"CmndData":"1202000400000000","DpType2Id18":0,"18":{"DpId":18,"DpIdType":2,"DpIdData":"00000000"}}}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA0307000813020004000000002A","Cmnd":7,"CmndData":"1302000400000000","DpType2Id19":0,"19":{"DpId":19,"DpIdType":2,"DpIdData":"00000000"}}}
22:48:40 MQT: tele/induccionhorno/RESULT = {"TuyaReceived":{"Data":"55AA030700081402000400037C4BF5","Cmnd":7,"CmndData":"1402000400037C4B","DpType2Id20":228427,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00037C4B"}}}

DpType2Id20":228427  -> 228.427 VAC  (shows >3000VAC)

The new version, doesn't have that problem, it shows as it should.

Also I'm expiriencing the same problems with the Total kWh going crazy in the new and old versions of the DD238-2 Wifi.

inorhan, 2021-01-30:
Hello.
Did someone take a look at the "resolution problem" ?  DpType2Id20":228427 -> 228.427 VAC (shows >3000VAC)}
Due to the wrong Voltage values, Apparent Power and Reactive power are incorrect too...
Regards

RaW79, 2024-07-15:
Hi
I´m testing on a Zemismart [SPM01-D2TW](https://www.zemismart.com/products/spm01-d2tw-zm) which was also reviewed by [blakkader](https://blakadder.com/zemismart-spm01/) Wifi Version but after HW change to ESP02S the fnId alignment seems to be impossible and I´m looking for a funktion to read out the energy meter which is povided by dPid1:
`16:41:25.408 RSL: RESULT = {"TuyaSend":"Done"}
16:41:25.488 DMP: 55 AA 03 07 00 08 01 02 00 04 00 00 00 05 1D
16:41:25.491 {"TuyaReceived":{"Data":"55AA0307000801020004000000051D","Cmnd":7,"CmndData":"0102000400000005","DpType2Id1":5,"1":{"DpId":1,"DpIdType":2,"DpIdData":"00000005"}}}
16:41:25.494 TYA: fnId=0 is set for dpId=1
16:41:25.496 TYA: RX value 5 from dpId 1 `

If I´m right here is suggested to use voltage and current to be calculated by Tasmota but this is my next problem because there are 3 data values combined in one dPId=6:
`16:41:25.507 DMP: 55 AA 03 07 00 0C 06 00 00 08 09 11 00 00 32 00 00 05 74
16:41:25.510 {"TuyaReceived":{"Data":"55AA0307000C06000008091100003200000574","Cmnd":7,"CmndData":"060000080911000032000005","DpType0Id6":"0x0911000032000005","6":{"DpId":6,"DpIdType":0,"DpIdData":"0911000032000005"}}}
16:41:25.511 TYA: fnId=33 is set for dpId=6`

Which means:
[image]

Kindly ask for your support/hints how to implement and read out dpIx=1 (energy counter) and if you can give hint for dpIx=6 would be great. Big Thx!