arendst/Tasmota #10005: TuyaMCU: Total kWh value gets random? value after reboot starting with version 9.1.0.2

krikk, 2020-11-28:
### PROBLEM DESCRIPTION
 Hiking DDS238-2 WIFI Din Rail Energy Meter with Tasmota, Version 9.1.0.2, reset to factory defaults, wifi config, config module as TuyaMCU, set TuyaRx and TuyaTx Gpio Right, config TuyaMCU with this:
```console
Backlog TuyaMCU 33,20; TuyaMCU 32,18; TuyaMCU 31,19;
```
after this i now get a Value of e.g 5392. for the total energy value... i also tried the same under 9.1.0 there this does not happen

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  Hiking DDS238-2 WIFI Din Rail Energy Meter
- [ ] Tasmota binary firmware version number used:  9.1.0.2 & 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:29:29 CMD: Backlog Template; Module; GPIO 255
20:29:29 SRC: WebConsole from 192.168.11.5
20:29:29 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:29:29 SRC: Backlog
20:29:29 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:29:29 RSL: stat/tasmota_1225D1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:29:30 SRC: Backlog
20:29:30 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:29:30 RSL: stat/tasmota_1225D1/RESULT = {"Module":{"54":"Tuya MCU"}}
20:29:30 SRC: Backlog
20:29:30 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:29:30 RSL: stat/tasmota_1225D1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
20:29:33 TYA: Send "55aa00000000ff"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:30:07 SRC: WebConsole from 192.168.11.5
20:30:07 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:30:07 RSL: stat/tasmota_1225D1/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1225D1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:06","StartupUTC":"2020-11-28T19:20:01","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-11-28T20:17:43","SaveCount":18,"SaveAddress":"FA000"}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"366/699"}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["www.krikkit.net",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1225D1-1489","IPAddress":"192.168.11.80","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"D8:BF:C0:12:25:D1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1225D1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS7 = {"StatusTIM":{"UTC":"2020-11-28T19:30:07","Local":"2020-11-28T20:30:07","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:19","Sunset":"16:57"}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS10 = {"StatusSNS":{"Time":"2020-11-28T20:30:07","ENERGY":{"TotalStartTime":"2020-11-28T20:19:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":6,"ApparentPower":9,"ReactivePower":7,"Factor":0.68,"Voltage":236,"Current":0.039}}}
20:30:07 RSL: stat/tasmota_1225D1/STATUS11 = {"StatusSTS":{"Time":"2020-11-28T20:30:07","Uptime":"0T00:10:06","UptimeSec":606,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"www.krikkit.net","BSSId":"DC:39:6F:10:FB:30","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
20:30:10 RSL: tele/tasmota_1225D1/STATE = {"Time":"2020-11-28T20:30:10","Uptime":"0T00:10:09","UptimeSec":609,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"www.krikkit.net","BSSId":"DC:39:6F:10:FB:30","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:30:10 RSL: tele/tasmota_1225D1/SENSOR = {"Time":"2020-11-28T20:30:10","ENERGY":{"TotalStartTime":"2020-11-28T20:19:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":6,"ApparentPower":9,"ReactivePower":7,"Factor":0.68,"Voltage":236,"Current":0.039}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 CFG: Loaded from flash at F7, Count 13
00:00:00 QPC: Count 1
[... 68 lines cut ...]
20:18:50 RSL: tele/tasmota_1225D1/STATE = {"Time":"2020-11-28T20:18:50","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"www.krikkit.net","BSSId":"DC:39:6F:10:FB:30","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:18:50 RSL: tele/tasmota_1225D1/SENSOR = {"Time":"2020-11-28T20:18:50","ENERGY":{"TotalStartTime":"2020-11-28T20:17:43","Total":8389.890,"Yesterday":0.000,"Today":8389.890,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
20:18:50 TYA: Send "55aa00000000ff"
20:18:50 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
20:18:50 TYA: Heartbeat

```

### TO REPRODUCE

reset to factory defaults, wifi config, config module as TuyaMCU, set TuyaRx and TuyaTx Gpio Right, config TuyaMCU with this:
```console
Backlog TuyaMCU 33,20; TuyaMCU 32,18; TuyaMCU 31,19;
```

### EXPECTED BEHAVIOUR
the Total kWh value should be 0 after reset

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

krikk, 2020-11-28:
updated with finding that this seems to happen also after power-cycle on 9.1.0.0

effelle, 2020-11-28:
Are you aware there is a command to reset the energy stats? 
Issue a `backlog energyreset1 0; energyreset2 0; energyreset3 0` to reset all to zero or choose what you want to reset.
```py
EnergyReset<x> | 0..42500 (for 3, Total, upper limit is 4250000) in watt-hours (Wh)
x = 1..5
1 <value> = ((p)re)set values for Today
2 <value> = ((p)re)set values for Yesterday
3 <value> = ((p)re)set values for Total
4 <standard>{,<off-peak>} = ((p)re)set tariff period values for Totals
5 <standard>{,<off-peak>} = ((p)re)set tariff period values for Exported
```

effelle, 2020-11-28:
Issuing a `backlog energyreset1 0; energyreset2 0; energyreset3 0 to reset all to zero or choose what you want to reset nromally will do the trick, but since is a Tuya MCU device you will be fine just until an update from the MCU comes in.
Have a look if you can reset your device data in nand if not . there is no way to do a reset.

ascillato2, 2020-11-29:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

krikk, 2020-11-29:
sorry, but i think you missunderstand my issue, i know the energyreset command and i did use this i did even to a complete device reset with reset 1, which should purge all data, but even after purging alle the energy data with 
```console
Backlog EnergyReset1 0; EnergyReset2 0; EnergyReset3 0; 
```
after powercycl/reboot i still have a "energy today" value of e.g. 3000 kWh (a completly wrong an unrealistic value), directly after reset and reboot... so this seemst to be an issue for me...

arendst, 2020-11-29:
I think the issue is caused by the fact that at the time Tuya reports it's active power the RealTimeClock is not in sync.

I'll change the tuyamcu driver and add a test for valid RTC before active power will be calculated.

arendst, 2020-11-29:
Give it a try.

effelle, 2020-11-29:
Could be, but on my last update I moved the `TUYA_CMD_QUERY_PRODUCT` under `FUNC_MQTT_INIT` and at this time the RTC should be already in sync. Since is a less than 24h change I doubt the OP has already tested it.
Let's call it as additional fail-safe! :) 
As a side note on next update there will be the support for external Total Energy (and SetOption72 for Tuya too).

arendst, 2020-11-29:
OK. You're right. I didn't know. At least it doesn't hurt to test for valid time anyway.

Let's see what OP comes up with.

krikk, 2020-11-30:
just did an update to current build (9.1.0.2 Timestamp 20201130 18:17) and could not reproduce the issue ...so big thanks to @arendst / @effelle for fixing this issue!

...still hoping for the Tolal Energy Support :)

effelle, 2020-11-30:
Perfect, thanks for reporting.
Closing this issue as it has been fixed.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-30 by effelle