arendst/Tasmota #10010: Enable data with more than 32 bits in xdrv_17_rcswitch.ino

joetriskaide, 2020-11-29:
**Have you looked for this feature in other issues and in the docs?**  Yes
The rc-switch library in tasmoa is capable of sending such payload , the CmndRfSend command in xdrv_17_rcswitch.ino not.

**Is your feature request related to a problem? Please describe.**  
I won a Dooya 1603 remote which uses payload data of 40 bits

**Describe the solution you'd like**  
xdrv_17_rcswitch  xdrv_17_rcswitch.ino
line 101, change to  `unsigned long long data = 0;`
line 112 change to ' `data = root.getULong(PSTR(D_JSON_RF_DATA), data);`'
**Describe alternatives you've considered**  
None

**Additional context**  
I tried the changes on gitpod and they work

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-29:
If the code size increase is not too big a provided PR will be merged

closed 2020-11-29 by arendst

nosaj66au, 2021-09-06:
Hi Guys, Been working and testing in this area it appear the fix above only addresses the transition section of the code, enabling 40 bits of data to be transmitted. Can you please also add it to the receive section.
Jason

PR #10012 (2020-11-29): Enable payload of >32 bit length
## Description:

**Related issue (if applicable):** fixes #10010

## Checklist:
  - [x ] The pull request is done against the latest dev branch
  - [x ] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x ] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [ n/a] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x ] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla) [... cut]