arendst/Tasmota #10011: Add new Protocol for Dooya 1603 433MHz remote control  in rcswitch library

joetriskaide, 2020-11-29:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
Yes, Dooya 1603 ist not supported in rcswitch library

**Describe the solution you'd like**  
Add a new protocol for Dooya 1603 to RCSwitch .cpp:
`{ 20, 0, { 0, 0 }, 1, { 239, 78 }, {20, 35 }, {35, 20}, false, 10000 } // 35 Dooya 5-Channel blinds remote DC1603`

**Describe alternatives you've considered**  
None

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**  
I tried this with the changes described in  #10010  and can confirm it works

Jason2866, 2020-11-29:
If you provide a PR there is a high chance it will be merged 😀

joetriskaide, 2020-11-29:
After implementing the code for #10010, it turns out that protocol 33 in RCSwitch.cpp works for the Dooya 1603 too. I'll do some more intense tests and may withdraw this request.

ascillato2, 2020-11-29:
> After implementing the code for #10010, it turns out that protocol 33 in RCSwitch.cpp works for the Dooya 1603 too.

Great to known that it is working now for you!

closed 2020-11-29 by ascillato2