arendst/Tasmota #10018: Thermostat Mode : Telemetry output (why power off ?)

Bibinsa, 2020-11-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Thermostat Mode : output power while TEMPTARGETSET > TEMPMEASUREDSET.
Need to know why.

**Describe the solution you'd like**  
A periodic JSON telemetry output whith main parameters of Ramp-Up/Pi strategy counters.
A "reason" why output is power off.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
Tasmota 9.1.0 with USE_THERMOSTAT

**(Please, remember to close the issue when the problem has been addressed)**  

Hi ! I'm using Thermostat feature since a fex days now.

I wonder if there is any option to activate telemetry with the different counters used (cycle reset, max time, ...) and for any reasons that trigger Power ON/OFF ?
If not, could it be possible to implement that ?

Thanks.

ascillato2, 2020-12-07:
@arijav 

Can you take a look on this? Thanks.

PoorNewb23, 2020-12-07:
That would be lovely to see.

arijav, 2020-12-08:
I take the point and will have a look

arijav, 2020-12-08:
@Bibinsa : Please check 2 topics to identify if there is really an issue

1) The PI controller do switch the output OFF even if the desired temperature is not reached. The cycle time is 30 minutes, for how long the controller outputs ON fully depends on the delta between target temp and set temp and the time you have tried to reach your set temp. This is how a PI controller works, to be able to reach the set temperature without a big overshoot.

2) After reading the first point, in case you still think you have an issue, please activate the DEBUG of the thermostat (define in your "user_config_override.h" file "#define DEBUG_THERMOSTAT", capture the serial output and please explain in detail what is the behavior you think is wrong.

Bibinsa, 2020-12-09:
@arijav Thanks for your help.
T'm trying Hybrid system for a few days but I think 8 have to tweak a little bit the driver as I'm running a Heating Pump (is that correct in english ?) with a floor heating.

I will try with DEBUG_THERMOSTAT and let you know if I find something.

arijav, 2020-12-10:
The driver is by default configure to actuate heating floor valves (as I've created it and tested it initially for this application in my apartment). For driving a heating pump you will have to check surely the application. It is important to understand and configure well the required period for heating (which depends on the time constant, long for slow changes, short for very fast reaction systems), any constraints about switching ON/OFF the heating pump and so on.

At best read the documentation of the driver in detail, the main parameters to adapt to your system are shown there, as well as the documentation of your heating pump. 

For the information I got I do not see any bug in the driver itself but more the need of adapting the parameters to your specific setup needs.

closed 2020-12-10 by ascillato2