arendst/Tasmota #10020: Add Option to Disable Unused RF Receive Protocols

BBBits, 2020-11-30:
**Have you looked for this feature in other issues and in the docs?**  
a little
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
There are over 30 Hard Coded RF protocols supported.
There can be a Lot of Decoded NOISE that gets into log and out to MQTT .
I propose a 64bit bit mask to allow only the protocols that you use be processed .
the changes in RCSwitch are basic .
In it's basic form , it would be a manual process for the user to change the value .
I.E SetOptionXYZ 0x123456789  
only 34 bits required currently

It will also save time while in interrupt for Rx Pin
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

BBBits, 2020-12-04:
Hi,
I have a working version of the idea using 2 32bit vars in settings that are tagged as being not used.
So is the idea OK or is it something I just use myself ?

Basic usage is this

Query what the Protocol Mask Is
CMD: rfrxprotocol
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x000000000"}}

Set the Protocol Mask  ( can be decimal or hex 0x12345   value )
CMD: rfrxprotocol 3
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x000000003"}}

Query If a Particular Protocol is Enabled
CMD: rfrxprotocol2
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x000000003","#2":"Enabled"}}

Disable a Protocol
CMD: rfrxprotocol2 0
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x000000001","#2":"Disabled"}}

Enable All Protocols
CMD: rfrxprotocol all
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x7FFFFFFFF"}}

List All protocols
CMD: rfrxprotocol list
MQT: stat/tasmota_378864/RESULT = {"RfRxProtocol":{"Enabled-Mask":"0x7FFFFFFFF","#1":1,"#2":1,"#3":1,"#4":1,"#5":1,"#6":1,"#7":1,"#8":1,"#9":1,"#10":1,"#11":1,"#12":1,"#13":1,"#14":1,"#15":1,"#16":1,"#17":1,"#18":1,"#19":1,"#20":1,"#21":1,"#22":1,"#23":1,"#24":1,"#25":1,"#26":1,"#27":1,"#28":1,"#29":1,"#30":1,"#31":1,"#32":1,"#33":1,"#34":1,"#35":1}}

arendst, 2020-12-04:
If implementing this pls have a look at how ``I2CDriver`` enable/disable is implemented. The goal is to use as less code as possible so your current implementation provides too much data. No need to see the mask. Just protocol numbers like:
```
CMD: rfprotocol
MQT: stat/wemos9/RESULT = {"RfProtocol":"7,9,10,11"}

CMD: rfprotocol9 0
MQT: stat/wemos9/RESULT = {"RfProtocol":"7,!9,10,11"}
```

BBBits, 2020-12-05:
I made some changes and this is the result
=====================================
I used the pow macro and was surprised with the code size
it is very hungy for an inline !
545776 mySwitch.enabled_protocol_mask.value=
pow(2,mySwitch.getNumProtos())-1;
then changed to
540912 mySwitch.enabled_protocol_mask.value= (1ULL <<
mySwitch.getNumProtos())-1
540432 my build without the feature changes

so extra is about 500

CMD: rfrxprotocol 0
MQT: stat/tasmota_D722A4/RESULT = {"RfRxProtocol":"None Enabled"}
CMD: rfrxprotocol1 1
MQT: stat/tasmota_D722A4/RESULT = {"RfRxProtocol":"1"}
CMD: rfrxprotocol 0x185
MQT: stat/tasmota_D722A4/RESULT = {"RfRxProtocol":"1,3,8,9"}
CMD: rfrxprotocol a
MQT: stat/tasmota_D722A4/RESULT =
{"RfRxProtocol":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35"}

arendst, 2020-12-05:
Great. Pls provide the PR.

ascillato2, 2020-12-07:
https://github.com/arendst/Tasmota/pull/10063

closed 2020-12-07 by ascillato2