arendst/Tasmota #10022: Wireguard support

Monacoslo, 2020-11-30:
**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
Sometimes Tasmota devices are not in home network, but they need direct connection to home network and the rest of smart home system.

**Describe the solution you'd like**  
Wireguard is becoming very popular VPN system and we need to have wireguard client installed directly on Tasmotized devices in order to be able to connect to home network on abroad locations. Home Assistant has great Wireguard integration.

**Describe alternatives you've considered**  
This option will be great alternative for complicated integrations like via CloudMQTT.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-30:
A wip PR of a leightweight port to an ESP8266 is welcome. I doubt it is possible to run Tasmota and Wireguard together.
The ESP8266 has very limited resources

ascillato2, 2020-11-30:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-30 by ascillato2

grantzvolsky, 2023-03-03:
Would it be difficult to add WireGuard support for ESP32 using this library? https://github.com/ciniml/WireGuard-ESP32-Arduino

barbudor, 2023-03-04:
May be
But I don't really see the advantage over MQTT TLS which is already supported
I'm running a site with 4 tasmota connected to a cloud based MQTT /TLS broker. My main home is also bridged to that broker so I can monitor from anywhere.

Jason2866, 2023-03-04:
Not everything possible with ESP32 is usefull to add. What's the use case where wireguard is the needed solution?

O2Graphics, 2023-06-22:
Would be great to have Wireguard support in Tasmota!
Wi-Fi isn't a secure technology, so people run a VPN (Wireguard or OpenVPN) when connecting to a Wi-Fi router.

MQTT TLS solves a problem: encrypting data between a device and an home automation system.

Wireguard solves another problem: if someone can somehow hack into a Wi-Fi router, he won't have access to anything in the network. Unless he has a Wireguard client's keys, of course.

Jason2866, 2023-06-22:
Secure your wifi. Tasmota is not secure when reachable from unwanted sources.
Adding pseudo security will not change this.

O2Graphics, 2023-06-22:
Wi-Fi isn't secure, it's crackable, and we can't do anything about this problem.
A VPN isn't pseudo security.

s-hadinger, 2023-06-22:
This topic is coming back regularly but there wasn't a convincing use case for wireguard on Tasmota. 

So you have a specific scenario? Turning on Wireguard for all devices on the wifi wlan is unheard of me. And it doesn't look at the nominal use case for wireguard

O2Graphics, 2023-06-22:
@s-hadinger, yes, I wanted to give another use case for Wireguard :-)

But the reason given by @Monacoslo in his first message is also very valid: being able to safely connect a device which is outside home. Without having to open others ports on the home network.

Currently I have this case, but I have access to the router where the device is connected, so I have been able to setup a Wireguard tunnel directly on the router.
But in the future I may not always have access to a device where I can setup such a tunnel. So having a Tasmota device able to directly connect through Wireguard would be very useful! (example: a Tasmota device running outside from a solar panel and connected to a free Wi-Fi).

Monacoslo, 2023-06-22:
Actually very interesting use case. So like having Wireguard router already built in tasmota device :)

O2Graphics, 2023-06-22:
I think it's a bit outside of the Tasmota project. :-)

I don't know if there are ready made projects like this, but you could code one by looking at some examples (by searching "esp32 access point", "esp32 gateway", "esp32 wireguard client",…)

I'm not sure if an ESP32 can have two differents Wi-Fi at the same time: one to be used as an access point for Tasmota devices and the others to connect to the modem of your ISP.
If it's not possible, there are some ESP32 with a RJ45 connector and I guess it should be possible this way.

Or maybe just a single ESP32 which wouldn't act as an access point, but would connect to your ISP's Wi-Fi. The Tasmota device would also connect to the ISP's Wi-Fi and would use the ESP32 as a Wireguard client.

PS: Those are just random ideas. I don't know a lot about ESP32, I just made a Wi-Fi access point years ago with a web server showing some webpages to clients (and some hardware attached to the ESP32 like keypad, display, servos,…).

s-hadinger, 2023-06-22:
The main problem is that those scenarios require to connect to public wifi and the public internet. DON'T DO THAT!

Tasmota was never designed nor hardened to be internet facing. The security level of esp32 is not at the right level for such scenario. 

It's even worse. If you esp32 device gets hacked, the wireguard tunnel gives to the attacker a direct access to your remote network. Not good...

btsimonh, 2024-01-07:
basic (even unencrypted) VPN support may be a good feature - e.g. to mount an esp32 device in a location where it wants to send to a device inside a VPN which has no public internet presence, and you don't have a device on the same local LAN which could be a bridge.  (e.g. I have a house in spain, but would want to deliver data - images - to my RPi in england behind a firewall.).
But, there are far more improtant featus for TAS to fucus on...
Personally, I'd setup an OpenWRT router to provide a wireless AP exposing the VPN, and routing to internet as required.  OpenWRT routers are available for <£5 for this kind of application.  This also then isolates your IOT network from your main LAN.

g6094199, 2025-01-02:
> May be But I don't really see the advantage over MQTT TLS which is already supported I'm running a site with 4 tasmota connected to a cloud based MQTT /TLS broker. My main home is also bridged to that broker so I can monitor from anywhere.

mqtt tls is NOT an option if you have already a runing network with mqtt and wireguard. never change a running environment.

e.g.
why fiddle around with a pain in the arse implementation of mosquitto (in HA) to run tls when you already have stable running wireguard endpoint(s). 
also opening another port on WAN, even its tls, is also often NO option when used in corporate environments and shouldn´t also on private ones.

s-hadinger, 2025-04-27:
Wireguard is now available: https://github.com/arendst/Tasmota/pull/23347 and https://tasmota.github.io/docs/Wireguard/

g6094199, 2026-10-26:
fyi

wiregurad NEEDS to go with a network watchdog if the endpoint is a dns and it  resides on a dynamic ip. wireguard  only resolves the name once. it doesnt track ip changes.  so you need  to track the endpoint via a watchguard and restart the wiregurad service or simply the device on a endpoints ip change.

this info also should go to the wiki.

s-hadinger, 2026-10-27:
@g6094199 Good point, I have actually patched WIreguard to supprot re-resolving of DNS

https://github.com/arendst/Tasmota/pull/25204