arendst/Tasmota #10030: H801 blink command Power2 3 is not working

Rothammel, 2020-12-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have tried the blink command.
Power1 3 is working fine. the red channel is now blinking, but Power2 3 gives me an error:

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): H801
- [x] Tasmota binary firmware version number used: Tasmota 9.1.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer initial then OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:40:19 MQT: stat/eingang/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:40:19 MQT: stat/eingang/RESULT = {"Module":{"20":"H801"}}
10:40:19 MQT: stat/eingang/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"288":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"420":"PWM5"},"GPIO5":{"321":"Led_i2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"419":"PWM4"},"GPIO15":{"416":"PWM1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:41:50 MQT: stat/eingang/STATUS = {"Status":{"Module":20,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"eingang","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:41:50 MQT: stat/eingang/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:26","StartupUTC":"2020-12-01T09:29:24","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-11-30T18:38:13","SaveCount":130,"SaveAddress":"FB000"}}
10:41:50 MQT: stat/eingang/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020.11.30 18:52:51","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"341/699"}}
10:41:50 MQT: stat/eingang/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000001","00006000","00000000"]}}
10:41:50 MQT: stat/eingang/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:41:50 MQT: stat/eingang/STATUS5 = {"StatusNET":{"Hostname":"eingang-5536","IPAddress":"192.168.0.213","Gateway":"192.168.0.4","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.2","Mac":"3C:71:BF:31:95:A0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:41:50 MQT: stat/eingang/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3195A0","MqttUser":"Stan","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:41:50 MQT: stat/eingang/STATUS7 = {"StatusTIM":{"UTC":"2020-12-01T09:41:50","Local":"2020-12-01T10:41:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
10:41:50 MQT: stat/eingang/STATUS10 = {"StatusSNS":{"Time":"2020-12-01T10:41:50"}}
10:41:50 MQT: stat/eingang/STATUS11 = {"StatusSTS":{"Time":"2020-12-01T10:41:50","Uptime":"0T00:12:26","UptimeSec":746,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":38,"MqttCount":2,"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":153,"Channel":[100,0,0,0,0],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WiFi","BSSId":"A4:71:74:06:57:C3","Channel":5,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
-> console -> Power2 3 -> Enter

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-01:
Please, provide the output of the Console log output when you experience your issue; if applicable:
(Please use weblog 4 for more debug information)

Rothammel, 2020-12-01:
> BlinkCount

23:29:36 CMD: Gruppe 0, Index 1, Befehl "BLINKCOUNT", Daten ""
23:29:36 MQT: stat/eingang/RESULT = {"BlinkCount":0}

> BlinkTime

23:30:37 CMD: Gruppe 0, Index 1, Befehl "BLINKTIME", Daten ""
23:30:37 MQT: stat/eingang/RESULT = {"BlinkTime":40}

ascillato, 2020-12-02:
Hi,

Thanks you very much for completing the template and explaining your issue. :+1: 

What is happening is not a bug. Tasmota thinks that your device is just 1 light that have 5 PWM channels, so the POWER command will work only when using `POWER1`. 

If you set your color to blue, for example, and then you perform `POWER1 3`, you will see the **blink** on blue and not on red. Red seems that was your last color before using the blink feature.

The blink feature is just for the whole device of several channels, or for relays.

To behave as you expects, you have to tell Tasmota to think that every channel is independent (like if you have 3 lights instead of 1 device).

So, if you go to the console and type `setoption68 1`, your device will restart and you will be able to perform `power2 3` without any issue.

If you want more information about this, please check https://tasmota.github.io/docs/Commands/#setoption68

Thanks.

ascillato2, 2020-12-02:
Closing this issue as it has been answered.

Happy holidays and lights preparations ;)

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-02 by ascillato2

Rothammel, 2020-12-02:
thank you, I think a reference to the SetOption68 option should be added to the wiki