arendst/Tasmota #10032: Device Groups not working with FritzBox 6660 (possible multicast problem)

thomasschuette, 2020-12-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Device Groups don't work using at least a Fritzbox 6660 as Wifi router.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Anything ESP8266 powered (Sonoff switches, ESP-01, ...)
- [ ] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA flash
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:01:06 CMD: Backlog Template; Module; GPIO 255
18:01:06 MQT: stat/esp0102/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:01:07 MQT: stat/esp0102/RESULT = {"Module":{"0":"Generic"}}
18:01:07 MQT: stat/esp0102/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"224":"Relay1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:04:02 CMD: status 0
18:04:02 MQT: stat/esp0102/STATUS = {"Status":{"Module":0,"DeviceName":"ESP-Test","FriendlyName":["ESP-Test"],"Topic":"esp0102","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:04:02 MQT: stat/esp0102/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:15","StartupUTC":"2020-12-01T17:03:47","Sleep":100,"CfgHolder":4617,"BootCount":55,"BCResetTime":"2020-11-30T17:42:01","SaveCount":269,"SaveAddress":"FA000"}}
18:04:02 MQT: stat/esp0102/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020.11.07 12:04:01","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"415/699"}}
18:04:02 MQT: stat/esp0102/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hawaii50",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000008","00006008","00000000"]}}
18:04:02 MQT: stat/esp0102/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
18:04:02 MQT: stat/esp0102/STATUS5 = {"StatusNET":{"Hostname":"Test01","IPAddress":"192.168.178.171","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"EC:FA:BC:C1:3D:C4","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
18:04:02 MQT: stat/esp0102/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.225","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C13DC4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:04:02 MQT: stat/esp0102/STATUS7 = {"StatusTIM":{"UTC":"2020-12-01T17:04:02","Local":"2020-12-01T18:04:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
18:04:02 MQT: stat/esp0102/STATUS10 = {"StatusSNS":{"Time":"2020-12-01T18:04:02"}}
18:04:02 MQT: stat/esp0102/STATUS11 = {"StatusSTS":{"Time":"2020-12-01T18:04:02","Uptime":"0T00:00:15","UptimeSec":15,"Heap":24,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Hawaii50","BSSId":"2C:91:AB:90:13:3E","Channel":9,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash at least two ESP8266 devices with Tasmota 9.1.0 connected to a Wifi network operated by a Fritzbox 6660 and try to set up a device group following the instructions found here https://tasmota.github.io/docs/Device-Groups/

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Working device groups out of the box.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I've asked about the issue in discord #general but got no response. Original message as follows:
--
Hi all, I'm tinkering with device groups since yesterday but to no avail. I followed the steps at https://tasmota.github.io/docs/Device-Groups/ for troubleshooting but still no luck. As I already started to suspect, that maybe my router (Fritzbox 6660) somehow filters the multicast packets out, I tried my luck with an ancient Linksys WRT45GL and yepp, it's working there. Switching back to the FB - I'm back at the non functioning behavior. Has someone any idea what to do about the FB? The Live TV optimization I've already checked and unchecked, no success.
--
In the meanwhile I got a bit further. Examining a support log of the Fritzbox I found following promising looking lines:
--
multid mrouter routes:
-----------------------------------------------
239.255.255.250: 192.168.178.1 (lan) => NONE
--
Without finding anything more about this multicastish looking IP (netmasks, etc.) I compiled a binary with setting #define DEVICE_GROUPS_ADDRESS 239,255,255,250 instead of the default 239.255.250.250 and flashed it to two devices. The device groups started working instantly after flashing and rebooting my test devices. One more attempt with changing only the host portion of the IP to 239.255.255.245 reverted erverything back to the non functional state.
So, yes, I already found my solution for this issue but I think that's not the most straight forward option, especially for those who barely manage to tasmotize their device and get it to do something usefuly with it. Asking AVM to change their firmware to a somewhat more multicast friendly routing is most probably not leading to anywhere I think (though, I didn't talk to them about this for the time being), I have not the slightest idea about any implications the use of this FB mulitcast address might have with other applications and I can't cross check with a different FB model as I don't own one.

Is there maybe a way to give users a run-time control of the DEVICE_GROUPS_ADDRESS / PORT without the need compile own binaries?
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-01:
Sorry we cant help. Tasmota Device Groups needs routing of UDP packets. 
It is not a Tasmota issue. You have to change settings in your Fritzbox to enable UDP routing.

Jason2866, 2020-12-01:
Using 239.255.255.250 does conflict with Alexa (Tasmota HueEmulation) and can not be used.
Probably the reason why AVM added this entry. Without the Alexa devices would not work.

ascillato, 2020-12-01:
> Using 239.255.255.250 does conflict with Alexa

So, based on that information, Device Groups must change its IP to 239.255.255.245 permanently. This is going to be a breaking change but IMHO it is needed. In that case, there is no need to add a new command to change "on the fly" the Multicast IP (that will use flash, ram and eeprom settings and will be used only once while the user is configuring the device).

pcdiem, 2020-12-01:
According to thomasschuette, "One more attempt with changing only the host portion of the IP to 239.255.255.245 reverted erverything back to the non functional state." So, changing it to 239.255.255.245 won't help with his router.

Jason2866, 2020-12-01:
Do i overlook something? At the moment we have a non conflicting setting
```
#define USE_DEVICE_GROUPS                        // Add support for device groups (+5k5 code)
  #define DEVICE_GROUPS_ADDRESS 239,255,250,250  // Device groups multicast address
  #define DEVICE_GROUPS_PORT 4447                // Device groups multicast port
```

ascillato, 2020-12-01:
Ok, so,

Alexa uses 239.255.255.250
DevGroup uses 239.255.25**0**.250

There is no conflict on that.

@thomasschuette

Are you sure that you are using the same IP Multicast than Alexa?

BBBits, 2020-12-02:
ok , so it works with another router so there are no obvious errors in the tasmota multicast registration.
perhaps the fritzbox has a bug and only stores the first registered IP ?
probably a tiresome exercise but , if you could disable all alexa stuff and reboot fritz , 
wonder if the devgroup ip ends up in the   multid mrouter routes: ...

thomasschuette, 2020-12-02:
Thanks for all your input! Just for clarification:

Default 239.255.250.250 - does not work for me,
compiled with 239.255.255.250 (from FBs debug data) - works
compiled with 239.255.255.245 (random host ip but in the same subnet) - does not work.

@ascillato

> Are you sure that you are using the same IP Multicast than Alexa and not the default in Tasmota?

Yes, I am sure. I tried forth and back and with several devices. Default 9.1.0 does not work. Compiled with the "Alexa multicast IP" it works. The settings of the devices are always the same.
Up to now I haven't seen any negative impact by using the "Alexa IP." The devices work as usual (a Sonof 3-gang wall switch and three Avatar AWP08L Plugs), device groups work, Alexa works, Alexa control works as usual with all devices (I haven't tried a device discovery yet).

@BBBits

> probably a tiresome exercise but , if you could disable all alexa stuff and reboot fritz

No problem, I'm going to try that when I'm back home later today and come back with the result.

Rummaging deeper through the FB debug logs, I found a couple of entries with the Alexa multicast IP seemingly associated with an IP out of my internal network but which is not used by any device (192.168.178.20). That seems a bit funky - but maybe just because of my limited knowledge. Also I tried to set a static route for the default multicast IP, that fails because I somehow can't use the router IP (192.168.178.1) as a gateway. Not sure, if that is a valid approach at all.

I'll post some excerpts from the debugs logs later today. I might also be able to test everything on a different FB model (DSL version, I'm on cable internet myself) but not sure if I managed that by today.

Would be a bummer if the device groups won't work with FBs, as at least for cable internet, there are really not much other choices in Germany than AVM :'‑( Pure modems are only available for business clients afaik.

Jason2866, 2020-12-02:
Alexa device discovery is the critical part. There is a high chance this functions gets "disturbed" by device groups activity.
@pcdiem  What changes are needed (code size increase?) to do a Setoption to change the Multicast IP address to the Alexa one?
I know it would be a ugly workaround. But since AVM Fritzbox is the most used router in Germany it does not hit only some users...

thomasschuette, 2020-12-02:
@Jason2866 
I'll give the discovery a try this evening and report back. Any special scenario for that? E.g. usually an Alexa discovery run takes about 30 to 45 seconds. Does it help to switch something in a device group on and off every second during the discovery and monitor if things are messing up? For something more sophisticated (wireshark) I fear I don't have enough time atm.

pcdiem, 2020-12-02:
There are no conflicts when device groups uses the same multicast IP address as Alexa (239.255.255.250, which is actually the UPnP discovery multicast address). In fact, device groups originally used the Tasmota emulation code for the UDP packet reception. The decision was later made to move device groups to a different multicast IP because users didn't want to have to allow all the UPnP packets in their networks.

Adding support to allow configuring the device groups multicast IP address isn't major. However, adding support for allowing the user to specify the device groups multicast IP as 239.255.255.250 is a bigger deal because that would require it to switch to using the existing emulation code if it's compiled in and enabled.

I believe the most efficient approach to allow 239.255.255.250 to be used as the device groups multicast IP would be to have a SetOption to enable it. That way it only requires one setting bit and doesn't require the code to support a new command.

thomasschuette, 2020-12-02:
Ok, so, I just finished some testing.

@BBBits 
Did throw Alexa out of the house, power cycled the router, tried device groups with stock 9.1.0 on two Sonof Basic RF with device groups enabled and checked - didn't work. Also no change in the FBs debug logs about the multid mrouter routes.

@Jason2866
Used two of my plugs with self compiled firmware with 239.255.255.250, put them in an endles loop to switch on and off every second and started Alexas device discovery. The relays were clicking happily all the time, nothing obvious in the console output and Alexa just did her job, found the two new devices and let me control them by voice.

One last despaired idea was to set a static route in the FB to net 239.255.250.250 mask 255.255.255.0 gw 192.168.178.20 (just a try...) or gw 192.168.178.0 - no luck either.

The last thing left to do is trying the stock 9.1.0 with a different FB - maybe tomorrow. If that works, I'll go, find a corner to cry silently and raise a ticket with AVM. If not, I guess the SetOption would be a great idea.

Entries from the debug log:

```
##### BEGIN SECTION multicast Multicast

Multicast
-----------------

/proc/net/ip_mr_vif:
-----------------------------------------------
Interface      BytesIn  PktsIn  BytesOut PktsOut Flags Local    Remote
 0 lan               0       0         0       0 00008 00000020 7FC40444
 1 guest             0       0         0       0 00008 00000021 7FC40444
 2 dsl               0       0         0       0 00008 00000023 738ED83C

/proc/net/ip_mr_cache:
-----------------------------------------------
Group    Origin   Iif     Pkts    Bytes    Wrong Oifs
FAFFFFEF 14B2A8C0 -1         0        0        0
-----------------

multid mrouter global:
-----------------------------------------------
Upstream                  : dsl
IGMP tos                  : c0
Robustness                : 2
Query Interval            : 125
Querier Check Interval    : 300
Query Response Interval   : 10
Group Interval            : 260
Last Member Query Interval: 1
Upstream IGMP Version     : 3
Homenet IGMP Version      : 3
Enforce SSM               : yes
Fast IGMP                 : yes

multid mrouter upstream:
-----------------------------------------------
dsl              VIF 2 IGMP Version 3 IfIndex 35 upstream
                 192.168.178.1 192.168.178.1

multid mrouter routes:
-----------------------------------------------
239.255.255.250: 192.168.178.1 (lan) => NONE

multid mrouter vifs:
-----------------------------------------------
lan              VIF 0 IGMP Version 3 IfIndex 32 querier
                 192.168.178.1/255.255.255.0
guest            VIF 1 IGMP Version 3 IfIndex 33 querier
                 192.168.189.1/255.255.255.0
dsl              VIF 2 IGMP Version 3 IfIndex 35 upstream
                 192.168.178.1 192.168.178.1

/proc/kdsld/netiface/dsl/igmpcompr/groups:
-----------------------------------------------

bridge mdb:
-----------------------------------------------
dev lan port vap0 grp 239.255.255.250 temp 
dev lan port vap0 grp 239.255.255.250 temp 
dev lan port vap0 grp 239.255.255.250 temp 
dev lan port vap0 grp 239.255.255.250 temp 
dev lan port vap0 grp 239.255.255.250 temp 
dev lan port eth1 grp 239.255.255.250 temp 
dev lan port vap1 grp ff02::fb temp 
dev lan port vap1 grp ff02::fb temp 
dev lan port vap1 grp ff02::fb temp 
dev lan port vap1 grp ff02::fb temp 
##### END SECTION multicast
```
And 
`IPv4 Hdr       : 192.168.178.20 239.255.255.250 proto 17 tos 00
`
multiple times

thomasschuette, 2020-12-06:
Well, today I had the chance to test two ESP-01 with stock 9.1.0 on a FB 7590 (DSL Version) and yep, there it works.

The same setup on my 6660 does not work. Except the difference between DSL and cable, everything is basically the same but the network. I'm running on 192.168.178.x, the 7590 on 192.168.2.x
The multicast section in the 7590 does not look diferent to my 6660.

I'm out of ideas. Guess it's time to find that silent corner, raise a ticket with AVM and start life with self compiled images (which isn't that bad - but I have to confess, i'm kind of lazy sometimes).

So, since now most probably the percentage of impacted routers dropped significantly: The work for a setoption might be to much - but maybe an entry in the docs might be helpful for others?

ascillato, 2020-12-06:
> but maybe an entry in the docs might be helpful for others?

Agree, that is the best solution for now. Please, help us and add that to the docs. Thanks.

closed 2020-12-07 by ascillato2

BBBits, 2020-12-07:
Perhaps another option that would be workable might be a flag to enable Broadcast instead of multicast.

when you look at the amount of garbage on most networks , a few broadcasts from tasmota is not going to be an issue.
Much easier to debug as well !!

marcohald, 2024-10-27:
I know this is an rather old and closed issue, but still a top match on my google search.
I had the same Problem that the Messages send are not received by the other Device.
The soloution was turning on "WLAN-Übertragung für Live TV optimieren" / "Optimize Wi-Fi transmission for live TV".
Point 3 in this article https://en.avm.de/service/knowledge-base/dok/FRITZ-Box-7590/592_Reception-interference-when-watching-TV-over-IPTV-connection/
[image]