arendst/Tasmota #10033: Shutterposition problem on newer versions

yinonco, 2020-12-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
**ISSUE 1:** Iv'e used two new shelly 2.5 as a rollershutter controller with tasmota.
on versions 9.1.0 and 9.1.0.2 (newest dev) when I am opening the shutter to 30% and then to 80% the shutter opens to about 100cm. But, when I open it in stages - from 30 to 50 to 70 to 80 it opens much **MORE** (almost all the way - about 127cm).
then of course when I try to close the shutter again it wont close all the way because it thinks it is only opened to 80 while it is actually more than that.
To make sure nothing which I configured making that happen Iv'e re-flashed the device and didnt configured anything except than:
```
SO80 1
shutterrelay1 1
interlock 1,2
interlock 1
shutteropenduration 19.5
shuttercloseduration 18.7
```
and it is still the same.
than I tried to flash version 8.5.0 and looks like it works just fine!

Note that the problem is that the shutter is opening more than it should and not less - therefore the problem is not motor delay 

**ISSUE 2:** Another problem is that I get hardware watchdog restarts - even when I removed all the sensors from the device template.

Thanks.
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [X] Tasmota binary firmware version number used: 9.1.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:36:04 CMD: Backlog Template; Module; GPIO 255
21:36:04 RSL: stat/tasmota_1A32AE/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
21:36:04 RSL: stat/tasmota_1A32AE/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
21:36:04 RSL: stat/tasmota_1A32AE/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

No rules used
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:36:51 CMD: Status 0
21:36:51 RSL: stat/tasmota_1A32AE/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_1A32AE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T00:32:33","StartupUTC":"2020-12-01T20:04:18","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-12-01T20:42:30","SaveCount":89,"SaveAddress":"F4000"}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-01T15:09:36","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"409/699"}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cohen_HB",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1A32AE-4782","IPAddress":"192.168.1.69","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"D8:BF:C0:1A:32:AE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1A32AE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS7 = {"StatusTIM":{"UTC":"2020-12-01T20:36:51","Local":"2020-12-01T21:36:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS10 = {"StatusSNS":{"Time":"2020-12-01T21:36:51","ANALOG":{"Temperature":57.1},"ENERGY":{"TotalStartTime":"2020-12-01T20:42:30","Total":0.007,"Yesterday":0.000,"Today":0.007,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":80,"Direction":0,"Target":80},"TempUnit":"C"}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS11 = {"StatusSTS":{"Time":"2020-12-01T21:36:51","Uptime":"0T00:32:33","UptimeSec":1953,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Cohen_HB","BSSId":"98:1E:19:7F:3C:1E","Channel":1,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:36:51 RSL: stat/tasmota_1A32AE/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":195,"Close":187,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
explained above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
shold open to the same opsition +-

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-02:
Thread continues at https://github.com/arendst/Tasmota/issues/9906

closed 2020-12-02 by ascillato2

yinonco, 2020-12-02:
Hi, I saw issue #9906 but as far as I understand it doesnt describes the same issue

stefanbode, 2020-12-14:
I agree that there is an issue with the stop command. The 0,5sek was quite well observed. I fixed it and submitted an update. The reason for the delay is that you cannot change directly from up to down. there is 0.5sec minimum stop. But the delay was at the wrong position in the code

stefanbode, 2020-12-15:
I checked the duration of sending the stop until it actually stops. this is 0.2 sek or 4 TASMOTA cycles (50ms). There is a potential delay if the MQTT server responds slow. I try to find a way to avoid sending data to the MQTT before the relay is stopped to minimize time between stop command and the real stop. Anyhow currently the position is correctly calculated as far as I can see with the latest fix.

Bascht74, 2020-12-15:
@stefanbode Thanks for the fix, I was observing this behavior in #10112, too. I will check whether that issue there is eliminated as well.

yinonco, 2020-12-16:
@stefanbode Hi I've updated to the newest dev and the position looks ok now. But, there is a new problem with that version (maybe related to the fix), now when I do `shutteropen `and then `shutterstop `- most of the times the shutter is stopping and then closing just a bit (and the opposite for `shutterclose`)

Bascht74, 2020-12-16:
@stefanbode This change solved my problem, too. But there is one glitch left. I will make a new issue out of it, as it has nothing to do with this problem here. btw: Thanks for your great work with the shutters! The shelly implementation was pain in the ass (no updates while moving, no halfway-correction for german shutters, no shutter commands for a remote shelly-button switch (close/stop/open/stop) etc...) Glad that I switched over...
Sebastian