arendst/Tasmota #10034: Include time in JSON for ZigBee sensor data via MQTT

alexhk, 2020-12-01:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
When publishing sensor data via MQTT it makes  sense to include the time of the reading in the JSON. Tasmota does this with the standard sensor data (see example below), but with Zigbee2Tasmota we are missing the time when sensor data is published to MQTT.

**Describe the solution you'd like**  

I'd like to see the timestamp included as "Time" in the JSON, just like the other sensors in Tasmota.

**Describe alternatives you've considered**  

I considered using the time when the MQTT message was received, but not all MQTT clients have this option (ie. Mosquitto doesn't support to [display a timestamp](https://szunyog.github.io/iot/mqtt/oneliners/mosquitto_sub-with-timestamp) of the event as far as I can tell).

In my specific example, I am setting up an MQTT consumer in Telegraf to get data into InfluxDB, which should be able to use the time of receiving the data. But The reading of the sensor might actually be older under certain circumstances, so the best would still be to include the Time with the sensor JSON directly from Zigbee2Tasmota .

**Additional context**  

Standard/Tasmota sensors published via MQTT:

```
tele/Sonoff-TH16-3/SENSOR

{
  **"Time": "2020-12-01T21:28:15"**,
  "DS18B20-1": {
    "Id": "01192128C21C",
    "Temperature": 18.1
  },
  "DS18B20-2": {
    "Id": "011921329BAC",
    "Temperature": 18.1
  },
  "TempUnit": "C"
}
```

Zigbee2Tasmota sensors published via MQTT:

```
tele/AqaraTH2/SENSOR

{
  "ZbReceived": {
    "AqaraTH2": {
      "Device": "0xE680",
      "Name": "AqaraTH2",
      "BatteryVoltage": 2.99,
      "BatteryPercentage": 92,
      "Temperature": 22.14,
      "Humidity": 61.14,
      "Pressure": 971,
      "SeaPressure": 970,
      "Endpoint": 1,
      "LinkQuality": 107
    }
  }
}
```

s-hadinger, 2020-12-15:
Sorry but I don't see why it is useful to add the date. Zigbee JSON message are already verbose enough, they will become very long.

closed 2020-12-15 by ascillato2

weeman1337, 2022-08-06:
> Sorry but I don't see why it is useful to add the date

It can be useful to know the date the sensor data appeared.

s-hadinger, 2022-08-09:
It is done now

sfromis, 2022-08-09:
Does `SetOption144` being on conflict with other setoptions? I'm trimming the payloads to avoid the `ZbReceived` object level (and using the ZbName as key). Testing with `SetOption144` gave unchanged payloads like 
`xxxxxxxx/tele/SENSOR = {"Ikea-Opal":{"Device":"0x665A","Name":"Ikea-Opal","Power":0` ....
Using precompiled `"Version":"12.0.2.4(zbbridge)","BuildDateTime":"2022-08-09T16:50:34"`

s-hadinger, 2022-08-10:
Precisely, the timestamp can only be added if you have `ZbReceived`, otherwise `SetOption144 1` has no effect, as it would break the structure of the returned JSON.

PR #16187 (2022-08-09): Zigbee ``SetOption144 1`` includes a timestamp in `ZbReceived` messages
## Description:

`SetOption144 1` adds a timestamp in all `ZbReceived` message. Note that the timestamp is not added if `ZbReceived` is not present (depending on options and topic used).

Example:

With `SetOption144 1`

```
17:24:49.597 RSL: SENSOR = {"Time":"2022-08-09T17:24:49","ZbReceived":{"0x4846":{"Device":"0x4846","Name":"SNZB-02","Humidity":77.69,"BatteryPercentage":60,"Endpoint":1,"LinkQuality":196}}}
```

With `SetOption144 0``
```
17:26:19.748 RSL: SENSOR = {"ZbReceived":{"0x4846":{"Device":"0x4846","Name":"SNZ [... cut]