arendst/Tasmota #10035: Sonoff S31 Energy Values all 0

hepcat72, 2020-12-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

All energy values are 0. They have never shown any value greater than 0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S31
- [x] Tasmota binary firmware version number used: 2.7.4.5
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: https://github.com/themadinventor/esptool.git
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

04:09:41 CMD: Backlog Template; Module; GPIO 255
04:09:42 RSL: stat/tasmota_B7F899/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
04:09:42 RSL: stat/tasmota_B7F899/RESULT = {"Module":{"41":"Sonoff S31"}}
04:09:42 RSL: stat/tasmota_B7F899/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

I started with rules, but reset the device, so there are currently no rules
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

04:11:09 CMD: Status 0
04:11:09 RSL: stat/tasmota_B7F899/STATUS = {"Status":{"Module":41,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B7F899","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:24:17","StartupUTC":"2020-12-02T02:46:52","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-12-02T03:17:42","SaveCount":18,"SaveAddress":"FA000"}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-01T15:09:36","Boot":7,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"348/699"}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Rob's Network",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS5 = {"StatusNET":{"Hostname":"tasmota_B7F899-6297","IPAddress":"10.0.1.44","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"A4:CF:12:B7:F8:99","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B7F899","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS7 = {"StatusTIM":{"UTC":"2020-12-02T03:11:09","Local":"2020-12-02T04:11:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:24","Sunset":"16:54"}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS10 = {"StatusSNS":{"Time":"2020-12-02T04:11:09","ENERGY":{"TotalStartTime":"2020-12-02T03:17:42","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
04:11:09 RSL: stat/tasmota_B7F899/STATUS11 = {"StatusSTS":{"Time":"2020-12-02T04:11:09","Uptime":"0T00:24:17","UptimeSec":1457,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Rob's Network","BSSId":"64:A5:C3:6D:C6:EA","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

00:00:00 CFG: Loaded from flash at FA, Count 18
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.1.0.2(tasmota)-2_7_4_7(2020-12-01T15:09:36)
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 Rob's Network Channel 11 BSSId 64:A5:C3:6D:C6:EA in mode 11N as tasmota_B7F899-6297...
00:00:02 WIF: Connected
00:00:03 HTP: Web server active on tasmota_B7F899-6297 with IP address 10.0.1.44
04:12:57 RSL: tele/tasmota_B7F899/INFO1 = {"Module":"Sonoff S31","Version":"9.1.0.2(tasmota)","FallbackTopic":"cmnd/DVES_B7F899_fb/","GroupTopic":"cmnd/tasmotas/"}
04:12:57 RSL: tele/tasmota_B7F899/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_B7F899-6297","IPAddress":"10.0.1.44"}
04:12:57 RSL: tele/tasmota_B7F899/INFO3 = {"RestartReason":"Software/System restart"}
04:12:57 RSL: stat/tasmota_B7F899/RESULT = {"POWER":"ON"}
04:12:57 RSL: stat/tasmota_B7F899/POWER = ON
04:12:59 QPC: Reset
[... 85 lines cut ...]
~121.50 volts

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Since this is my first time flashing anything, let alone tasmota, there's a chance I did something wrong, but I've been working with IoT devices for years.  I'm a scientific programmer and I'm comfortable on the command line.  I saved my entire terminal session on my mac if that would be useful.  Just ask.

It took me over 2 hours to flash the sonoff, only because I had shaky hands and the precision work was really difficult.  I'm not looking forward to soldering the second one.

There's also a chance that the outlet is a dud.  I haven't taken the second one out of the shrink wrap, so I can try that one with the eWeLink app to see if it works.  I never even set up the first outlet with eWeLink.  I didn't even have an account until after this problem presented itself.

I appreciate your issue form.  I did search all of those sites and read through what looked like the same issue that appeared to never have been resolved and found similar, but not the same issues.  This whole experience has been really frustrating.

I followed essentially [the youtube from your doc](https://www.youtube.com/watch?v=kKtLKjI4wA0), aside from the flashing software, since it was windows specific.  Oh - also, I did not go through the MQTT stuff since I intend to use a GET webhook to my node red instance.

I chatted with @sfromis on discord, but I think he signed off before I encountered this issue.  I also tried doing a calibration, but nothing came of it.

Oh - not sure if this is relevant.  My first attempt to flash, I had the wires mixed up.  Could that have fried it?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-02:
Hi,

Thanks for completing the template and sharing all the info. Very appreciated.

Your issue is that you don't have a Sonoff S31, you have a Sonoff S31 Lite. The "LITE" version is a cheaper version that don't have the power monitoring chip. So, your flash procedure, your steps and your device are all working as expected. The only thing is that your device don't have that feature. Sorry.

ascillato, 2020-12-02:
The Sonoff S31 (with power monitoring) is this: https://www.amazon.com/Cigovd-SmartPlug-Monitoring-Upgraded-Intelligent/dp/B081DH47BQ/ref=sr_1_27?dchild=1&keywords=sonoff+s31&qid=1606883695&sr=8-27

ascillato2, 2020-12-02:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-02 by ascillato2

hepcat72, 2020-12-02:
I'm confused. The box it came in says it has power monitoring. The link I used to buy it was the one linked from the YouTube video linked in your doc. How do you know it's the LITE version?

hepcat72, 2020-12-02:
Well at least I didn't ruin it. I guess it's somewhat of a relief that the one you linked doesn't arrive until January. I was not looking forward to soldering the second outlet. Do you know if there's a way to flash OTA from a Mac?

sfromis, 2020-12-02:
The Amazon link you posted says "Sonoff S31 Lite Wi-Fi Smart Plug".

hepcat72, 2020-12-02:
Yeah, I saw that. I swear it said energy monitoring when I ordered it, but I noticed that the link in the youtube tutorial currently goes to a "not found" page on Amazon, so I clearly must have searched for it.  Thinking back, I now recall having searched for awhile and only finding one amazon product listing for the S31. And if you search for it now, you get a few S31's that don't mention "lite" - but I don't trust them, and you don't ever get anything that says "16A" in the description/product name, so I appreciate you guys providing a direct link.  I'm bummed that it's currently out of stock and won't arrive until mid January.  So long to wait... I searched again and still could not find a 16A version.

And I'm glad I didn't waste your time on discord @sfromis chasing my wild goose. :)

In the meantime - any tips on OTA via a mac would be greatly appreciated.  I looked at several blog posts about TUYA and didn't seem to be able to figure it out, mac-wise.

hepcat72, 2020-12-02:
I ordered 3 this time.

meingraham, 2020-12-02:
No OTA method to replace the factory firmware with Tasmota is possible for the "older" generations of Sonoff devices. Some of their newer "DIY" devices have provided an OTA capability... but there have been many glitches which have prevented many users from using Sonoff's mechanism to successfully flash custom firmware... and have ended up resorting to the tried & true wired flashing method. All of this to say that for the S31, the only way to flash it is to carefully solder onto those edge contacts.