arendst/Tasmota #10039: Support for Tasmota Zigbee Aqara Xiaomi WAll switch QBKG11LM

tyjtyj, 2020-12-02:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Support for QBKG11LM
https://www.zigbee2mqtt.io/devices/QBKG11LM.html
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

```
"ModelId":"lumi.plug"
00:46:24 MQT: tele/mi_plug1/SENSOR = {"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1","0600/0000":1,"ActivePower":0,"EnergyTotal":31,"Endpoint":1,"LinkQuality":128}} (retained)
```
0600/0000 : Power

```
"ModelId":"lumi.ctrl_ln1.aq1" QBKG11LM
00:42:51 MQT: tele/QBKG11LM/SENSOR = {"QBKG11LM":{"Device":"0xXXXX","Name":"QBKG11LM","Xiaomi_64":0,"Xiaomi_98":0,"Xiaomi_95":4.9,"Xiaomi_9A":16,"Endpoint":1,"LinkQuality":131}} (retained)
```

Xiaomi_64: Power State(On/Off)
Xiaomi_98: ActivePower
Xiaomi_95: EnergyTotal

**Additional context**  
_Add any other context or screenshots about the feature request here._
Similar to https://github.com/arendst/Tasmota/issues/9941

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-12-02:
Will take a look. Can you share the result of `ZbStatus2 QBKG11LM`

s-hadinger, 2020-12-02:
Sorry, I also need the logs using `WebLog 3` level when you receive the message.

tyjtyj, 2020-12-03:
Here is weblog 3

QBKG11LM and miplug should have temperature sensor too but I dont see any value showing it. 

QBKG11LM
zbstatus2 

```
20:33:57 CMD: Group 0, Index 2, Command "ZBSTATUS", Data "master_bathrm_light"
20:33:57 MQT: stat/zb/RESULT = {"ZbStatus2":[{"Device":"0xAA94","Name":"master_bathrm_light","IEEEAddr":"0x00158D00044E403D","ModelId":"lumi.ctrl_ln1.aq1","Manufacturer":"LUMI","Endpoints":[1,2,3,5],"Config":["O01"]}]}
```

```
11:59:31 ZIG: {"ZbEZSPReceived":"450000040100000101000100005BECD794AAFFFF341C5F11370A01FF422B64100103281F98390000744195392D5F9D4005210F009A2010082121110727000000000000000009210002"}
11:59:31 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xAA94","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":120,"securityuse":0,"seqnumber":91,"fc":"0x1C","manuf":"0x115F","transact":55,"cmdid":"0x0A","payload":"01FF422B64100103281F98390000744195392D5F9D4005210F009A2010082121110727000000000000000009210002"}}
11:59:31 ZIG: ZbZCLRawReceived: {"0xAA94":{"0000/FF01":"64100103281F98390000744195392D5F9D4005210F009A2010082121110727000000000000000009210002","Endpoint":1,"LinkQuality":120}}
11:59:31 MQT: tele/master_bathrm_light/SENSOR = {"master_bathrm_light":{"Device":"0xAA94","Name":"master_bathrm_light","Xiaomi_64":1,"Xiaomi_98":15.25,"Xiaomi_95":4.92,"Xiaomi_9A":16,"Endpoint":1,"LinkQuality":120}} (retained)
```

Miplug

```
20:31:58 ZIG: {"ZbEZSPReceived":"45000004010C00020100010000E388BE9CDCFFFF0A18E90A5500399A993342"}
20:31:58 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":12,"srcaddr":"0xDC9C","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":55,"securityuse":0,"seqnumber":227,"fc":"0x18","manuf":"0x0000","transact":233,"cmdid":"0x0A","payload":"5500399A993342"}}
20:31:58 ZIG: ZbZCLRawReceived: {"0xDC9C":{"000C/0055":44.9,"Endpoint":2,"LinkQuality":55}}
20:31:58 MQT: tele/mi_plug1/SENSOR = {"mi_plug1":{"Device":"0xDC9C","Name":"mi_plug1","EnergyTotal":44.9,"Endpoint":2,"LinkQuality":55}} (retained)
```

tyjtyj, 2020-12-03:
additional notes. QBKG11LM will send out activepower shortly after turn power:1

20:40:10 ZIG: {"ZbEZSPReceived":"45000004010C00020100010000D1F0D894AAFFFF0A18A20A550039B81E8741"}
20:40:10 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":12,"srcaddr":"0xAA94","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":123,"securityuse":0,"seqnumber":209,"fc":"0x18","manuf":"0x0000","transact":162,"cmdid":"0x0A","payload":"550039B81E8741"}}
20:40:10 ZIG: ZbZCLRawReceived: {"0xAA94":{"000C/0055":16.89,"Endpoint":2,"LinkQuality":123}}
20:40:11 MQT: tele/master_bathrm_light/SENSOR = {"master_bathrm_light":{"Device":"0xAA94","Name":"master_bathrm_light","**AnalogValue**":16.89,"Endpoint":2,"LinkQuality":123}} (retained)

AnalogValue: ActivePower

PR #10136 (2020-12-14): Zigbee support for QBKG11LM #10039
## Description:

Support for QBKG11LM lumi.ctrl_ln1.aq1.

**Related issue (if applicable):** fixes #10039

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md [... cut]
closed 2020-12-14 by s-hadinger