arendst/Tasmota #10044: Config Sonoff TH16 with Sonoff Sensor Si7021

AvdH, 2020-12-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After upgrading my Sonoff TH16 to version 9.1.0 I can no longer configure the Si7021 sensor on GPIO14. I can't find it in the dropdown list. Previously it was 03 Si7021. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff TH16
- [ x] Tasmota binary firmware version number used: 9.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Upgrade by file upload
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
17:08:28 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:08:29 RSL: stat/sonoff/RESULT = {"Module":{"4":"Sonoff TH"}}
17:08:29 RSL: stat/sonoff/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  17:10:10 RSL: stat/sonoff/STATUS = {"Status":{"Module":4,"DeviceName":"Sonoff","FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:10:10 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:57","StartupUTC":"2020-12-03T16:08:13","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2020-12-03T15:42:01","SaveCount":139,"SaveAddress":"FA000"}}
17:10:10 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"9.1.0(lite)","BuildDateTime":"2020-11-07T11:58:37","Boot":6,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"328/699"}}
17:10:10 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vanderHeide",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
17:10:10 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21,37","Sensors":"3"}}
17:10:10 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7315","IPAddress":"192.168.2.238","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"89.101.251.229","Mac":"DC:4F:22:0E:5C:93","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:10:10 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E5C93","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:10:10 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"2020-12-03T16:10:10","Local":"2020-12-03T17:10:10","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"16:54"}}
17:10:10 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-12-03T17:10:10"}}
17:10:10 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-12-03T17:10:10","Uptime":"0T00:01:57","UptimeSec":117,"Vcc":3.462,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"54:67:51:CB:3A:BC","Channel":5,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Update to version 9.1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-03:
Please update to Tasmota-sensors.bin

Thanks.

ascillato2, 2020-12-03:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-03 by ascillato2