arendst/Tasmota #10048: Sonoff Bridge will not deactivate RfRaw-Mode when `RfRaw 0`-Command is sent via mqtt

jldmn, 2020-12-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I'm currently trying to send raw data via the Sonoff RF-Bridge an then automatically change back to the default Rf-mode which would be activated by running RfRaw 0. This somehow works perfectly when executed via the web console. But it does not work when send as a mqtt command from home-assistant or any other mqtt-client. In fact I'm unable to even run the single RfRaw 0 command via mqtt and the device is still in RfRaw: On mode.

I took a quick look into the CmndRfRaw() function inside the xdrv_06_snfbridge.ino and thought that maybe SnfBridge.receive_raw_flag = 0; is missing in the case 0´ block. But maybe this is also handled via fall-trough to case 1`.

Maybe someone who is more into the code could help me with this problem?
Or can someone explain me from where the CmndRfRaw() function is called when commanda are received via mqtt?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Brige
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:18:03 MQT: stat/rf/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:18:03 MQT: stat/rf/RESULT = {"Module":{"25":"Sonoff Bridge"}}
21:18:04 MQT: stat/rf/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
21:18:33 MQT: stat/rf/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on system#boot do RfRaw AAB155 endon"}
21:18:33 MQT: stat/rf/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
21:18:33 MQT: stat/rf/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:19:15 MQT: stat/rf/STATUS = {"Status":{"Module":25,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"rf","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:19:15 MQT: stat/rf/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://172.16.42.42:9541/data/firmwares/tasmota.bin","RestartReason":"Power On","Uptime":"0T04:15:01","StartupUTC":"2020-12-03T16:04:14","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2020-12-02T20:43:44","SaveCount":77,"SaveAddress":"FB000"}}
21:19:15 MQT: stat/rf/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"376/699"}}
21:19:15 MQT: stat/rf/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Reetdachkate_NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
21:19:15 MQT: stat/rf/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:19:15 MQT: stat/rf/STATUS5 = {"StatusNET":{"Hostname":"rf-4372","IPAddress":"172.16.30.6","Gateway":"172.16.30.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.30.1","Mac":"2C:F4:32:C4:B1:14","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
21:19:15 MQT: stat/rf/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.42.42","MqttPort":1883,"MqttClientMask":"rf_bridge","MqttClient":"rf_bridge","MqttUser":"client","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:19:15 MQT: stat/rf/STATUS7 = {"StatusTIM":{"UTC":"2020-12-03T20:19:15","Local":"2020-12-03T21:19:15","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"16:54"}}
21:19:15 MQT: stat/rf/STATUS10 = {"StatusSNS":{"Time":"2020-12-03T21:19:15"}}
21:19:15 MQT: stat/rf/STATUS11 = {"StatusSTS":{"Time":"2020-12-03T21:19:15","Uptime":"0T04:15:01","UptimeSec":15301,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Reetdachkate_NoT","BSSId":"76:AC:B9:24:09:4C","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:19:24 MQT: tele/rf/STATE = {"Time":"2020-12-03T21:19:24","Uptime":"0T04:15:10","UptimeSec":15310,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Reetdachkate_NoT","BSSId":"76:AC:B9:24:09:4C","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:22:12 CMD: rfraw 1
21:22:12 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:22:18 CMD: rfraw 0
21:22:18 MQT: stat/rf/RESULT = {"RfRaw":"OFF"}
21:22:21 CMD: rfraw 1
21:22:21 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:22:34 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:22:38 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:22:54 CMD: rfraw 0
21:22:54 MQT: stat/rf/RESULT = {"RfRaw":"OFF"}
21:22:59 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:23:02 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:23:03 MQT: stat/rf/RESULT = {"RfRaw":"ON"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Set Bridge to `RfRaw: On` via either console or mqtt by running `RfRaw 1`
2. Try to get it back to `RfRaw: Off` via mqtt by sending `RfRaw 0` command.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect the Bridge to go back into `RfRaw: Off` State after receiving and executing the given command.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-12-03:
You omitted the key details about how you actually send the command via MQTT. A common misunderstanding is to not realize that the command name goes into the topic, and the argument(s) goes into the payload, not just blank-separated. Or use Backlog.

jldmn, 2020-12-03:
That's in fact a good point. topic: `cmnd/rf/RfRaw` message: `rfraw 1`.
I also tried: `Backlog RfRaw 0`

ascillato2, 2020-12-03:
please, set weblog to 4 and post the output of the console when you send the rfraw command

ascillato2, 2020-12-03:
> That's in fact a good point. topic: `cmnd/rf/RfRaw` message: `RfRaw 0`.

that is incorrect. it MUST be

topic: `cmnd/rf/RfRaw` message: `0`.

jldmn, 2020-12-03:
Ok, seems like this is the solution.

```
21:35:02 SRC: MQTT
21:35:02 CMD: Group 0, Index 1, Command "RFRAW", Data "rfraw 0"
21:35:02 MQT: stat/rf/RESULT = {"RfRaw":"ON"}
21:35:14 MQT: Data Size 1, "cmnd/rf/RfRaw 0"
21:35:14 SRC: MQTT
21:35:14 CMD: Group 0, Index 1, Command "RFRAW", Data "0"
21:35:14 MQT: stat/rf/RESULT = {"RfRaw":"OFF"}
21:35:14 DMP: A0 55
```
But why does it work with all other commands?

ascillato2, 2020-12-03:
>  Data "rfraw 0"

that is a syntax error. MUST be just 0

ascillato2, 2020-12-03:
correct syntax:

topic: `cmnd/rf/RfRaw` message: `0`.

jldmn, 2020-12-03:
And how do i use Backlog then to first send raw data and then change back to `RfRaw: Off` ?

ascillato2, 2020-12-03:
You must send:

topic: `cmnd/rf/RfRaw`
message: `0`

ascillato2, 2020-12-03:
> 21:35:02 CMD: Group 0, Index 1, Command "RFRAW", Data "rfraw 0"

explanation:

that is the same as writing in the console `RFRAW RFRAW 0`

ascillato2, 2020-12-03:
the argument is invalid, so Tasmota is answering just the actual state of RFRAW

closed 2020-12-03 by ascillato2

jldmn, 2020-12-03:
It's working now, thank you very very much! :)

sfromis, 2020-12-03:
Using Backlog would be
```
topic: cmnd/rf/Backlog
message: RfRaw 0
```