arendst/Tasmota #10049: logging corruption - observed on ESP32 with more that one 'task'

btsimonh, 2020-12-04:
### PROBLEM DESCRIPTION
Whilst developing a new driver (xdrv_47_BLE_ESP32), serial logging corruption was observed.
the specific corruption was something like:
00:00:04 Startscan from on
00:00:04 Startscan from onesec

With ESP32, since multiple 'tasks' are possible (and essential for some operations like BLE if we are not to hold up the whole main process), it is possible to call, for example, Addlog_P from multiple threads.

We need to do some analysis of TasmotaGlobal and the data therein to see what data accesses must be protected by some form of mutex/semaphore when compiling for ESP32, else we will introduce instability and crash and burn badly and randomly.

Immediate danger would be:
Any logging from anything not on the main thread.
Any MQTT activity from anything not on the main thread.

Until we have some form of protection, all logging from secondary threads should be avoided, as should any activity against MQTT or other global data.
One issue is that it may not be obvious when using a 3rd party API like NimBLE if a callback is on the main thread or on a secondary task.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 dev board
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
n/a

### EXPECTED BEHAVIOUR
n/a

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
I would assume this only affects ESP32, as ESP8266 does not have multiple threads? (but i'm no expert).

Example of possible Semaphore use:
(what is port_MAX_DELAY anyway? - must check).
```
// this protects our global data, which can be accessed by multiple tasks 
SemaphoreHandle_t  GlobalsMutex;
...
  GlobalsMutex = xSemaphoreCreateMutex();

// example from support.ino
  xSemaphoreTake(GlobalsMutex, portMAX_DELAY);
    memmove(TasmotaGlobal.log_data + strlen(syslog_preamble), TasmotaGlobal.log_data, sizeof(TasmotaGlobal.log_data) - strlen(syslog_preamble));
    TasmotaGlobal.log_data[sizeof(TasmotaGlobal.log_data) -1] = '\0';
    memcpy(TasmotaGlobal.log_data, syslog_preamble, strlen(syslog_preamble));
    PortUdp_write(TasmotaGlobal.log_data, strlen(TasmotaGlobal.log_data));

  xSemaphoreGive(GlobalsMutex); // release mutex

```
The take/give could be a #define, making if compile to nothing when not ESP32?
There are a lot of references to .log_data, .mqtt_data, etc. so this is no small undertaking.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-04:
Learn to use preplog_p in those situations

btsimonh, 2020-12-04:
ok, so had a look.
PrepLog_P will change the data in log_data, but not actually log it, the logging begin deferred to when the next AddLog_P is called?  i.e. won't run serial from an invalid thread, or try to post on syslog.

It's not a heavy protection, as the copy into log_data would not be atomic, but it feels like it should avoid the worst of the issues (probably stemming from serial, udp, etc.. so understood.

I wonder if we can detect if we are in the main thread in AddLog_P, and simply reject and set an error flag if not (logged at next valid call) - at least them a 'bad' driver would not kill us.

However, for this BLE stuff, the logging is fairly heavy, and PrepLog_P will overwrite each time, so I'm putting in a temporary 'SafeLog_P' in the xdrv_47 driver which queues a few logs (<15 <40chars) for later emission.  Let's see if it works....

btsimonh, 2020-12-04:
I wonder if https://www.freertos.org/a00021.html#xTaskGetCurrentTaskHandle would give us a result which we could use to ban access from other tasks/threads?

ascillato2, 2020-12-06:
Closing this issue as a PR has been proposed for review. (#10067).

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-06 by ascillato2

btsimonh, 2020-12-06:
thanks @ascillato2