arendst/Tasmota #10053: Support for mcp4725 - 3.3v 12 bit i2c dac with EEPROM

fdepalo, 2020-12-04:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hello, we are about to release a new i/o board that includes among all other chips already supported by Tasmota mcp4725 for dual or quad 0-10v output ( mainly for valves, dimmers, fan with 0-10v input )

It is possible to add support to tasmota? 

If dev requires so, we can send a board prototype for tests and development/integration.

Thank You

[MCP4725.pdf](https://github.com/arendst/Tasmota/files/5642582/MCP4725.pdf)

sfromis, 2020-12-04:
A very common way for having new sensors added to Tasmota is when someone with interest creates a driver, and submits a PR for consideration.

ascillato2, 2020-12-04:
Hi,

This is an open source project. If you want to add a new driver, please provide a pull request.

Free software doesn't mean free support or free development for companies. Tasmota is built with free contributions of end-users. If you want to use Tasmota for your own product, you are allowed to do it (see license) but that doesn't meant free support or free development. That will be your job.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md).

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-04 by ascillato2