arendst/Tasmota #10056: RF Receiver not working on GPIO1 (RX)

artucast, 2020-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
GPIO1 is not working configures as RF Receiver.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:14:46 MQT: stat/receptor-rf/STATUS = {"Status":{"Module":0,"DeviceName":"Sensor RF","FriendlyName":["Sensor RF"],"Topic":"receptor-rf","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:14:46 MQT: stat/receptor-rf/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:25","StartupUTC":"2020-12-04T22:14:21","Sleep":50,"CfgHolder":4617,"BootCount":204,"BCResetTime":"2020-11-20T14:30:30","SaveCount":352,"SaveAddress":"FA000"}}
19:14:46 MQT: stat/receptor-rf/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"388/699"}}
19:14:46 MQT: stat/receptor-rf/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Castano LS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
19:14:46 MQT: stat/receptor-rf/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
19:14:46 MQT: stat/receptor-rf/STATUS5 = {"StatusNET":{"Hostname":"Sensor RF","IPAddress":"192.168.1.3","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:EF:F9:58","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:14:46 MQT: stat/receptor-rf/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"Receptor RF","MqttClient":"Receptor RF","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:14:46 MQT: stat/receptor-rf/STATUS7 = {"StatusTIM":{"UTC":"2020-12-04T22:14:46","Local":"2020-12-04T19:14:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-03:00","Sunrise":"04:26","Sunset":"12:54"}}
19:14:46 MQT: stat/receptor-rf/STATUS10 = {"StatusSNS":{"Time":"2020-12-04T19:14:46"}}
19:14:46 MQT: stat/receptor-rf/STATUS11 = {"StatusSTS":{"Time":"2020-12-04T19:14:46","Uptime":"0T00:00:25","UptimeSec":25,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Castano LS","BSSId":"14:CC:20:78:91:8A","Channel":4,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Hi Guys. 
I'm trying to get this working.
I only have available pins RX and TX in my Canwing CW001.
I want to use one of the as a RF receiver. If I connect the data pin of the receiver to TX, the device does not boot. If I connect it after boot, it works perfectly.
If I connect it to the RX pin, it does nothing.
Any solution? 
It will be great if there is any way to leave it connected to TX pin and make it boot.

Thans a lot.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-12-04:
Boot fails if TX pulled low. Did you try disabling serial logging? `SerialLog 0`

artucast, 2020-12-04:
> Boot fails if TX pulled low. Did you try disabling serial logging? `SerialLog 0`

I know. But I'ts not pulled low, unless there is some ground "leaking" through the data pin of the receiver. Any ideas on how to solve this?
I tried SerialLog 0 and is still not working with RX .

ascillato, 2020-12-05:
Sorry, that is a hardware limitation of the ESP8266. That pin is used for boot selection and it can't be modified by software. You need to use another gpio. Sorry.

GPIO limitations are explained in the docs.

closed 2020-12-05 by ascillato2