arendst/Tasmota #10066: Strange MQTT CMD occurrence on ESP32 with script support

vslovak, 2020-12-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Strange MQTT CMD observed on ESP32 with option Script compiled. Occurrence is accidental. ESP32 compiled with Display and Script options. As source is given Serial (weblog 4), but nothing is connected to Serial

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32-DevKit Modul
- [ ] Tasmota binary firmware version number used: 9.1.0.2(display)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: 1_0_4_2/v3.3.4-232-g44ec7972b
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:34:47 MQT: stat/kotel4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
19:34:47 MQT: stat/kotel4/RESULT = {"Module":{"1":"ESP32-DevKit"}}
19:34:47 MQT: stat/kotel4/RESULT = {"GPIO0":{"32":"Tlačítko1"},"GPIO1":{"0":"Není"},"GPIO2":{"0":"Není"},"GPIO3":{"0":"Není"},"GPIO4":{"0":"Není"},"GPIO5":{"0":"Není"},"GPIO9":{"0":"Není"},"GPIO10":{"0":"Není"},"GPIO12":{"224":"Relé1"},"GPIO13":{"448":"PWM_i1"},"GPIO14":{"544":"LedLink"},"GPIO15":{"225":"Relé2"},"GPIO16":{"0":"Není"},"GPIO17":{"0":"Není"},"GPIO18":{"0":"Není"},"GPIO19":{"0":"Není"},"GPIO20":{"0":"Není"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"Není"},"GPIO24":{"0":"Není"},"GPIO25":{"2560":"MX31855 DO"},"GPIO26":{"2496":"MX31855 CS"},"GPIO27":{"2528":"MX31855 CLK"},"GPIO28":{"0":"Není"},"GPIO29":{"0":"Není"},"GPIO30":{"0":"Není"}," ...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:34:05 MQT: stat/kotel4/STATUS = {"Status":{"Module":1,"DeviceName":"kotel4","FriendlyName":["kotel4","Tasmota2","Tasmota3","Tasmota4"],"Topic":"kotel4","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:34:05 MQT: stat/kotel4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:00:39","StartupUTC":"2020-12-05T18:33:26","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-12-05T19:04:37","SaveCount":19}}
19:34:05 MQT: stat/kotel4/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(display)","BuildDateTime":"2020-11-29T16:40:29","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"457/699"}}
19:34:05 MQT: stat/kotel4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vlsiot",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000500060000005A00000000000000","00804000","00007000","00000000"]}}
19:34:05 MQT: stat/kotel4/STATUS4 = {"StatusMEM":{"ProgramSize":1089,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000405","07AA858E","000C1FA1","000000C3","00080000","00000080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13","Sensors":"1,2,5,6,39"}}
19:34:05 MQT: stat/kotel4/STATUS5 = {"StatusNET":{"Hostname":"kotel4","IPAddress":"192.168.8.210","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.1","Mac":"A8:03:2A:D6:87:A0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:34:05 MQT: stat/kotel4/STATUS6 = {"StatusMQT":{"MqttHost":"rpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D687A0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:34:05 MQT: stat/kotel4/STATUS7 = {"StatusTIM":{"UTC":"2020-12-05T18:34:05","Local":"2020-12-05T19:34:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:26","Sunset":"15:47"}}
19:34:05 MQT: stat/kotel4/STATUS10 = {"StatusSNS":{"Time":"2020-12-05T19:34:05","ANALOG":{"Temperature1":3.5,"Temperature2":null},"MAX6675":{"ProbeTemperature":59.0,"ReferenceTemperature":null,"Error":0},"TempUnit":"C"}}
19:34:05 MQT: stat/kotel4/STATUS11 = {"StatusSTS":{"Time":"2020-12-05T19:34:05","Uptime":"0T00:00:39","UptimeSec":39,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"vlsiot","BSSId":"A8:5E:45:8F:AB:99","Channel":5,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

00:00:00 CFG: Loaded, Počítej 26
00:00:00 QPC: Count 1
00:00:00 Script: nv=21, tv=7, vns=135, ram=543
00:00:00 I2C: SH1106 znalezeno v 0x3c
00:00:00 Projekt tasmota kotel4 Verze 9.1.0.2(display)-1_0_4_2
00:00:00  Script version 0.10
00:00:00 CMD: nt 1
00:00:00 RSL: stat/kotel4/RESULT = {"Command":"Unknown"}
00:00:01 WIF: Připojování k AP1 vlsiot Channel 5 BSSId A8:5E:45:8F:AB:99 v módu 11  jako kotel4...
00:00:04 WIF: ...připojeno
00:00:04 HTP: Aktivní Web server kotel4 na IP adrese 192.168.8.210
19:56:20 MQT: Připojování...
19:56:20 MQT: ...připojeno
19:56:20 MQT: tele/kotel4/LWT = Online (Zachováno)
19:56:20 MQT: cmnd/kotel4/POWER = 
19:56:20 MQT: tele/kotel4/INFO1 = {"Module":"ESP32-DevKit","Version":"9.1.0.2(display)","FallbackTopic":"cmnd/DVES_D687A0_fb/","GroupTopic":"cmnd/tasmotas/"}
19:56:20 MQT: tele/kotel4/INFO2 = {"WebServerMode":"Admin","Hostname":"kotel4","IPAddress":"192.168.8.210"}
19:56:20 MQT: tele/kotel4/INFO3 = {"RestartReason":"Software reset CPU"}
19:56:20 MQT: stat/kotel4/RESULT = {"POWER1":"OFF"}
19:56:20 MQT: stat/kotel4/POWER1 = OFF
19:56:20 MQT: stat/kotel4/RESULT = {"POWER2":"OFF"}
19:56:20 MQT: stat/kotel4/POWER2 = OFF
19:56:20 MQT: stat/kotel4/RESULT = {"POWER3":"OFF"}
19:56:20 MQT: stat/kotel4/POWER3 = OFF
19:56:20 MQT: stat/kotel4/RESULT = {"POWER4":"ON"}
19:56:20 MQT: stat/kotel4/POWER4 = ON
19:56:20 CMD: /RESULT = {"POWERz5
19:56:20 MQT: stat/kotel4/RESULT = {"Command":"Unknown"}

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-05:
Set seriallog to 0 and try again please

vslovak, 2020-12-06:
Hi,
yes, set seriallog to 0 solve the mentioned behavior. I change the HW design to add pull up to RX pin (GPIO3). It resolve the reason of the issue.

closed 2020-12-06 by vslovak