arendst/Tasmota #10068: 支持控制舵机PWM

fengnaiping, 2020-12-06:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

MattWestb, 2020-12-06:
GT: "Support control servo PWM"

Jason2866, 2020-12-06:
Servo can be controlled. Just set a gpio to pwm. Fir the command `PWMfrequency 50` 
and you can controll a servo

MattWestb, 2020-12-06:
GT: 伺服是可以控制的。 只需将gpio设置为pwm。 触发命令 `PWMfrequency 50`
您可以使用 `dimmer` 命令控制伺服

ascillato2, 2020-12-06:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-06 by ascillato2

TrDA-hab, 2021-01-07:
> GT: 是 可以 控制 的。 只需 将 gpio 设置 为 pwm。 触发 命令`PWMfrequency 50`
> 您可以使用`dimmer`命令 控制 伺服

https://github.com/arendst/Tasmota/discussions/10443