arendst/Tasmota #10070: iAQ Core i2c works only with address 90

DarthWeber, 2020-12-06:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
I have several ams IAQ-CORE C sensors with i2c adresses from 90...95. But only that one with 90 works perfect with tasmota. Is it possible to add support to those address range up top 95 or 99 ?
**Describe the solution you'd like**  
support AMS iAQ Core with address 91...95 or higher
**Describe alternatives you've considered**  
no idea except building and compile own versions. But that would mean to use 6 different versions
**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-12-06:
As things are now, the address is hardcoded in the driver `xsns_66_iAQ.ino`
`#define I2_ADR_IAQ         0x5a`
Supporting a non-static address would take someone working to expand the driver for this.

arendst, 2020-12-06:
I've searched for information about iAQ-Core address range but I cannot find anything other than a fixed 0x5A (=90).

https://www.sciosense.com/wp-content/uploads/2020/01/iaQ-Core-Datasheet.pdf

How do you change the I2C address of the iAQ-Core?

arendst, 2020-12-06:
Another nice one https://www.mouser.com/PCN/ScioSense_SC_001021_CN_1_Sciosense_End_of_Life_(EOL)_Notification_iAQ_Core_EOL_2020_002.pdf

blakadder, 2020-12-09:
when we did the driver there were no indications iAQ core uses any other address than 0x5A so it was coded that way

DarthWeber, 2020-12-09:
I did not change the adress, according to AMS it is not possible for end users to change the address. However, i was asked when i purchased 5 of them if i would like to have them with different address. I asked them if i can change it back it 0x5A but they declined.

Jason2866, 2020-12-10:
Sorry it is not from common interest and it will not be implemented.

closed 2020-12-10 by ascillato2