arendst/Tasmota #10073: Websend command to other non tasmota host  with username&password

mamrai1, 2020-12-06:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
example 2: WebSend [myserver.com] /fancy/data.php?log=1234 sends http://myserver.com/fancy/data.php?log=1234
but 
WebSend [myserver.com,username:password] /fancy/data.php?log=1234 fails!

**Describe the solution you'd like**  
WebSend [myserver.com,username:password] /fancy/data.php?log=1234 should send:
http://username:password@myserver.com/fancy/data.php?log=1234

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-06:
In order to send http://username:password@myserver.com/fancy/data.php?log=1234

have you tried the following?
```lua
WebSend [username:password@myserver.com] /fancy/data.php?log=1234
```

mamrai1, 2020-12-06:
Yes, tried this: websend [admin:password@192.168.1.41] /relay/0?turn=off
but i get {"WebSend":"Connect failed"}

mamrai1, 2020-12-06:
oh. the problem is my password that cntains the '@' character.. is there a way to fix it wiithout changing my pass?

ascillato, 2020-12-06:
Sorry, there isn't.

mamrai1, 2020-12-06:
ok thanks

closed 2020-12-06 by mamrai1