arendst/Tasmota #10076: tasmota-ir PIONEER protocol not detected in 9.1.0 was working in 8.5.1

linuxlurak, 2020-12-06:
### PROBLEM DESCRIPTION
with tasmota-ir 8.5.1 ir codes (example following) for PIONEER protocol was working, with 9.1.0 not anymore:
`irsend {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0,"RawData":[8504,4194,584,1540,560,464,578,1544,578,472,584,466,578,1544,578,498,556,1540,582,468,586,1512,586,464,580,1542,582,1542,580,470,584,1538,562,488,554,1544,578,1546,588,462,582,1540,582,468,590,458,582,468,586,462,580,468,554,496,560,1538,584,492,552,1546,588,1536,586,1538,560,1538,584,25190,8510,4188,580,1544,588,486,560,1538,562,514,530,494,560,1538,586,490,554,1544,578,496,558,1540,584,492,552,1520,580,1544,580,496,560,1538,584,492,552,1546,578,1520,580,496,560,1538,586,490,554,496,560,488,554,494,560,488,554,494,562,1536,584,466,554,1544,578,1546,588,1536,588,1536,584,25166,8512,4186,580,1544,578,470,584,1540,582,466,588,460,582,1542,556,492,552,1548,586,464,580,1544,578,472,584,1540,582,1518,582,466,576,1548,586,464,580,1544,578,1546,576,472,584,1516,586,464,580,468,586,464,580,468,586,464,582,468,586,1538,584,466,578,1546,576,1524,576,1548,584,1540,582],"RawDataInfo":[203,203,0]}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): YTF IR Bridge see: https://tasmota.github.io/docs/devices/YTF-IR-Bridge/
- [x] Tasmota binary firmware version number used: 8.5.1. working / 9.1.0 not workind
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
stat/tasmota_43E2B3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"51":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"17":"Button1"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
none used.

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:01:25 CMD: STATUS 0 
21:01:25 SRC: WebConsole from 192.168.10.185
21:01:25 CMD: Group 0, Index 1, Command "STATUS", Data "0 "
21:01:25 MQT: stat/tasmota_43E2B3/STATUS = {"Status":{"Module":62,"DeviceName":"tasmota-ir","FriendlyName":["tasmota-ir"],"Topic":"tasmota_43E2B3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v9.0.1/tasmota-ir.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:29","StartupUTC":"2020-12-06T19:52:56","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-10-05T20:26:48","SaveCount":76,"SaveAddress":"FB000"}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS2 = {"StatusFWR":{"Version":"8.5.1(ir)","BuildDateTime":"2020-10-02T10:12:16","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"427/699"}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["uxj18zny4i3_2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000300","00006000","00000000"]}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","04000001","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":""}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS5 = {"StatusNET":{"Hostname":"tasmota-ir","IPAddress":"192.168.10.86","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.4","Mac":"EC:FA:BC:43:E2:B3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_43E2B3","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS7 = {"StatusTIM":{"UTC":"2020-12-06T20:01:25","Local":"2020-12-06T21:01:25","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:28","Sunset":"16:53"}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS10 = {"StatusSNS":{"Time":"2020-12-06T21:01:25"}}
21:01:25 MQT: stat/tasmota_43E2B3/STATUS11 = {"StatusSTS":{"Time":"2020-12-06T21:01:25","Uptime":"0T00:08:29","UptimeSec":509,"Vcc":3.524,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXXXXXXXXX.4","BSSId":"XXXXXXXXXxxx","Channel":1,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:01:27 WIF: Checking connection...

### TO REPRODUCE
_Steps to reproduce the behavior:_
send ir with this:
`irsend {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0,"RawData":[8504,4194,584,1540,560,464,578,1544,578,472,584,466,578,1544,578,498,556,1540,582,468,586,1512,586,464,580,1542,582,1542,580,470,584,1538,562,488,554,1544,578,1546,588,462,582,1540,582,468,590,458,582,468,586,462,580,468,554,496,560,1538,584,492,552,1546,588,1536,586,1538,560,1538,584,25190,8510,4188,580,1544,588,486,560,1538,562,514,530,494,560,1538,586,490,554,1544,578,496,558,1540,584,492,552,1520,580,1544,580,496,560,1538,584,492,552,1546,578,1520,580,496,560,1538,586,490,554,496,560,488,554,494,560,488,554,494,562,1536,584,466,554,1544,578,1546,588,1536,588,1536,584,25166,8512,4186,580,1544,578,470,584,1540,582,466,588,460,582,1542,556,492,552,1548,586,464,580,1544,578,472,584,1540,582,1518,582,466,576,1548,586,464,580,1544,578,1546,576,472,584,1516,586,464,580,468,586,464,580,468,586,464,582,468,586,1538,584,466,578,1546,576,1524,576,1548,584,1540,582],"RawDataInfo":[203,203,0]}`

### EXPECTED BEHAVIOUR
Have ir code sent out

s-hadinger, 2020-12-06:
You can't send raw data with this syntax. Choose either PIONEER protocol OR raw.

ascillato2, 2020-12-06:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-06 by ascillato2

linuxlurak, 2020-12-06:
Thanks for the clarification @s-hadinger! 

I can send with this command on tasmota-ir 8.5.1 but not in 9.1.0. There has been a change in handling JSON I guess. I actually can't see the point in this. Why not choose protocol data over raw data if both is given and send the IR code. At the moment there is nothing that gets sent. It's no big deal for me since you helped me understanding this.  

So do not just copy the console output and send it back. Just take the protocol part or the raw part. For the sake of completeness:

PIONEER volume down (as an example) works like this on 8.5.1 and 9.5.1:
irsend `{"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0}`

This works on 8.5.1 but does not work anymore on 9.1.0:
irsend {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0,"RawData":[8504,4194,584,1540,560,464,578,1544,578,472,584,466,578,1544,578,498,556,1540,582,468,586,1512,586,464,580,1542,582,1542,580,470,584,1538,562,488,554,1544,578,1546,588,462,582,1540,582,468,590,458,582,468,586,462,580,468,554,496,560,1538,584,492,552,1546,588,1536,586,1538,560,1538,584,25190,8510,4188,580,1544,588,486,560,1538,562,514,530,494,560,1538,586,490,554,1544,578,496,558,1540,584,492,552,1520,580,1544,580,496,560,1538,584,492,552,1546,578,1520,580,496,560,1538,586,490,554,496,560,488,554,494,560,488,554,494,562,1536,584,466,554,1544,578,1546,588,1536,588,1536,584,25166,8512,4186,580,1544,578,470,584,1540,582,466,588,460,582,1542,556,492,552,1548,586,464,580,1544,578,472,584,1540,582,1518,582,466,576,1548,586,464,580,1544,578,1546,576,472,584,1516,586,464,580,468,586,464,580,468,586,464,582,468,586,1538,584,466,578,1546,576,1524,576,1548,584,1540,582],"RawDataInfo":[203,203,0]}

Thanks again for this clarification and have a good one.

kpalczewski, 2021-02-12:
@linuxlurak  I can see from here:
https://github.com/arendst/Tasmota/pull/6917
that tasmota has special tasmota-ir version that supports pioneer.
you might also change the encoding:
https://tasmota.github.io/docs/IRSend-RAW-Encoding/#pioneer-ir-enconding