arendst/Tasmota #10080: Tuya dimmer doesn't send MQT message on manual brightness change fix for issue # 9583 / #9606 not fully fixed

fahubgit, 2020-12-07:
### PROBLEM DESCRIPTION
The fix in v9.1.0 to resolve the issue # 9583 for Tuya Module dimmers in v8.5.1 fixes the MQT update issue presents but causes other issues. In the new version, control of the dimmer works fine through the Tasmota web GUI and Home Assistant before the +/- buttons on the dimmer itself are pressed. (Using Treatlife DS02S dimmer). Once the buttons on the dimmer are used, the switch does not respond to slider controls on the Tasmota web GUI or Home Assistant for a random amount of time. Eventually, it starts responding but there is no telling when it will. During that time, the console shows "TYA: Send dim skipped value ### for dpid 2" when attempting to set a different brightness value. I also tried v9.1.0.2 and it had the same behavior.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Treatlife DS02S
- [ ] Tasmota binary firmware version number used: 9.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
05:59:46 MQT: stat/tasmota_E54EC2/RESULT = {"NAME":"DS02S Dimmer","GPIO":[0,2272,0,2304,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
05:59:46 MQT: stat/tasmota_E54EC2/RESULT = {"Module":{"0":"DS02S Dimmer"}}
05:59:46 MQT: stat/tasmota_E54EC2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:00:19 MQT: stat/tasmota_E54EC2/STATUS = {"Status":{"Module":0,"DeviceName":"office_light","FriendlyName":["Office Light"],"Topic":"tasmota_E54EC2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T10:11:55","StartupUTC":"2020-12-06T18:48:24","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-10-29T08:16:17","SaveCount":638,"SaveAddress":"FB000"}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"420/699"}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FiberPower",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["001A8009","2805C8000100060000005A00000000000000","00000210","00006000","00000000"]}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E54EC2-3778","IPAddress":"192.168.220.228","Gateway":"192.168.220.1","Subnetmask":"255.255.254.0","DNSServer":"192.168.220.1","Mac":"A4:CF:12:E5:4E:C2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.220.106","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E54EC2","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS7 = {"StatusTIM":{"UTC":"2020-12-07T05:00:19","Local":"2020-12-07T06:00:19","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:30","Sunset":"16:53"}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS10 = {"StatusSNS":{"Time":"2020-12-07T06:00:19"}}
06:00:19 MQT: stat/tasmota_E54EC2/STATUS11 = {"StatusSTS":{"Time":"2020-12-07T06:00:19","Uptime":"0T10:11:55","UptimeSec":36715,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":16,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:50:55 TYA: Send dim skipped value 200 for dpid 2
19:50:55 CFG: Saved to flash at F8, Count 601, Bytes 4096
19:50:58 SRC: WebGui from 192.168.220.101
19:50:58 CMD: Group 0, Index 1, Command "DIMMER", Data "74"
19:50:58 MQT: tele/tasmota_E54EC2/STATE = {"Time":"2020-12-06T19:50:58","Uptime":"0T00:02:34","UptimeSec":154,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:50:58 MQT: stat/tasmota_E54EC2/RESULT = {"POWER":"ON","Dimmer":74}
19:50:58 TYA: Send "55aa0006000802020004000002e4fb"
19:50:58 TYA: Send "55aa00000000ff"
19:50:58 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:50:58 TYA: Heartbeat
19:50:58 CFG: Saved to flash at F7, Count 602, Bytes 4096
19:50:58 {"TuyaReceived":{"Data":"55AA0307000802020004000002E4FF","Cmnd":7,"CmndData":"02020004000002E4","DpType2Id2":740,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000002E4"}}}
19:50:58 TYA: fnId=21 is set for dpId=2
19:50:58 TYA: RX value 740 from dpId 2 
19:51:07 HTP: Main Menu
19:51:07 WIF: Checking connection...
19:51:09 TYA: Send "55aa00000000ff"
19:51:09 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:51:09 TYA: Heartbeat
19:51:10 SRC: WebGui from 192.168.220.101
19:51:10 CMD: Group 0, Index 1, Command "DIMMER", Data "19"
[... 259 lines cut ...]
```

### TO REPRODUCE
Flash v9.1.0

Try the following (My results included):

Change dimmer setting twice in Home assistant - Works
Change dimmer setting twice in Tasmota GUI - Works
Change dimmer settings 4 times by pressing button on switch - Works
Change dimmer settings twice in Home Assistant - Doesn't work
Change dimmer settings once in Tasmota GUI - Doesn't work
Change dimmer settings once again in Tasmota GUI - Works
Change dimmer settings once again in Home Assistant - Works

### EXPECTED BEHAVIOUR
Device should continue to receive change brightness commands through any method without issue after using the physical buttons.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-12-07:
Can't reproduce it. At least not yet! :-)
A little more than a week ago we updated the Tuya Dimmer code, you wrote you tested already a 9.1.0.2 bin but if your dev version was not so recent would you mind to test the latest too?  Thanks!

ascillato2, 2020-12-08:
Latest Tasmota version: http://ota.tasmota.com/tasmota/tasmota.bin

fahubgit, 2020-12-08:
I just tried updating to the latest dev version again and it's still doing it. I have to get to work now but I will set up a test station with a light bulb and switch later and try to capture a video of it happening. Maybe that will provide some additional insight and help. I will provide the console readout from that as well of course.

Could you provide some insight on what the "TYA: Send dim skipped value ### for dpid 2" means? I tried to search some info on it but didn't come up with much. That seems to be the response whenever is dim command doesn't produce a result.

effelle, 2020-12-08:
Thanks for your time!
> Could you provide some insight on what the "TYA: Send dim skipped value ### for dpid 2" means? I tried to search some info on it but didn't come up with much. That seems to be the response whenever is dim command doesn't produce a result.

Tasmota will skip a value if too much close of the one registered on memory (normally -1/+1) . A value is also skipped if it cames too fast from a previous one to avoid flickering (normally 250ms).
In both cases it will not send the command to the MCU and then the dimmer with not change state.

fahubgit, 2020-12-13:
Sorry for the delay in updating this. It's been a busy week for me.

I have recorded a video of the behavior and uploaded it to YouTube for you to review. The video shows the behavior change from 8.5.1 to 9.1.0/9.1.0.2 and the difference after using the hardware buttons on both. I will include the console outputs below of all 3 firmware versions from the video to match up/review as needed.

Video link : https://youtu.be/V38gdh8woyI

fahubgit, 2020-12-13:
Console Output for v8.5.1

19:45:27 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON","Dimmer":25}
19:45:27 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:45:27","Uptime":"1T20:46:29","UptimeSec":161189,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":25,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:45:27 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:45:27 MQT: stat/tasmota_E54725/POWER = ON
19:45:38 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:45:38","Uptime":"1T20:46:40","UptimeSec":161200,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":88,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:45:38 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON","Dimmer":88}
19:45:39 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:45:38","Uptime":"1T20:46:40","UptimeSec":161200,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":104,"MqttCount":1,"POWER":"ON","Dimmer":88,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:45:39 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:45:39 MQT: stat/tasmota_E54725/POWER = ON
19:45:40 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:45:40","Uptime":"1T20:46:42","UptimeSec":161202,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":19,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:45:40 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON","Dimmer":19}
19:45:40 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:45:40","Uptime":"1T20:46:42","UptimeSec":161202,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":19,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:45:40 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:45:40 MQT: stat/tasmota_E54725/POWER = ON
19:46:15 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:46:15","Uptime":"1T20:47:17","UptimeSec":161237,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"POWER":"OFF","Dimmer":19,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:46:15 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF"}
19:46:15 MQT: stat/tasmota_E54725/POWER = OFF
19:49:03 MQT: tele/tasmota_E54725/HASS_STATE = {"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Module or Template":"DS02S Dimmer","RestartReason":"Power On","Uptime":"1T20:50:05","Hostname":"tasmota_E54725-1829","IPAddress":"192.168.220.225","RSSI":"100","Signal (dBm)":"-46","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":1,"LoadAvg":19}
19:49:11 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:49:11","Uptime":"1T20:50:13","UptimeSec":161413,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":19,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:50:16 CMD: WebLog 4
19:50:16 MQT: stat/tasmota_E54725/RESULT = {"WebLog":4}
19:50:17 CFG: Saved to flash at F4, Count 168, Bytes 4096
19:50:21 TYA: Send "55aa00000000ff"
19:50:21 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:50:21 TYA: Heartbeat
19:50:25 WIF: Checking connection...
19:50:31 MQT: Received Topic "cmnd/tasmota_E54725/Dimmer", Data Size 2, Data "67"
19:50:31 SRC: MQTT
19:50:31 CMD: Group 0, Index 1, Command "DIMMER", Data "67"
19:50:31 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:50:31","Uptime":"1T20:51:33","UptimeSec":161493,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":67,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:50:31 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF","Dimmer":67}
19:50:31 TYA: Send "55aa00060008020200040000029eb5"
19:50:31 MQT: Received Topic "cmnd/tasmota_E54725/POWER", Data Size 2, Data "ON"
19:50:31 SRC: MQTT
19:50:31 CMD: Group 0, Index 1, Command "POWER", Data "ON"
19:50:31 TYA: Send "55aa0006000501010001010e"
19:50:31 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:50:31","Uptime":"1T20:51:33","UptimeSec":161493,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":28,"MqttCount":1,"POWER":"ON","Dimmer":67,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:50:31 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:50:31 MQT: stat/tasmota_E54725/POWER = ON
[... cut ...]
19:51:44 {"TuyaReceived":{"Data":"55AA03070008020200040000016882","Cmnd":7,"CmndData":"0202000400000168","DpType2Id2":360,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000168"}}}
19:51:44 TYA: fnId=21 is set for dpId=2
19:51:44 TYA: RX value 360 from dpId 2 
19:51:45 WIF: Checking connection...
19:51:49 TYA: Send "55aa00000000ff"
19:51:49 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:51:49 TYA: Heartbeat
19:51:50 CMD: WebLog 2
19:51:50 SRC: WebConsole from 192.168.220.101
19:51:50 CMD: Group 0, Index 1, Command "WEBLOG", Data "2"
19:51:50 MQT: stat/tasmota_E54725/RESULT = {"WebLog":2}

fahubgit, 2020-12-13:
Console output for 9.1.0

00:00:00 CFG: Loaded from flash at FB, Count 182
00:00:00 QPC: Reset
00:00:00 SER: Set to 8N1 115200 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota hallway_light Version 9.1.0(tasmota)-2_7_4_5
00:00:00 WIF: Connecting to AP1 FiberPower Channel 2 BSSId 88:DC:96:8C:29:90 in mode 11N as tasmota_E54725-1829...
00:00:01 WIF: Connected
19:54:56 HTP: Web server active on tasmota_E54725-1829 with IP address 192.168.220.225
19:54:57 MQT: Attempting connection...
19:54:57 MQT: Connected
19:54:57 MQT: tele/tasmota_E54725/LWT = Online (retained)
19:54:57 MQT: cmnd/tasmota_E54725/POWER = 
19:54:57 MQT: tele/tasmota_E54725/INFO1 = {"Module":"DS02S Dimmer","Version":"9.1.0(tasmota)","FallbackTopic":"cmnd/DVES_E54725_fb/","GroupTopic":"cmnd/tasmotas/"}
19:54:57 MQT: tele/tasmota_E54725/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_E54725-1829","IPAddress":"192.168.220.225"}
19:54:57 MQT: tele/tasmota_E54725/INFO3 = {"RestartReason":"Software/System restart"}
19:54:57 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF"}
19:54:57 MQT: stat/tasmota_E54725/POWER = OFF
19:54:58 MQT: homeassistant/light/E54725_LI_1/config = {"name":"Hallway Light","stat_t":"tele/tasmota_E54725/STATE","avty_t":"tele/tasmota_E54725/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/tasmota_E54725/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"E54725_LI_1","dev":{"ids":["E54725"]},"bri_cmd_t":"cmnd/tasmota_E54725/Dimmer","bri_stat_t":"tele/tasmota_E54725/STATE","bri_scl":100,"on_cmd_type":"last","bri_val_tpl":"{{value_json.Dimmer}}"} (retained)
19:54:58 MQT: homeassistant/sensor/E54725_status/config = {"name":"hallway_light status","stat_t":"tele/tasmota_E54725/HASS_STATE","avty_t":"tele/tasmota_E54725/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota_E54725/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"E54725_status","dev":{"ids":["E54725"],"name":"hallway_light","mdl":"DS02S Dimmer","sw":"9.1.0(tasmota)","mf":"Tasmota"}} (retained)
19:55:01 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:01","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"POWER":"OFF","Dimmer":36,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:24 CMD: WebLog 4
19:55:24 MQT: stat/tasmota_E54725/RESULT = {"WebLog":4}
19:55:25 TYA: Send "55aa00000000ff"
19:55:25 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:55:25 TYA: Heartbeat
19:55:25 CFG: Saved to flash at F8, Count 186, Bytes 4096
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/Dimmer 51"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "DIMMER", Data "51"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF","Dimmer":51}
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/POWER ON"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "POWER", Data "ON"
19:55:33 TYA: Send "55aa0006000501010001010e"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":21,"MqttCount":1,"POWER":"ON","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:55:33 MQT: stat/tasmota_E54725/POWER = ON
19:55:33 {"TuyaReceived":{"Data":"55AA03070008020200040000016882","Cmnd":7,"CmndData":"0202000400000168","DpType2Id2":360,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000168"}}}
19:55:33 TYA: fnId=21 is set for dpId=2
19:55:33 TYA: RX value 360 from dpId 2 
19:55:33 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:55:33 TYA: fnId=11 is set for dpId=1
19:55:33 TYA: RX Relay-1 --> MCU State: On Current State:On
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/Dimmer 51"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "DIMMER", Data "51"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":27,"MqttCount":1,"POWER":"ON","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON","Dimmer":51}
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/POWER ON"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "POWER", Data "ON"
19:55:33 TYA: Send "55aa0006000501010001010e"
[... cut ...]
19:58:00 TYA: fnId=11 is set for dpId=1
19:58:00 TYA: RX Relay-1 --> MCU State: On Current State:On
19:58:00 {"TuyaReceived":{"Data":"55AA0307000802020004000000F009","Cmnd":7,"CmndData":"02020004000000F0","DpType2Id2":240,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000F0"}}}
19:58:00 TYA: fnId=21 is set for dpId=2
19:58:00 TYA: RX value 240 from dpId 2 
19:58:09 CMD: WebLog 2
19:58:09 SRC: WebConsole from 192.168.220.101
19:58:09 CMD: Group 0, Index 1, Command "WEBLOG", Data "2"
19:58:09 MQT: stat/tasmota_E54725/RESULT = {"WebLog":2}

fahubgit, 2020-12-13:
Console output for 9.1.0.2 from 12/11/2020

00:00:00 CFG: Loaded from flash at FB, Count 182
00:00:00 QPC: Reset
00:00:00 SER: Set to 8N1 115200 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota hallway_light Version 9.1.0(tasmota)-2_7_4_5
00:00:00 WIF: Connecting to AP1 FiberPower Channel 2 BSSId 88:DC:96:8C:29:90 in mode 11N as tasmota_E54725-1829...
00:00:01 WIF: Connected
19:54:56 HTP: Web server active on tasmota_E54725-1829 with IP address 192.168.220.225
19:54:57 MQT: Attempting connection...
19:54:57 MQT: Connected
19:54:57 MQT: tele/tasmota_E54725/LWT = Online (retained)
19:54:57 MQT: cmnd/tasmota_E54725/POWER = 
19:54:57 MQT: tele/tasmota_E54725/INFO1 = {"Module":"DS02S Dimmer","Version":"9.1.0(tasmota)","FallbackTopic":"cmnd/DVES_E54725_fb/","GroupTopic":"cmnd/tasmotas/"}
19:54:57 MQT: tele/tasmota_E54725/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_E54725-1829","IPAddress":"192.168.220.225"}
19:54:57 MQT: tele/tasmota_E54725/INFO3 = {"RestartReason":"Software/System restart"}
19:54:57 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF"}
19:54:57 MQT: stat/tasmota_E54725/POWER = OFF
19:54:58 MQT: homeassistant/light/E54725_LI_1/config = {"name":"Hallway Light","stat_t":"tele/tasmota_E54725/STATE","avty_t":"tele/tasmota_E54725/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/tasmota_E54725/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"E54725_LI_1","dev":{"ids":["E54725"]},"bri_cmd_t":"cmnd/tasmota_E54725/Dimmer","bri_stat_t":"tele/tasmota_E54725/STATE","bri_scl":100,"on_cmd_type":"last","bri_val_tpl":"{{value_json.Dimmer}}"} (retained)
19:54:58 MQT: homeassistant/sensor/E54725_status/config = {"name":"hallway_light status","stat_t":"tele/tasmota_E54725/HASS_STATE","avty_t":"tele/tasmota_E54725/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota_E54725/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"E54725_status","dev":{"ids":["E54725"],"name":"hallway_light","mdl":"DS02S Dimmer","sw":"9.1.0(tasmota)","mf":"Tasmota"}} (retained)
19:55:01 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:01","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"POWER":"OFF","Dimmer":36,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:24 CMD: WebLog 4
19:55:24 MQT: stat/tasmota_E54725/RESULT = {"WebLog":4}
19:55:25 TYA: Send "55aa00000000ff"
19:55:25 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
19:55:25 TYA: Heartbeat
19:55:25 CFG: Saved to flash at F8, Count 186, Bytes 4096
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/Dimmer 51"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "DIMMER", Data "51"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"OFF","Dimmer":51}
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/POWER ON"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "POWER", Data "ON"
19:55:33 TYA: Send "55aa0006000501010001010e"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":21,"MqttCount":1,"POWER":"ON","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON"}
19:55:33 MQT: stat/tasmota_E54725/POWER = ON
19:55:33 {"TuyaReceived":{"Data":"55AA03070008020200040000016882","Cmnd":7,"CmndData":"0202000400000168","DpType2Id2":360,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000168"}}}
19:55:33 TYA: fnId=21 is set for dpId=2
19:55:33 TYA: RX value 360 from dpId 2 
19:55:33 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
19:55:33 TYA: fnId=11 is set for dpId=1
19:55:33 TYA: RX Relay-1 --> MCU State: On Current State:On
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/Dimmer 51"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "DIMMER", Data "51"
19:55:33 MQT: tele/tasmota_E54725/STATE = {"Time":"2020-12-12T19:55:33","Uptime":"0T00:00:41","UptimeSec":41,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":27,"MqttCount":1,"POWER":"ON","Dimmer":51,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"FiberPower","BSSId":"88:DC:96:8C:29:90","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:55:33 MQT: stat/tasmota_E54725/RESULT = {"POWER":"ON","Dimmer":51}
19:55:33 MQT: Data Size 2, "cmnd/tasmota_E54725/POWER ON"
19:55:33 SRC: MQTT
19:55:33 CMD: Group 0, Index 1, Command "POWER", Data "ON"
19:55:33 TYA: Send "55aa0006000501010001010e"
[... cut ...]
19:58:00 TYA: fnId=11 is set for dpId=1
19:58:00 TYA: RX Relay-1 --> MCU State: On Current State:On
19:58:00 {"TuyaReceived":{"Data":"55AA0307000802020004000000F009","Cmnd":7,"CmndData":"02020004000000F0","DpType2Id2":240,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000F0"}}}
19:58:00 TYA: fnId=21 is set for dpId=2
19:58:00 TYA: RX value 240 from dpId 2 
19:58:09 CMD: WebLog 2
19:58:09 SRC: WebConsole from 192.168.220.101
19:58:09 CMD: Group 0, Index 1, Command "WEBLOG", Data "2"
19:58:09 MQT: stat/tasmota_E54725/RESULT = {"WebLog":2}

leshric, 2020-12-13:
I'm new to Tasmota and just bought four Treatlife DS02S dimmers.  Experiencing a similar issue.  I've got release 9.1.0.

If I use the manual switches, or if I put the dimmer on a value too low (under 10%) using Home Assistant then it locks up and starts ignoring messages.  Getting TYA: Send dim skipped value XXX for dpid 2.

But, if I go to the Tasmota Web GUI and move the slider a couple times, then it starts responding again in HA.  Here's an example:

```
****Set 50% in HomeAssistant - SUCCESS****

04:20:13 MQT: Data Size 50, "cmnd/tasmota_840AF0/Backlog NoDelay;Fade 0;NoDelay;Dimmer 50;NoDelay;Power1 ON"
04:20:13 SRC: MQTT
04:20:13 CMD: Group 0, Index 1, Command "BACKLOG", Data "NoDelay;Fade 0;NoDelay;Dimmer 50;NoDelay;Power1 ON"
04:20:13 SRC: Backlog
04:20:13 CMD: Group 0, Index 1, Command "FADE", Data "0"
04:20:13 MQT: stat/tasmota_840AF0/RESULT = {"Fade":"OFF"}
04:20:13 SRC: Backlog
04:20:13 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
04:20:13 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON","Dimmer":50}
04:20:13 SRC: Backlog
04:20:13 CMD: Group 0, Index 1, Command "POWER", Data "ON"
04:20:13 TYA: Send "55aa0006000501010001010e"
04:20:13 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON"}
04:20:13 MQT: stat/tasmota_840AF0/POWER = ON
04:20:13 {"TuyaReceived":{"Data":"55AA0307000802020004000000D2EB","Cmnd":7,"CmndData":"02020004000000D2","DpType2Id2":210,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000D2"}}}
04:20:13 TYA: fnId=21 is set for dpId=2
04:20:13 TYA: RX value 210 from dpId 2 
04:20:13 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
04:20:13 TYA: fnId=11 is set for dpId=1
04:20:13 TYA: RX Relay-1 --> MCU State: On Current State:On
04:20:14 TYA: Send "55aa0006000802020004000001f40a"
04:20:14 CFG: Saved to flash at F5, Count 295, Bytes 4096
04:20:14 {"TuyaReceived":{"Data":"55AA0307000802020004000001F40E","Cmnd":7,"CmndData":"02020004000001F4","DpType2Id2":500,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000001F4"}}}
04:20:14 TYA: fnId=21 is set for dpId=2
04:20:14 TYA: RX value 500 from dpId 2

****Clicked + Switch Once - SUCCESS****
 
04:20:17 {"TuyaReceived":{"Data":"55AA0307000802020004000002021D","Cmnd":7,"CmndData":"0202000400000202","DpType2Id2":514,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000202"}}}
04:20:17 TYA: fnId=21 is set for dpId=2
04:20:17 TYA: RX value 514 from dpId 2 
04:20:17 TYA: Send dim skipped value 500 for dpid 2

****Set 75% in HomeAssistant - FAIL****

04:20:20 MQT: Data Size 50, "cmnd/tasmota_840AF0/Backlog NoDelay;Fade 0;NoDelay;Dimmer 75;NoDelay;Power1 ON"
04:20:20 SRC: MQTT
04:20:20 CMD: Group 0, Index 1, Command "BACKLOG", Data "NoDelay;Fade 0;NoDelay;Dimmer 75;NoDelay;Power1 ON"
04:20:20 SRC: Backlog
04:20:20 CMD: Group 0, Index 1, Command "FADE", Data "0"
04:20:20 MQT: stat/tasmota_840AF0/RESULT = {"Fade":"OFF"}
04:20:20 SRC: Backlog
04:20:20 CMD: Group 0, Index 1, Command "DIMMER", Data "75"
04:20:20 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON","Dimmer":75}
04:20:20 SRC: Backlog
04:20:20 CMD: Group 0, Index 1, Command "POWER", Data "ON"
04:20:20 TYA: Send "55aa0006000501010001010e"
04:20:20 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON"}
04:20:20 MQT: stat/tasmota_840AF0/POWER = ON
04:20:20 {"TuyaReceived":{"Data":"55AA0307000802020004000002021D","Cmnd":7,"CmndData":"0202000400000202","DpType2Id2":514,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000202"}}}
04:20:20 TYA: fnId=21 is set for dpId=2
04:20:20 TYA: RX value 514 from dpId 2 
04:20:20 SRC: Switch
04:20:20 CMD: Group 0, Index 3, Command "DIMMER", Data "51"
04:20:20 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON","Dimmer":51}
04:20:20 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
04:20:20 TYA: fnId=11 is set for dpId=1
04:20:20 TYA: RX Relay-1 --> MCU State: On Current State:On
04:20:20 CFG: Saved to flash at F4, Count 296, Bytes 4096
04:20:23 TYA: Send "55aa00000000ff"
04:20:23 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
[... 58 lines cut ...]
04:20:36 SRC: Backlog
04:20:36 CMD: Group 0, Index 1, Command "FADE", Data "0"
04:20:36 MQT: stat/tasmota_840AF0/RESULT = {"Fade":"OFF"}
04:20:36 SRC: Backlog
04:20:36 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
04:20:36 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON","Dimmer":50}
04:20:36 SRC: Backlog
04:20:36 CMD: Group 0, Index 1, Command "POWER", Data "ON"
04:20:36 TYA: Send "55aa0006000501010001010e"
04:20:36 MQT: stat/tasmota_840AF0/RESULT = {"POWER":"ON"}
04:20:36 MQT: stat/tasmota_840AF0/POWER = ON
04:20:36 {"TuyaReceived":{"Data":"55AA0307000802020004000000D2EB","Cmnd":7,"CmndData":"02020004000000D2","DpType2Id2":210,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000D2"}}}
04:20:36 TYA: fnId=21 is set for dpId=2
04:20:36 TYA: RX value 210 from dpId 2 
04:20:36 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
04:20:36 TYA: fnId=11 is set for dpId=1
04:20:36 TYA: RX Relay-1 --> MCU State: On Current State:On
04:20:37 TYA: Send "55aa0006000802020004000001f40a"
04:20:37 CFG: Saved to flash at F8, Count 300, Bytes 4096
04:20:37 {"TuyaReceived":{"Data":"55AA031C00001E","Cmnd":28}}
04:20:37 TYA: RX unknown command
04:20:37 {"TuyaReceived":{"Data":"55AA0307000802020004000001F40E","Cmnd":7,"CmndData":"02020004000001F4","DpType2Id2":500,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000001F4"}}}
04:20:37 TYA: fnId=21 is set for dpId=2
04:20:37 TYA: RX value 500 from dpId 2
```

vajonam, 2020-12-18:
@effelle 

I am seeing this behavior when the MCU is stuck at some position and hence doesn't allow the dimmer to move the brightness.

Turning on device
```
21:58:30 CMD: Group 0, Index 1, Command "POWER", Data "1"
21:58:30 TYA: Send "55aa0006000501010001010e"
21:58:30 MQT: stat/dimmer_04/RESULT = {"POWER":"ON"}
21:58:30 MQT: stat/dimmer_04/POWER = ON (retained)
21:58:30 {"TuyaReceived":{"Data":"55AA03070008030200040000006680","Cmnd":7,"CmndData":"0302000400000066","DpType2Id3":102,"3":{"DpId":3,"DpIdType":2,"DpIdData":"00000066"}}}
21:58:30 TYA: fnId=21 is set for dpId=3
21:58:30 TYA: RX value 102 from dpId 3 
21:58:30 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
21:58:30 TYA: fnId=11 is set for dpId=1
21:58:30 TYA: RX Relay-1 --> MCU State: On Current State:On
```
Setting to 60% but dimmer is stuck at 16%. I wasn't even physically near dimmer  at this time. 

```
21:58:31 MQT: Data Size 22, "cmnd/dimmer_04/Backlog Dimmer 60;    Power 1;"
21:58:31 SRC: MQTT
21:58:31 CMD: Group 0, Index 1, Command "BACKLOG", Data "Dimmer 60;    Power 1;"
21:58:31 SRC: Backlog
21:58:31 CMD: Group 0, Index 1, Command "DIMMER", Data "60"
21:58:31 MQT: stat/dimmer_04/RESULT = {"POWER":"ON","Dimmer":60}
21:58:31 MQT: Data Size 22, "cmnd/dimmer_04/Backlog Dimmer 60;    Power 1;"
21:58:31 SRC: MQTT
21:58:31 CMD: Group 0, Index 1, Command "BACKLOG", Data "Dimmer 60;    Power 1;"
21:58:31 TYA: Send dim skipped value 153 for dpid 3
21:58:31 SRC: Backlog
21:58:31 CMD: Group 0, Index 1, Command "POWER", Data "1"
21:58:31 TYA: Send "55aa0006000501010001010e"
21:58:31 MQT: stat/dimmer_04/RESULT = {"POWER":"ON"}
21:58:31 MQT: stat/dimmer_04/POWER = ON (retained)
21:58:32 {"TuyaReceived":{"Data":"55AA03070008030200040000002842","Cmnd":7,"CmndData":"0302000400000028","DpType2Id3":40,"3":{"DpId":3,"DpIdType":2,"DpIdData":"00000028"}}}
21:58:32 TYA: fnId=21 is set for dpId=3
21:58:32 TYA: RX value 40 from dpId 3 
21:58:32 SRC: Switch
21:58:32 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
21:58:32 MQT: stat/dimmer_04/RESULT = {"POWER":"ON","Dimmer":16}
21:58:32 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
21:58:32 TYA: fnId=11 is set for dpId=1
21:58:32 TYA: RX Relay-1 --> MCU State: On Current State:On
21:58:32 SRC: Backlog
21:58:32 CMD: Group 0, Index 1, Command "DIMMER", Data "60"
21:58:32 MQT: stat/dimmer_04/RESULT = {"POWER":"ON","Dimmer":60}
21:58:32 TYA: Send dim skipped value 153 for dpid 3
21:58:32 MQT: Data Size 22, "cmnd/dimmer_04/Backlog Dimmer 90;    Power 1;"
21:58:32 SRC: MQTT
21:58:32 CMD: Group 0, Index 1, Command "BACKLOG", Data "Dimmer 90;    Power 1;"
21:58:32 SRC: Backlog
21:58:32 CMD: Group 0, Index 1, Command "POWER", Data "1"
21:58:32 TYA: Send "55aa0006000501010001010e"
21:58:32 MQT: stat/dimmer_04/RESULT = {"POWER":"ON"}
21:58:32 MQT: stat/dimmer_04/POWER = ON (retained)
21:58:32 {"TuyaReceived":{"Data":"55AA03070008030200040000002842","Cmnd":7,"CmndData":"0302000400000028","DpType2Id3":40,"3":{"DpId":3,"DpIdType":2,"DpIdData":"00000028"}}}
21:58:32 TYA: fnId=21 is set for dpId=3
21:58:32 TYA: RX value 40 from dpId 3 
21:58:32 SRC: Switch
21:58:32 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
21:58:32 MQT: stat/dimmer_04/RESULT = {"POWER":"ON","Dimmer":16}
21:58:32 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
21:58:32 TYA: fnId=11 is set for dpId=1
21:58:32 TYA: RX Relay-1 --> MCU State: On Current State:On
21:58:32 SRC: Backlog
21:58:32 CMD: Group 0, Index 1, Command "DIMMER", Data "90"
21:58:32 MQT: stat/dimmer_04/RESULT = {"POWER":"ON","Dimmer":90}
21:58:32 TYA: Send dim skipped value 230 for dpid 3
21:58:32 SRC: Backlog
[... 273 lines cut ...]
22:00:12 TYA: RX value 40 from dpId 3 
22:00:12 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
22:00:12 TYA: fnId=11 is set for dpId=1
22:00:12 TYA: RX Relay-1 --> MCU State: On Current State:On
22:00:13 CFG: Saved to flash at FB, Count 902, Bytes 4096
22:00:14 MQT: Data Size 7, "cmnd/dimmer_04/Backlog Power 0"
22:00:14 SRC: MQTT
22:00:14 CMD: Group 0, Index 1, Command "BACKLOG", Data "Power 0"
22:00:14 SRC: Backlog
22:00:14 CMD: Group 0, Index 1, Command "POWER", Data "0"
22:00:14 TYA: Send "55aa0006000803020004000000283e"
22:00:14 TYA: Send "55aa0006000501010001000d"
22:00:14 MQT: stat/dimmer_04/RESULT = {"POWER":"OFF"}
22:00:14 MQT: stat/dimmer_04/POWER = OFF (retained)
22:00:14 {"TuyaReceived":{"Data":"55AA03070008030200040000002842","Cmnd":7,"CmndData":"0302000400000028","DpType2Id3":40,"3":{"DpId":3,"DpIdType":2,"DpIdData":"00000028"}}}
22:00:14 TYA: fnId=21 is set for dpId=3
22:00:14 TYA: RX value 40 from dpId 3 
22:00:14 {"TuyaReceived":{"Data":"55AA03070008030200040000002842","Cmnd":7,"CmndData":"0302000400000028","DpType2Id3":40,"3":{"DpId":3,"DpIdType":2,"DpIdData":"00000028"}}}
22:00:14 TYA: fnId=21 is set for dpId=3
22:00:14 TYA: RX value 40 from dpId 3 
22:00:14 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
22:00:14 TYA: fnId=11 is set for dpId=1
22:00:14 TYA: RX Relay-1 --> MCU State: Off Current State:Off
22:00:14 MQT: Data Size 8, "cmnd/dimmer_04/Backlog Power 1;"
```

vajonam, 2020-12-20:
@effelle 

got some more concrete steps to reprroduce.

- turn on dimmer with hardware button.
- you cannot then change the dimmer using Dimmer commands or MQTT commands. 
- turn off the dimmer by sliding 0 on webui or mqtt dimmer 0, then light goes off and them dimmer commands.

Seems that when a power on happens the MCU isn't sending the dimmer state when powered on from  hardware buttons. then it ignores the dimmer commands (skip) not sure why..

effelle, 2020-12-20:
I'll have a look after the holidays.
Meanwhile, what happens if you enable `SetOption59`?

fahubgit, 2020-12-28:
I used auto discovery on all my switches I installed in Home Assistant (SetOption19), which says it automatically turns on SetOption59. What exactly are you looking for with that and how can I test/troubleshoot it for you? Anything in something like MQTT Explorer?

vajonam, 2020-12-29:
Over the last few days I have been able to reproduce this. 

So20 1; So54 1; So59 1; 

- Turn on using faceplace/hardware button - works and is controllable via MQTT (Home Assistant)
- Turn off using POWER 0 from MQTT
- Turn on using POWER 1 from MQTT
- Now the dimmer will ignore all future dimming changes until, Power 0, and them Dimmer x is set and then power is set to 1.

fahubgit, 2021-01-18:
This issue is still showing awaiting feedback even though there's been feedback from at least two of us for a while and the bug tag was removed even though this is most certainly a bug. I'm just trying to figure out if there anything else we can provide for you guys to help with this issue? Please advise. I, and I'm sure others, are happy to help.

That being said, hopefully this will help...

Since the troubleshooting/resolution seemed to have stalled, I decided to do a little more investigating and thought that an older version of the firmware may exhibit different behavior, even if not the expected behavior (I wasn't sure if older firmware was able to work with the separate Tuya MCU). I figured it was worth a try. I decided to start at v8.1.0, so I downgraded to that firmware and to my surprise, everything seemed to be functioning as expected. Like everything. I was able to change the dimmer levels from the devices web page, Home Assistant, or the Alexa app and have it properly take effect and update the change in both Home Assistant and the Alexa app. I then tried using the hardware buttons on the switch and those changes were being reflected right away too. I then tested the last part, which is what was still not working in the newer firmwares (v9.1.0 and above), which is ability to change the dimmer level from Home Assistant or the Alexa app after using the hardware buttons, and the was working perfectly as well. I was able to use both immediately after utilizing the hardware buttons.

I then started trying increasingly higher firmware versions until it broke and all worked until v8.5.1, which was the newest version when I started using Tasmota coincidently. So whatever happened to break this broke in version v8.5.1, and still exists in v9.1.0 and at least v9.1.0.2 (the last dev version I tested).

For anyone still dealing with this issue, until it gets resolved, if having a fully functional switch is more important to you than being on the newest version of firmware, then I would suggest downgrading your firmware to v8.5.0 until this issue is resolved and we can upgrade again. That's what I'll be doing for all my devices for now.

arendst, 2021-01-22:
Thx for your research. It appears the tuya driver was massively refactored between 8.5.0 and 8.5.1 so apparently something went wrong there.

The Tuya driver has been managed by two skilled programmers both have left the scene so solving your issue is a kind of a challange....

vajonam, 2021-01-22:
that is not great news for us tuyamcu users :-( hopefully @effelle  will find his way back :-). However the current state of play means the tuyamcu stuff isn't completely broken, but some edge cases cause bit of frustration. downgrading to 8.5.0 might not be an option as some of the tuya refactoring was useful in simplifying tuya enums etc.

closed 2021-01-30 by github-actions[bot]