arendst/Tasmota #10081: TuyaMCU: TuyaSend3 X, X command not working

SergyiSviderskii, 2020-12-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
TuyaMCU: TuyaSend3 X, X command not working.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ yes ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ yes] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ yes] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ no] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [no ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Tuya thermostat____
- [ yes] Tasmota binary firmware version number used: _ 8.5.1, and  9.1.0___
  - [ yes] Pre-compiled
  - [yes ] Self-compiled
    - [yes ] IDE / Compiler used: __GitPod___
- [ yes] Flashing tools used: CH341A_____
- [ yes] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
1:38:06 CMD: Backlog Template
11:38:06 SRC: WebConsole from 192.168.0.106
11:38:06 CMD: Группа 0, Индекс 1, Команда "BACKLOG", Данные "Template"
11:38:06 SRC: Backlog
11:38:06 CMD: Группа 0, Индекс 1, Команда "TEMPLATE", Данные ""
11:38:06 MQT: stat/thermostatbathroom/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

11:40:29 CMD: Module
11:40:29 SRC: WebConsole from 192.168.0.106
11:40:29 CMD: Группа 0, Индекс 1, Команда "MODULE", Данные ""
11:40:29 MQT: stat/thermostatbathroom/RESULT = {"Module":{"54":"Tuya MCU"}}

11:41:47 CMD: GPIO 255
11:41:47 SRC: WebConsole from 192.168.0.106
11:41:47 CMD: Группа 0, Индекс 1, Команда "GPIO", Данные "255"
11:41:47 MQT: stat/thermostatbathroom/RESULT = {"GPIO0":{"0":"-нет-"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"-нет-"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"-нет-"},"GPIO5":{"0":"-нет-"},"GPIO9":{"0":"-нет-"},"GPIO10":{"0":"-нет-"},"GPIO12":{"0":"-нет-"},"GPIO13":{"0":"-нет-"},"GPIO14":{"0":"-нет-"},"GPIO15":{"0":"-нет-"},"GPIO16":{"0":"-нет-"},"GPIO17":{"0":"-нет-"}}

```
- [ no] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [yes ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:44:29 CMD: Группа 0, Индекс 1, Команда "STATUS", Данные "0"
11:44:29 MQT: stat/thermostatbathroom/STATUS = {"Status":{"Module":54,"DeviceName":"thermostatbathroom","FriendlyName":["Tasmota_thermostatbathroom"],"Topic":"thermostatbathroom","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:44:29 MQT: stat/thermostatbathroom/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:48:35","StartupUTC":"2020-12-07T06:55:54","Sleep":50,"CfgHolder":4617,"BootCount":3388,"BCResetTime":"2020-09-25T20:04:27","SaveCount":9336,"SaveAddress":"F5000"}}
11:44:29 MQT: stat/thermostatbathroom/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-12-02T08:34:14","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"484/699"}}
11:44:29 MQT: stat/thermostatbathroom/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HOME","HOME_IoT"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00C28009","2805C8000100060000005A00000000000000","00010220","00006000","00000000"]}}
11:44:29 MQT: stat/thermostatbathroom/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000419","07C06592","042E8001","000000C3","00000000","C000F981","00000004","00000000"],"Drivers":"1,2,4,6,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,4,5,6"}}
11:44:29 MQT: stat/thermostatbathroom/STATUS5 = {"StatusNET":{"Hostname":"thermostat_bathroom","IPAddress":"192.168.0.105","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.151.101","Mac":"84:F3:EB:EC:39:8E","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
11:44:29 MQT: stat/thermostatbathroom/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant.local","MqttPort":1883,"MqttClientMask":"thermostatbathroom_%06X","MqttClient":"thermostatbathroom_EC398E","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:44:29 MQT: stat/thermostatbathroom/STATUS7 = {"StatusTIM":{"UTC":"2020-12-07T09:44:29","Local":"2020-12-07T11:44:29","StartDST":"2020-10-25T00:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"09:30","Sunset":"17:53"}}
11:44:29 MQT: stat/thermostatbathroom/STATUS10 = {"StatusSNS":{"Time":"2020-12-07T11:44:29"}}
11:44:29 MQT: stat/thermostatbathroom/STATUS11 = {"StatusSTS":{"Time":"2020-12-07T11:44:29","Uptime":"0T02:48:35","UptimeSec":10115,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"HOME","BSSId":"00:1D:0F:EA:5A:92","Channel":7,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:18"}}}

```
- [yes ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

12:07:30 CMD: TuyaSend3 119,050614080015113015
12:07:30 SRC: WebConsole from 192.168.0.106
12:07:30 CMD: Группа 0, Индекс 3, Команда "TUYASEND", Данные "119,050614080015113015"
12:07:30 TYA: Send "55aa000600167703001230353036313430383030313531313330313530"
12:07:30 MQT: stat/thermostatbathroom/RESULT = {"TuyaSend":"Done"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

there is:
12:07:30 TYA: Send "55aa000600167703001230353036313430383030313531313330313530"
expected:
12:07:30 TYA: Send "55AA0006000D7700000905060E08000F0B1E0FFA"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

extra three after each digit.

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-12-07:
Tuyasend3 send a string in encoded hex format, then in your example:
```py
input: 050614080015113015
output: 303530363134303830303135313133303135
```
That is the default conversion for tuya string types. 
As important note, you dpId is labeled `119` and is a custom one added by the manufacturer, not Tuya. This means I have no idea what is expecting instead of the default values. 
How did  you end up with the string `55AA0006000D7700000905060E08000F0B1E0FFA`?
Did you find somewhere else?   
Show me something more and if is a common protocol I'll think on add it.

SergyiSviderskii, 2020-12-07:
data: 060015080016090020 for two times in a hex, to the command: 55AA0006000D77000009 and the sum.

checked on:
SerialSend5 55AA0006000D7700000905060E08000F0B1E0FFA
all is well:
16:16:18 CMD: SerialSend5 55AA0006000D7700000905060E08000F0B1E0FFA
16:16:18 SRC: WebConsole from 192.168.0.106
16:16:18 CMD: Группа 0, Индекс 5, Команда "SERIALSEND", Данные "55AA0006000D7700000905060E08000F0B1E0FFA"
16:16:18 MQT: stat/thermostatbathroom/RESULT = {"SerialSend":"Done"}

16:13:19 MQT: tele/thermostatbathroom/RESULT = {"TuyaReceived":{"Data":"55AA0107000D7700000905060E08000F0B1E0FFC","Cmnd":7,"CmndData":"7700000905060E08000F0B1E0F","DpType0Id119":"0x05060E08000F0B1E0F","119":{"DpId":119,"DpIdType":0,"DpIdData":"05060E08000F0B1E0F"}}}
16:13:19 TYA: fnId=0 is set for dpId=119

SergyiSviderskii, 2020-12-07:
there are more such registers:
there are more such registers:

there are more such registers:

16:38:16 MQT: tele/thermostatbathroom/RESULT = {"TuyaReceived":{"Data":"55AA0107000D7700000905060E08000F0B1E0FFC","Cmnd":7,"CmndData":"7700000905060E08000F0B1E0F","DpType0Id119":"0x05060E08000F0B1E0F","119":{"DpId":119,"DpIdType":0,"DpIdData":"05060E08000F0B1E0F"}}}
16:38:16 TYA: fnId=0 is set for dpId=119
16:38:16 MQT: tele/thermostatbathroom/RESULT = {"TuyaReceived":{"Data":"55AA0107000D780000098D1E0F11000F16000F94","Cmnd":7,"CmndData":"780000098D1E0F11000F16000F","DpType0Id120":"0x8D1E0F11000F16000F","120":{"DpId":120,"DpIdType":0,"DpIdData":"8D1E0F11000F16000F"}}}
16:38:16 TYA: fnId=0 is set for dpId=120
16:38:16 MQT: tele/thermostatbathroom/RESULT = {"TuyaReceived":{"Data":"55AA0107000D7900000906000F08000F0B1E0FFA","Cmnd":7,"CmndData":"7900000906000F08000F0B1E0F","DpType0Id121":"0x06000F08000F0B1E0F","121":{"DpId":121,"DpIdType":0,"DpIdData":"06000F08000F0B1E0F"}}}
16:38:16 TYA: fnId=0 is set for dpId=121
16:38:16 MQT: tele/thermostatbathroom/RESULT = {"TuyaReceived":{"Data":"55AA0107000D7A0000090D1E0F11000F16000F16","Cmnd":7,"CmndData":"7A0000090D1E0F11000F16000F","DpType0Id122":"0x0D1E0F11000F16000F","122":{"DpId":122,"DpIdType":0,"DpIdData":"0D1E0F11000F16000F"}}}
16:38:16 TYA: fnId=0 is set for dpId=122

this is a schedule of 6 time points and 7 or 5 + 2 or 6 + 1 weeks.

effelle, 2020-12-07:
> data: 060015080016090020 for two times in a hex, to the command: 55AA0006000D77000009 and the sum.

I'm sorry, I don't understand what you mean here. Care to elaborate a bit more? 

> there are more such registers:

Ok, but as I said you are using custom dpId (119, 120, 121, 122) and we can't support them since can be anything depending by the manufacturer. 
I'm afraid your only solution will be to use SerialSend5 with the known codes.

SergyiSviderskii, 2020-12-07:
"data: 060015080016090020 for two times in a hex, to the command: 55AA0006000D77000009 and the sum." 
my English is bad.
                                         060015080016090020
55AA0006000D7700000906000F080010090014 + two bytes sum.

I'm understood, thank you.

there is one more question TuyaMCU, will prepare and ask.
thanks again.

SergyiSviderskii, 2020-12-07:
"How did you end up with the string 55AA0006000D7700000905060E08000F0B1E0FFA?"

05 06 14 08 00 15 11 30 15 => 05 06 0E 08 00 0F 0B 1E 0F
55AA0006000D7700000905060E08000F0B1E0F FA,    FA-sum.

closed 2020-12-08 by SergyiSviderskii