arendst/Tasmota #10089: Exception 29 on a Novostella E27 Bulb

imduffy15, 2020-12-07:
### PROBLEM DESCRIPTION

The bulb reboots regularly with the following shown for `status 12`:

```
00:17:54 RSL: stat/tasmota_A06CF6/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40100430","40235f57","40235eec","40235e93","40234ff4","4023501d","40232a88","4022ee01","4023377f","40000f58","402331f2","4023c953","4023c20f","4021b82c","40000f49","40000f49","40000e19","40104df5","40104dfb","4010000d","4023b0f0","4023b0a1","40100214","40105595","4000050c","402212f0","40221117","40220c18","40100214","40101a3c","40104cd7"]}}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tuya Convert / OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
00:20:41 CMD: Backlog Template; Module; GPIO 255
00:20:41 RSL: stat/tasmota_A06CF6/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:20:41 RSL: stat/tasmota_A06CF6/RESULT = {"Module":{"1":"Sonoff Basic"}}
00:20:41 RSL: stat/tasmota_A06CF6/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
00:22:03 CMD: status 0
00:22:03 RSL: stat/tasmota_A06CF6/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A06CF6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:07","StartupUTC":"2020-12-07T23:21:56","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-12-08T00:06:20","SaveCount":13,"SaveAddress":"FB000"}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS2 = {"StatusFWR":{"Version":"9.1.0(minimal)","BuildDateTime":"2020-11-07T11:58:20","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"352/699"}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["QUINTESSENTIAL",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","10000000","00006000","00000000"]}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS4 = {"StatusMEM":{"ProgramSize":378,"Free":624,"Heap":33,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","00000580","04000003","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2","Sensors":""}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS5 = {"StatusNET":{"Hostname":"tasmota_A06CF6-3318","IPAddress":"192.168.20.146","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.254","Mac":"BC:DD:C2:A0:6C:F6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A06CF6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS7 = {"StatusTIM":{"UTC":"2020-12-07T23:22:03","Local":"2020-12-08T00:22:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS10 = {"StatusSNS":{"Time":"2020-12-08T00:22:03"}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS11 = {"StatusSTS":{"Time":"2020-12-08T00:22:03","Uptime":"0T00:00:07","UptimeSec":7,"Vcc":3.462,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"QUINTESSENTIAL","BSSId":"E0:63:DA:CA:C6:7A","Channel":6,"RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:04"}}}
00:22:03 RSL: stat/tasmota_A06CF6/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40100430","40235f57","40235eec","40235e93","40234ff4","4023501d","40232a88","4022ee01","4023377f","40000f58","402331f2","4023c953","4023c20f","4021b82c","40000f49","40000f49","40000e19","40104df5","40104dfb","4010000d","4023b0f0","4023b0a1","40220ce2","40100214","40101a3c","40104cef","401025d3","401027b4","40100214","40102c76","40104c3d"]}}
00:22:06 RSL: tele/tasmota_A06CF6/STATE = {"Time":"2020-12-08T00:22:06","Uptime":"0T00:00:10","UptimeSec":10,"Vcc":3.471,"Heap":33,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"QUINTESSENTIAL","BSSId":"E0:63:DA:CA:C6:7A","Channel":6,"RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:04"}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Unsure, just leave the device powered on for a couple of minutes and it crashes and restarts

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The device to not crash/restart

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I have tried `reset 3` I have tried flashing older versions of tasmota, going to espurna and coming back but I seem to be unable to shift it.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-08:
Hi,

Exception 29 is sometimes due to a bad flashing or not erasing before flashing. 

Please, update Tasmota to latest (9.1.0.2) http://ota.tasmota.com/tasmota/tasmota.bin and then go to the console and type reset 6 in order to clean your flash.

After that, please, configure your device again. Thanks.

imduffy15, 2020-12-08:
Hi @ascillato2 I have tried your suggestion but I’m afraid it did not resolve the issue. The device continues to throw the exception and restart. 

Thanks,
Ian

ascillato, 2020-12-08:
In that case, seems that your device has a hardware issue not related to Tasmota or Espurna. Are you able to claim the warranty?

imduffy15, 2020-12-08:
Oh that is interesting and good to know! Not able to claim the warranty ad it’s been flashed with tasmota but not to worry, will just replace it.  

Thank you for your time and assistance.

sfromis, 2020-12-08:
Another risk is that your settings drive the bulb harder than the power supply in it supports, leading to overheating and crashing.

closed 2020-12-08 by ascillato2