arendst/Tasmota #10090: Ikea remote is unknown device after pairing to Sonoff Zigbee Bridge

Carl3k, 2020-12-08:
### PROBLEM DESCRIPTION
After pairing an Ikea remote (type E1524) to a Sonoff Zigbee Bridge ZbStatus2 reports "Unknown device".

### REQUESTED INFORMATION

- [#] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [#] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [#] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [#] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [#] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [#] Device used: Sonoff Zigbee Bridge
- [#] Tasmota binary firmware version number used: 9.1.0
  - [#] Pre-compiled
- [#] Flashing tools used: esptool.py
- [#] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
MQT: stat/zbbridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
MQT: stat/zbbridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
MQT: stat/zbbridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```

```
- [#] Provide the output of this command: `Status 0`:
```
MQT: stat/zbbridge/STATUS = {"Status":{"Module":75,"DeviceName":"zbbridge","FriendlyName":["zbbridge"],"Topic":"zbbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
MQT: stat/zbbridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:05:01","StartupUTC":"2020-12-08T06:39:15","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-12-04T17:54:55","SaveCount":20,"SaveAddress":"F7000"}}
MQT: stat/zbbridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0(zbbridge)","BuildDateTime":"2020-11-07T12:01:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"396/699"}}
MQT: stat/zbbridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MOM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
MQT: stat/zbbridge/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
MQT: stat/zbbridge/STATUS5 = {"StatusNET":{"Hostname":"zbbridge","IPAddress":"192.168.2.178","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.9","Mac":"98:F4:AB:E2:00:7F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
MQT: stat/zbbridge/STATUS6 = {"StatusMQT":{"MqttHost":"house.mom.home","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E2FF7F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
MQT: stat/zbbridge/STATUS7 = {"StatusTIM":{"UTC":"2020-12-08T06:44:16","Local":"2020-12-08T07:44:16","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
MQT: stat/zbbridge/STATUS10 = {"StatusSNS":{"Time":"2020-12-08T07:44:16"}}
MQT: stat/zbbridge/STATUS11 = {"StatusSTS":{"Time":"2020-12-08T07:44:16","Uptime":"0T00:05:01","UptimeSec":301,"Vcc":3.514,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MOM","BSSId":"E4:95:6E:12:00:66","Channel":1,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}
MQT: tele/zbbridge/STATE = {"Time":"2020-12-08T07:44:24","Uptime":"0T00:05:09","UptimeSec":309,"Vcc":3.523,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MOM","BSSId":"E4:95:6E:12:00:66","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}

```
- [#] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
CMD: ZbPermitJoin 1
SRC: WebConsole from 192.168.2.164
CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
ZIG: ZbEZSPSend 22003C
ZIG: adding packet to_send, to_ack:6, to_send:6, to_end:6
ZIG: ZbEZSPSend 3600FCFF00003600000000000000071E0103073C01
ZIG: adding packet to_send, to_ack:6, to_send:6, to_end:7
MQT: tele/zbbridge/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
zigbee.permit_end_time = 800540
MQT: stat/zbbridge/RESULT = {"ZbPermitJoin":"Done"}
ZIG: Something to_send, to_ack:6, to_send:6, to_end:0
ZIG: ZbEZSPSentRaw 6726000122003C
ZIG: {"ZbEZSPReceived2":"77268001220000"}
ZIG: new ack/data received, was 6 now 7
ZIG: freeing packet 6 from memory
ZIG: ZbEZSPSentRaw 80
ZIG: {"ZbEZSPReceived":"220000"}
ZIG: Something to_send, to_ack:7, to_send:7, to_end:0
ZIG: ZbEZSPSentRaw 702700013600FCFF00003600000000000000071E0103073C01
ZIG: {"ZbEZSPReceived2":"002780013600007D"}
ZIG: new ack/data received, was 7 now 0
ZIG: freeing packet 7 from memory
ZIG: ZbEZSPSentRaw 81
ZIG: {"ZbEZSPReceived":"3600007D"}
ZIG: {"ZbEZSPReceived2":"10279001450005000036000000000000007DFF000000FFFF03073C01"}
ZIG: ZbEZSPSentRaw 82
ZIG: {"ZbEZSPReceived":"450005000036000000000000007DFF000000FFFF03073C01"}
[... 174 lines cut ...]
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401060001FF00012B3BE0000000"}
ZIG: Sniffing group 0x3B2B
WIF: Checking connection...
ZIG: {"ZbEZSPReceived2":"572E90013F0003FDFF0401060001FF00012B3BE1000000"}
ZIG: ZbEZSPSentRaw 86
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401060001FF00012B3BE1000000"}
ZIG: Sniffing group 0x3B2B
MQT: tele/zbbridge/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- ZbPermitJoin 1
- pressing pairing button on remote 4 times within 5 seconds, red LED on remote front blinks
- ZbStatus2 shows MQT: stat/zbbridge/RESULT = {"ZbStatus":"Unknown device"}

### EXPECTED BEHAVIOUR
The remote is recognised by the bridge, button presses are reported.

### SCREENSHOTS
Both devices do not have a screen.

### ADDITIONAL CONTEXT
Bridge has been flashed based on instructions from https://zigbee.blakadder.com/Sonoff_ZBBridge.html .

s-hadinger, 2020-12-08:
Type `ZbStatus2 0x281D` or `ZbStatus` to get the whole list of devices

Carl3k, 2020-12-08:
That indeed shows some details:
```
CMD: ZbStatus
MQT: stat/zbbridge/RESULT = {"ZbStatus1":[{"Device":"0x281D"}]}

CMD: ZbStatus2 0x281D
MQT: stat/zbbridge/RESULT = {"ZbStatus2":[{"Device":"0x281D","IEEEAddr":"0xD0CF5EFFFEEBE7BB","ModelId":"TRADFRI remote control","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[]}]}
```

Problem is that pressing buttons on the remote only activates the Zigbee LED on the bridge. No MQTT message is generated. Has something else to be done besides pairing?

s-hadinger, 2020-12-08:
No, from your logs the binding went ok.

Please reach out via Discord.

Carl3k, 2020-12-09:
Will ask on Discord as adviced.

For reference, these are log entries for button presses after setting WebLog to 4:

center button
```
ZIG: {"ZbEZSPReceived2":"233A90013F0003FDFF0401060001FF00012B3BFB000000"}
ZIG: ZbEZSPSentRaw 83
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401060001FF00012B3BFB000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"333A90013F0003FDFF0401060001FF00012B3BFC000000"}
ZIG: ZbEZSPSentRaw 84
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401060001FF00012B3BFC000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"433A90013F0003FDFF0401060001FF00012B3BFD000000"}
ZIG: ZbEZSPSentRaw 85
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401060001FF00012B3BFD000000"}
ZIG: Sniffing group 0x3B2B
```

top button
```
ZIG: {"ZbEZSPReceived2":"533A90013F0003FDFF0401080001FF00012B3BFE000000"}
ZIG: ZbEZSPSentRaw 86
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3BFE000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"633A90013F0003FDFF0401080001FF00012B3BFF000000"}
ZIG: ZbEZSPSentRaw 87
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3BFF000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"733A90013F0003FDFF0401080001FF00012B3B00000000"}
ZIG: ZbEZSPSentRaw 80
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3B00000000"}
ZIG: Sniffing group 0x3B2B
```

bottom button
```
ZIG: {"ZbEZSPReceived2":"033A90013F0003FDFF0401080001FF00012B3B01000000"}
ZIG: ZbEZSPSentRaw 81
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3B01000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"133A90013F0003FDFF0401080001FF00012B3B02000000"}
ZIG: ZbEZSPSentRaw 82
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3B02000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"233A90013F0003FDFF0401080001FF00012B3B03000000"}
ZIG: ZbEZSPSentRaw 83
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401080001FF00012B3B03000000"}
ZIG: Sniffing group 0x3B2B
```

left button
```
ZIG: {"ZbEZSPReceived2":"333A90013F0003FDFF0401050001FF00012B3B04000000"}
ZIG: ZbEZSPSentRaw 84
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B04000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"433A90013F0003FDFF0401050001FF00012B3B05000000"}
ZIG: ZbEZSPSentRaw 85
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B05000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"533A90013F0003FDFF0401050001FF00012B3B06000000"}
ZIG: ZbEZSPSentRaw 86
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B06000000"}
ZIG: Sniffing group 0x3B2B
```

right button
```
ZIG: {"ZbEZSPReceived2":"633A90013F0003FDFF0401050001FF00012B3B07000000"}
ZIG: ZbEZSPSentRaw 87
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B07000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"733A90013F0003FDFF0401050001FF00012B3B08000000"}
ZIG: ZbEZSPSentRaw 80
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B08000000"}
ZIG: Sniffing group 0x3B2B
ZIG: {"ZbEZSPReceived2":"033A90013F0003FDFF0401050001FF00012B3B09000000"}
ZIG: ZbEZSPSentRaw 81
ZIG: {"ZbEZSPReceived":"3F0003FDFF0401050001FF00012B3B09000000"}
ZIG: Sniffing group 0x3B2B
```

s-hadinger, 2020-12-09:
From your logs it looks like you have an old firmware in your device where group addresses are fixed in advance.

Your group is 0x3B2B or 15147.

Type `ZbListen 15147` and you will see the events coming in MQTT. Be aware that you need to put a rule to run `ZbListen1 15147` at each restart:

Example:
```
Rule1 on zbstate#status==0 do backlog zblisten1 15147 endon 
Rule1 1
```

Carl3k, 2020-12-09:
Brilliant, thanks a lot for your analysis. Setting ZbListen indeed works, MQTT messages are created.

I bought the remote a long time ago. Now after I got the Sonoff Zigbee Bridge it was the only device to test functionality of the setup. It is very likely that firmware in the remote is old.

Is there any other way to do a firmware update on the device than finding an Ikea bridge or someone with a Zigbee2mqtt setup?

s-hadinger, 2020-12-09:
Software update of IKEA devices works with an IKEA gateway. But I believe you can also use Zigbee2MQTT.

OTA Zigbee update is not supported by Tasmota and will probably never be. ESP8266 is too limited for this.

Carl3k, 2020-12-09:
Thanks a lot for confirming the options to update Ikea devices. You are a true Zigbee expert. :)

closed 2020-12-09 by Carl3k

JoeWiseman, 2021-05-21:
Sorry to open this issue again ....
I've got the same problems described above ... So I thought about buying a IKEA-gateway to upgrade my remotes. All went fine with that, so that the firmware has been updated today successfully.
When connecting the remote to the ZBbridge (which was flashed as told at: https://zigbee.blakadder.com/Sonoff_ZBBridge.html eith the lates firmware: ncp-uart-sw_6.7.8_115200.ota and works allready fine with Tuya temperature sensors) the same problem described above appears. Of course the same workaround is helping.... But I thought when uprading firmware it should work without the listen-option....
Is there any way to implement this, so that in future versions it will be possible to use the IKEA tradfri without  that workaround?
Regards
Joe